MichaIng/DietPi #2683: Neo2 image aplay -l shows wrong processor

avworldme, 2019-04-01:
I have a Neo2 confirmed with the H5 processor.  However when I issue aplay -l I get:

card 0: Codec [H3 Audio Codec], device 0: CDC PCM Codec-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: allwinnerhdmi [allwinner,hdmi], device 0: 1c22800.i2s-i2s-hifi i2s-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Notice the H3Audio Codec instead of H5.  Is that normal?

Also there is no option to enable the Audio Hat [PCM5102](http://wiki.friendlyarm.com/wiki/index.php/NanoHat_PCM5102A)

Although on the FriendyElec image there is that option.

MichaIng, 2019-04-04:
@avworldme 
According to the link, I2S needs to be enabled, let me investigate how to achieve that on NEO2.

avworldme, 2019-04-04:
yes, in their npi-config setup i have to first enable i2s then the audio.  if you need a Neo2 and an audio hat I am more than happy to donate mine

MichaIng, 2019-04-04:
@avworldme 
Please try to disable all the serial console. AFAIK it uses the same GPIO pins:
`dietpi-config > Advanced Options > Serial console`

The default UART0 serial console is on the other GPIO fin row but the way we currently enable serial login consoles on all known serial devices might block the other one too. I am already working on a better configuration menu that allows to toggle serial console individually on all found serial devices: https://github.com/MichaIng/DietPi/pull/2678

avworldme, 2019-04-04:
I disabled the serial and ran a firmware update.  Still only get the above devices. Are there any additional steps after disable serial?

MichaIng, 2019-04-04:
I just checked the `npi-config` code. When you use it, doesn't it throw some error? It accesses dtoverlays at /boot but there are non with out image. I now downloaded the FriendlyCore image and it has only one single overlay in /boot/overlays. So I believe now that those are contained in `/boot/sun50i-h5-nanopi-neo2.dtb` somehow. Are those perhaps created as dtb(o) files when activated, does `/boot/overlays` exist on your system?

The fresh install (other thread) did not solve something, did it?

avworldme, 2019-04-05:
I don't get any errors when I use npi-config on FriendlyCore Xenial 4.14 image.  It activates the hat and it works.  On my FreindlyCore image is as you described.  /boot/overlays does exist and it has one file in it:  sun50i-h5-gpio-dvfs-overlay.dtb

MichaIng, 2019-04-06:
@avworldme 
> I don't get any errors when I use npi-config on FriendlyCore Xenial 4.14 image.

Ah I meant if you get an error on DietPi when runnin `npi-config`. And your should since:
- It contains to get the board info:
```
if [ -f /etc/friendlyelec-release ]; then
  # get friendlyelec's boardt model
  # example: 
  # BOARD="NanoPC-T2"
  # LINUXFAMILY=nanopi2
  . /etc/friendlyelec-release                       
else
    echo "Not found /etc/friendlyelec-release, Unable to determine the board model."
    return 1
fi
KERN_VERSION=`uname -r`
BOARD_LC=$(echo $BOARD|tr '[:upper:]' '[:lower:]')
```
- But this file does not exist on our image.
- So when running npi-config you should only see `Not found /etc/friendlyelec-release, Unable to determine the board model.` and the script should exit immediately.

It is not too hard to create this file with the correct info though:
```
cat << _EOF_ > /etc/friendlyelec-release
BOARD='NanoPi-NEO2'
LINUXFAMILY='Allwinnersun50iw2Family'
_EOF_
```
_____________
When done the above check if the config tool works. Generally you can check the important overlay states via:
```
fdtget /boot/sun50i-h5-nanopi-neo2.dtb pcm5102a status # Should be enabled
fdtget /boot/sun50i-h5-nanopi-neo2.dtb i2s0 status # Should be just an alias of above, not sure npi-config is not consistent there
fdtget /boot/sun50i-h5-nanopi-neo2.dtb i2c1 status # Needs to be DISABLED for I2S to work!
```

MichaIng, 2020-03-03:
New image based on Armbian Buster, might provide the required dtoverlay: https://dietpi.com/downloads/testing/DietPi_NanoPiNEO2-ARMv8-Buster.7z

MichaIng, 2020-06-24:
I mark this as closed as of new image. Feel free to reopen if required.

closed 2020-06-24 by MichaIng