MichaIng/DietPi #2685: Running dietpi-update fails on RPi-3

xanvincent, 2019-04-02:
### Creating a bug report/issue

#### Required Information
root@carbon:~# cat /DietPi/dietpi/.version
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH=master
G_GITOWNER=MichaIng

root@carbon:~# echo $G_DISTRO_NAME
stretch

root@carbon:~# uname -a
Linux carbon 4.19.30-v7+ #1208 SMP Thu Mar 21 22:32:04 GMT 2019 armv7l GNU/Linux

root@carbon:~# echo $G_HW_MODEL_DESCRIPTION
RPi 3 Model B+ (armv7l)

PowerSupply: 5V 2A

SDCard: Sandisk

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

Issue is with dietpi-update
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`
output: -bash: $'\r\E[K\E[0m\E[90m[\E[0m\E[32m': command not found

#### Steps to reproduce

1. run `dietpi-update`

#### Expected behaviour

- Update should complete as usual

#### Actual behaviour

root@carbon:~# dietpi-update
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[ INFO ] DietPi-Update | No response from: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6 ()
[ INFO ] DietPi-Update | Checking mirror: https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
[ INFO ] DietPi-Update | No response from: https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6 ()
[FAILED] DietPi-Update | Unable to access update servers. Please check your connection, then run dietpi-update again.

#### Extra details

-I can wget this file, although it throws a certificate error:
root@carbon:~# wget https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
--2019-04-02 11:56:03--  https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
Resolving dietpi.com (dietpi.com)... 104.27.179.199, 104.27.178.199, 2606:4700:30::681b:b2c7, ...
Connecting to dietpi.com (dietpi.com)|104.27.179.199|:443... connected.
ERROR: The certificate of ‘dietpi.com’ is not trusted.
ERROR: The certificate of ‘dietpi.com’ hasn't got a known issuer.

-I can ping dietpi.com successfully, so this isn't DNS-related.
root@carbon:~# ping dietpi.com
PING dietpi.com (104.27.179.199) 56(84) bytes of data.
64 bytes from 104.27.179.199: icmp_seq=1 ttl=53 time=22.9 ms
64 bytes from 104.27.179.199: icmp_seq=2 ttl=53 time=26.1 ms
^C
--- dietpi.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 22.978/24.544/26.111/1.574 ms

-Only software currently running are pihole and mariadb.

MichaIng, 2019-04-05:
@xanvincent 
Hmm it works fine here.
- Your system clock is also correct.

> ERROR: The certificate of ‘dietpi.com’ hasn't got a known issuer.

Hmm 🤔, `curl -L https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6` fails with the same error I guess?

Please try: `G_AGI --reinstall ca-certificates`

xanvincent, 2019-04-05:
@MichaIng 
That reinstall line did the trick. I guess my certs install was just broken. dietpi-update works fine now.

closed 2019-04-05 by xanvincent