MichaIng/DietPi #2690: Audiophonics I-Sabre 9038Q2M DAC driver | Now in kernel

Fourdee, 2019-04-05:
https://github.com/raspberrypi/linux/commit/d65a0f76d3adcf86a6f5c614c68edb3aeb3b8590

- Remove compile method.
- request/prompt/do rpi-update during selection of audio card.

Fourdee, 2019-04-06:
- q2m driver compatible with k2m?

Fourdee, 2019-04-06:
Pending ```rpi-update```:
```
root@DietPi:~# uname -a
Linux DietPi 4.19.32-v7+ #1210 SMP Tue Apr 2 14:07:04 BST 2019 armv7l GNU/Linux
```

MichaIng, 2019-04-06:
@Fourdee 
Actually I am not keen to add `rpi-update` again. This just ships a rats tail and it is still beta/unstable/testing. As long as the custom build works, I would leave it as is and remove after RPi kernel 4.19 has been released. This should not take long anymore.

What we could try is to simply copy the dtbo and check whether it's compatible with the current stable 4.14 kernel or not?

MichaIng, 2019-04-11:
Firmware update just arrived, sadly still on Kernel 4.14 😢.

`Waiting for user reply` = Waiting for 4.19 kernel package release. In any case we keep this in current milestone to not forget.

MichaIng, 2019-05-17:
Yey, RPi kernel 4.19 has just been released: https://github.com/Hexxeh/rpi-firmware/tree/stable
As I am anyway processing the RPi images I will redo them directly with this kernel.

MichaIng, 2019-05-25:
Possibly related: https://github.com/MichaIng/DietPi/issues/2855

Actually the dtoverlay is a different one:
- `i-sabre-q2m` vs `i-sabre-k2m`

So selecting `i-sabre-k2m` from dietpi-config like the initial PR intended it will not work. So either, if those are compatible, we need to only allow `i-sabre-q2m` selection, or internally rewrite it.

MichaIng, 2019-06-17:
Resolved with: https://github.com/MichaIng/DietPi/pull/2920

closed 2019-07-03 by MichaIng