MichaIng/DietPi #2691: DietPi-Software | WireGuard: install broke ipv6

schnuckz, 2019-04-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.22.3`
- Distro version | 9.8
- Kernel version | 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B (armv7l)
- Power supply used | 5V 2,4A
- SDcard used | SanDisk ultra 32gb

#### Additional Information (if applicable)
- Software title | Wireguard
- Was the software title installed freshly or updated/migrated?
no update, fresh install
- Can this issue be replicated on a fresh installation of DietPi?
yes

#### Steps to reproduce

1. install pihole
2. ping6 google.com works fine
3. install Wireguard
4. ping6 google.com dosent work

#### Expected behaviour
- no ipv6 address reachable

#### Actual behaviour
- no ipv6 connection
- also no ipv6 tunneling over Wireguard 

#### Extra details

**ip a / before install or after uninstall Wireguard, IPv6 connection works, ping6 works**

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 192.168.131.5/24 brd 192.168.131.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fd00::xxxx:xxxx:xxxx:xxxx/64 scope global mngtmpaddr dynamic 
       valid_lft 7166sec preferred_lft 3566sec
    inet6 2003:xx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx/64 scope global mngtmpaddr dynamic 
       valid_lft 7166sec preferred_lft 1685sec
    inet6 fe80::xxxx:xxxx:xxxx:xxxx/64 scope link 
       valid_lft forever preferred_lft forever
```

**ip a / after install Wiregaurd, no IPv6 connection, no ping6** 

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 192.168.131.5/24 brd 192.168.131.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::xxxx:xxxx:xxxx:xxxx/64 scope link 
       valid_lft forever preferred_lft forever
3: wg0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet 10.9.0.1/24 scope global wg0
       valid_lft forever preferred_lft forever
```

MichaIng, 2019-04-06:
@schnuckz 
Many thanks for your report. Is the below also true when stopping the WireGuard service?
`systemctl stop wg-quick@wg0`

MichaIng, 2019-04-06:
Hmm seems to work well here:
```
root@VM-Stretch:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.29/24 brd 192.168.178.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 xxxx:xxxx:xxxx:xxxx:xxx:xxxx:xxxx:xxxx/64 scope global mngtmpaddr dynamic
       valid_lft 7193sec preferred_lft 3593sec
    inet6 fe80::xxx:xxxx:xxxx:xxxx/64 scope link
       valid_lft forever preferred_lft forever
```
After installing WireGuard:
```
root@VM-Stretch:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.29/24 brd 192.168.178.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 xxxx:xxxx:xxxx:xxxx:xxx:xxxx:xxxx:xxxx/64 scope global mngtmpaddr dynamic
       valid_lft 7193sec preferred_lft 3593sec
    inet6 fe80::xxx:xxxx:xxxx:xxxx/64 scope link
       valid_lft forever preferred_lft forever
3: wg0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN group default qlen 1
    link/none
    inet 10.9.0.1/24 scope global wg0
       valid_lft forever preferred_lft forever
```
After reboot:
```
root@VM-Stretch:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.29/24 brd 192.168.178.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::xxx:xxxx:xxxx:xxxx/64 scope link
       valid_lft forever preferred_lft forever
3: wg0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN group default qlen 1
    link/none
    inet 10.9.0.1/24 scope global wg0
       valid_lft forever preferred_lft forever
```
- Ah now after reboot I have the issue here as well.

MichaIng, 2019-04-06:
IPv6 forwarding is the issue:
```
sysctl net.ipv6.conf.all.forwarding=0 net.ipv6.conf.default.forwarding=0
ifdown eth0 && ifup eth0
ip a # Now shows the assinged IPv6 address again
```

Indeed this seems to be the wanted behaviour: https://askubuntu.com/a/463654
This actually fits to some report that `::/0` in WireGuard client configs  breaks some connections.

@schnuckz 
Please try:
`sysctl net.ipv6.conf.all.accept_ra=2 net.ipv6.conf.default.accept_ra=2`
Then in case restart the interface:
`ifdown eth0 && ifup eth0`
And check if IPv6 connections work again, as well for clients when forwarded through the VPN.

schnuckz, 2019-04-07:
Hi @MichaIng 

thank you for your fast reply,

this works for me:

> Please try:
> `sysctl net.ipv6.conf.eth0.accept_ra=2`
> Then in case restart the interface:
> `ifdown eth0 && ifup eth0`
> And check if IPv6 connections work again, as well for clients when forwarded through the VPN.
> It works well here 😃.

But this works not for me, after an reboot no ipv6 connection
> So if it works to make the change persistent:
> 
> `echo -e 'net.ipv6.conf.all.accept_ra=2\nnet.ipv6.conf.default.accept_ra=2' >> /etc/sysctl.d/d`

also I changed wg0.conf, PostUp und PostDown to make ipv6 forwarding working:
```
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE; ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -t nat -D POSTROUTING -o eth0 -j MASQUERADE; ip6tables -D FORWARD -i %i -j ACCEPT; ip6tables -t nat -D POSTROUTING -o eth0 -j MASQUERADE
```

MichaIng, 2019-04-07:
> But this works not for me, after an reboot no ipv6 connection

Strange. Yeah somehow applying this particular setting to "all" does not work as expected as "all" does in other cases (e.g. ip_forwarding itself). Not sure for the reason.

Actually I found:
```
forwarding - BOOLEAN
	Enable IP forwarding on this interface.  This controls whether packets
	received _on_ this interface can be forwarded.
```
- So it should be sufficient to enable forwarding for `wg0`. Not sure if eth0 requires this to forward "answers" as well but at least we only define netfilter forwarding rules for wg0.

Can you please try:
`sysctl net.ipv6.conf.all.forwarding=0 net.ipv6.conf.default.forwarding=0 net.ipv6.conf.eth0.forwarding=0 net.ipv6.conf.wg0.forwarding=1`

> also I changed wg0.conf, PostUp und PostDown to make ipv6 forwarding working:

Damn you are right, this was missing as well.

I will setup a test server now to test all of this up and down.

MichaIng, 2019-04-08:
@schnuckz 
Okay so `ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o eth0 -j MASQUERADE` are required for sure and `sysctl net.ipv6.conf.<iface>.accept_ra=2` on the main interface.

On my home network I can't get IPv6 forwarding via WireGuard running with this. All peers can ping IPv6 hosts/IPs outside of WireGuard, the VPN server itself as well, the rules as above are all correct, but accessing or pinging IPv6 through VPN fails. Not sure if this is as I disabled DHCPv6 + ULA in router settings for my local network.

However I think it is best to move all the sysctl rules to the WireGuard PostUp config as well, so it can be done only for the required interfaces and especially the accept_ra=2 setting can be applied to the current  main interface. Otherwise we could need to guess or add any found (or possible) interface to the persistent sysctl.d config...
We will not change back the rules on PostDown just in case these rules were desired for another reason already, so to not override custom user settings. So enable the settings on demand with wg0 startup and leave them after stopping. iptables rules can be removed since those rules can stack/double. So it is impossible to remove/override a rule created by user as long as PostUp and PostDown match.

schnuckz, 2019-04-11:
With the following configs its works on my raspberry.

/etc/sysctl.d/dietpi-wireguard.conf
```
net.ipv4.ip_forward=1
net.ipv6.conf.all.forwarding=1
net.ipv6.conf.default.forwarding=1
net.ipv6.conf.all.accept_ra=2
net.ipv6.conf.default.accept_ra=2
net.ipv6.conf.eth0.accept_ra=2
```

wg0.conf 
```
[Interface]
Address = 10.9.0.1/24, fdxx:xxxx:xxxx:xxxx::1/64
PrivateKey = <privatekey>
ListenPort = 51820

PostUp = sysctl net.ipv6.conf.all.accept_ra=2 && sysctl net.ipv6.conf.default.accept_ra=2 && sysctl net.ipv6.conf.eth0.accept_ra=2 && iptables -A FORWARD -i %i -j ACCEPT; iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE; ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -t nat -D POSTROUTING -o eth0 -j MASQUERADE; ip6tables -D FORWARD -i %i -j ACCEPT; ip6tables -t nat -D POSTROUTING -o eth0 -j MASQUERADE

# Client 1
[Peer]
PublicKey = <publickey-client1>
AllowedIPs = 10.9.0.2/32, fdxx:xxxx:xxxx:xxxx::2/128

# Client 2
[Peer]
PublicKey = <publickey-client1>
AllowedIPs = 10.9.0.3/32, fdxx:xxxx:xxxx:xxxx::3/128
```

wg0-client.conf 
```
[Interface]
Address = 10.9.0.2/32, fdxx:xxxx:xxxx:xxxx::2/128
PrivateKey = <privatekey-client>

# Comment the following to preserve the clients default DNS server, or force a desired one.
DNS = 10.9.0.1, fdxx:xxxx:xxxx:xxxx::1

# Kill switch: Uncomment the following, if the client should stop any network traffic, when disconnected from the VPN server
# NB: This requires "iptables" to be installed, thus will most likely not work on mobile phones.
#PostUp = iptables -I OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT && ip6tables -I OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT
#PreDown = iptables -D OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT && ip6tables -D OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT

[Peer]
PublicKey = <publickey-server>
# Tunnel all network traffic through the VPN:
#	AllowedIPs = 0.0.0.0/0, ::/0
# Tunnel access to server-side local network only:
#	AllowedIPs = 192.168.xxx.0/24
# Tunnel access to VPN server only:
#	AllowedIPs = 192.168.xxx.xxx/32
AllowedIPs = 0.0.0.0/0, ::/0
Endpoint = xxxxxxxxxx.xx:51820

# Uncomment the following, if you're behind a NAT and want the connection to be kept alive.
#PersistentKeepalive = 25
```

Client config alternativ
Tunnel access to local network only and Pihole
```
#AllowedIPs = 192.168.xxx.0/24, fdxx::/64, 10.9.0.0/24, fdxx:xxxx:xxxx:xxxx::/64
```

schnuckz, 2019-04-17:
Hi, this doesn't work for me. 
After reboot no connection over vpn. I will manually use the command systemctl restart wg-quick@wg0 to establish a connection, but no ipv6 only ipv4.

> ### ADMIN EDIT
> #### Solution:
> ```
> rm /etc/sysctl.d/dietpi-wireguard.conf
> sed -i '/^ListenPort/a\PostUp = sysctl net.ipv4.conf.%i.forwarding=1 net.ipv4.conf.$(sed -n 3p /DietPi/dietpi/.network).forwarding=1' /etc/wireguard/wg0.conf
> sed -i '/^ListenPort/a\PostUp = sysctl net.ipv6.conf.$(sed -n 3p /DietPi/dietpi/.network).accept_ra=2' /etc/wireguard/wg0.conf
> sed -i '/^ListenPort/a\PostUp = sysctl net.ipv6.conf.%i.forwarding=1 net.ipv6.conf.$(sed -n 3p /DietPi/dietpi/.network).forwarding=1' /etc/wireguard/wg0.conf
> sed -i '/^ListenPort/a\PostUp = ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o $(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE' /etc/wireguard/wg0.conf
> sed -i '/^ListenPort/a\PostDown = ip6tables -D FORWARD -i %i -j ACCEPT; ip6tables -t nat -D POSTROUTING -o $(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE' /etc/wireguard/wg0.conf
> systemctl restart wg-quick@wg0
> ```

MichaIng, 2019-04-18:
@schnuckz 
I recognize now that you have WiFi Hotspot AND WireGuard installed, right?
What does `cat /DietPi/dietpi/.network` contain actually?
And could you paste final: `cat /etc/wireguard/wg0.conf`

> I will manually use the command systemctl restart wg-quick@wg0 to establish a connection

So WireGuard service failed on boot?
What does `journalctl -u wg-quick@wg0` show after reboot (before starting it manually)?

Since `wg-quick@.service` starts after `network-online.target`, it should start after `dietpi-boot` as well, thus `/DietPi/dietpi/.network` should already contain updated values.

With WiFi Hotspot active, the IP forwarding should double with `/etc/sysctl.d/dietpi-wifi_hotspot.conf` (or something like that), however should no harm.

`sysctl net.ipv6.conf.$(sed -n 3p /DietPi/dietpi/.network).accept_ra=2'` should make even more sense with WiFi Hotspot in combination, since:
- `net.ipv6.conf.all.forwarding=1` might lead to `accept_ra=2` being enabled for `wlan0` as well.
- But for the Hotspot is is actually desired that the interface is NOT auto-configured by some router since the Pi itself should serve as router/DHCP server on this interface.
- So only `eth0` should be allowed to be auto-configured by router/DHCP server.

Lets see the final sysctl values: `sysctl -a | grep -E 'ipv6.*\.(forwarding|accept_ra) ='`

schnuckz, 2019-04-19:
@MichaIng 

> So WireGuard service failed on boot?
WireGuard service started, but the PostUp commands not working.

> I recognize now that you have WiFi Hotspot AND WireGuard installed, right?
No, WiFiHotspot (TOR) on a second raspberry. Only installed WireGuard & PiHole with Lighttpd.

```
cat /etc/sysctl.d/dietpi-wireguard.conf 
#net.ipv4.ip_forward=1
#net.ipv6.conf.all.forwarding=1
#net.ipv6.conf.default.forwarding=1
#net.ipv6.conf.all.accept_ra=2
#net.ipv6.conf.default.accept_ra=2
#net.ipv6.conf.eth0.accept_ra=2
```

```
cat /DietPi/dietpi/.network 
0
0
eth0
192.168.131.5
ETH_IP=192.168.131.5
WLAN_IP=
```

```
cat /etc/wireguard/wg0.conf
[Interface]
Address = 10.9.0.1/24, fdxx:xxxx:xxxx:xxxx::1/64
PrivateKey = xxxxxxxxxxxx
ListenPort = 51820

PostUp = sysctl net.ipv4.conf.%i.forwarding=1 net.ipv4.conf.$(sed -n 3p /DietPi/dietpi/.network).forwarding=1
PostUp = sysctl net.ipv6.conf.%i.accept_ra=2 net.ipv6.conf.$(sed -n 3p /DietPi/dietpi/.network).accept_ra=2
PostUp = sysctl net.ipv6.conf.%i.forwarding=1 net.ipv6.conf.$(sed -n 3p /DietPi/dietpi/.network).forwarding=1
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -t nat -A POSTROUTING -o $(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE
PostUp = ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o $(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -t nat -D POSTROUTING -o $(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE
PostDown = ip6tables -D FORWARD -i %i -j ACCEPT; ip6tables -t nat -D POSTROUTING -o $(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE

#working after reboot
#PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE; ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
#PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -t nat -D POSTROUTING -o eth0 -j MASQUERADE; ip6tables -D FORWARD -i %i -j ACCEPT; ip6tables -t nat -D POSTROUTING -o eth0 -j MASQUERADE

# Client 1 - MacBook xxxx
[Peer]
PublicKey = xxxxxxxxxxxxxxx
AllowedIPs = 10.9.0.2/32, fdxx:xxxx:xxxx:xxxx::2/128

# Client 2 - iPhone xxx
[Peer]
PublicKey = xxxxxx
AllowedIPs = 10.9.0.3/32, fdxx:xxxx:xxxx:xxxx::3/128

# Client 3 - iPad xxx
[Peer]
PublicKey = xxxxxx
AllowedIPs = 10.9.0.4/32, fdxx:xxxx:xxxx:xxxx::4/128

# Client 4 - iPhone xxxxx
[Peer]
PublicKey = xxxxxx
AllowedIPs = 10.9.0.5/32, fdxx:xxxx:xxxx:xxxx::5/128

# Client 5 - MacBook xxxxxx
[Peer]
PublicKey = xxxxxxxxxxx
AllowedIPs = 10.9.0.6/32, fdxx:xxxx:xxxx:xxxx::6/128
```

```
sysctl -a | grep -E 'ipv6.*\.(forwarding|accept_ra) ='
sysctl: reading key "net.ipv6.conf.all.stable_secret"
net.ipv6.conf.all.accept_ra = 1
net.ipv6.conf.all.forwarding = 0
sysctl: reading key "net.ipv6.conf.default.stable_secret"
net.ipv6.conf.default.accept_ra = 1
sysctl: reading key "net.ipv6.conf.eth0.stable_secret"
net.ipv6.conf.default.forwarding = 0
net.ipv6.conf.eth0.accept_ra = 2
sysctl: reading key "net.ipv6.conf.lo.stable_secret"
net.ipv6.conf.eth0.forwarding = 1
sysctl: reading key "net.ipv6.conf.wg0.stable_secret"
net.ipv6.conf.lo.accept_ra = 1
net.ipv6.conf.lo.forwarding = 0
net.ipv6.conf.wg0.accept_ra = 2
net.ipv6.conf.wg0.forwarding = 1
```

```
journalctl -u wg-quick@wg0
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Fri 2019-04-19 06:52:14 CES
T. --
Apr 19 06:51:40 mjgs-rpi01 systemd[1]: Starting WireGuard via wg-quick(8) for wg
0...
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: [#] ip link add wg0 type wireguard
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: [#] wg setconf wg0 /dev/fd/63
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: [#] ip address add 10.9.0.1/24 dev wg0
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: [#] ip address add fdxx:xxxx:xxxx:xxxx
::1/64 dev wg0
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: [#] ip link set mtu 1420 up dev wg0
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: [#] sysctl net.ipv4.conf.wg0.forwardin
g=1 net.ipv4.conf.$(sed -n 3p /DietPi/dietpi/.network).forwarding=1
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: net.ipv4.conf.wg0.forwarding = 1
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: net.ipv4.conf.eth0.forwarding = 1
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: [#] sysctl net.ipv6.conf.wg0.accept_ra
=2 net.ipv6.conf.$(sed -n 3p /DietPi/dietpi/.network).accept_ra=2
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: net.ipv6.conf.wg0.accept_ra = 2
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: net.ipv6.conf.eth0.accept_ra = 2
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: [#] sysctl net.ipv6.conf.wg0.forwardin
g=1 net.ipv6.conf.$(sed -n 3p /DietPi/dietpi/.network).forwarding=1
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: net.ipv6.conf.wg0.forwarding = 1
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: net.ipv6.conf.eth0.forwarding = 1
Apr 19 06:51:41 mjgs-rpi01 wg-quick[566]: [#] iptables -A FORWARD -i wg0 -j ACCE
PT; iptables -t nat -A POSTROUTING -o $(sed -n 3p /DietPi/dietpi/.network) -j MA
SQUERADE
Apr 19 06:51:42 mjgs-rpi01 wg-quick[566]: [#] ip6tables -A FORWARD -i wg0 -j ACC
EPT; ip6tables -t nat -A POSTROUTING -o $(sed -n 3p /DietPi/dietpi/.network) -j 
MASQUERADE
Apr 19 06:51:56 mjgs-rpi01 systemd[1]: Started WireGuard via wg-quick(8) for wg0
.
```

I replace `$(sed -n 3p /DietPi/dietpi/.network)` with `eth0` in wg0.conf and only postup iptables, also my `/etc/sysctl.d/dietpi-wireguard.conf` and it works with ipv6 and after reboot. See my comment lines.

MichaIng, 2019-04-19:
@schnuckz 
Okay so obviously `$(sed -n 3p /DietPi/dietpi/.network)` is not the issue (since also with `eth0` sysctl settings are not applied), but that somehow this is not applied on boot.

Could you check at which stage WireGuard starts in your boot order:
`journalctl`
Then hit space until you reach the part where systemd starts services and seek out for WireGuard and DietPi services. Please paste the ~20 lines around WireGuard.

In my case it is starting after DietPi-Boot and sysctl settings are applied as desired, but perhaps this is not assured as I thought via `network-online.target` and we need to shift it.

After you restarted wg-quick manually, does IPv6 forwarding then work?
Ah `net.ipv6.conf.wg0.accept_ra = 2` that should actually not be required, since the server should set everything. Perhaps this might even cause the issue in your case if some client sends router advertisements. Please try to remove `net.ipv6.conf.%i.accept_ra=2`.

closed 2019-05-12 by MichaIng