MichaIng/DietPi #2694: DNS Forwarding - Suggesting supporting DuckDNS

spectrumcomputing, 2019-04-07:
DietPi provides support for No-IP which is great, but the issue for me is that No-IP does not support letsencrypt. When I'm using Nextcloud I can't connect via the Android app because it insists on https.

#### Describe the solution you'd like:
Please could you provide support for DuckDNS which allows you to use letsencrypt.

#### Describe alternatives you've considered:
I have researched and there seems no way of getting No-IP to work with letsencrypt.

MichaIng, 2019-04-07:
@spectrumcomputing 
Thanks for your report.

Let's Encrypt DOES support No-IP for quite some years now. Do you face some issue when trying to receive a certificate? If so it is not related to No-IP and then please paste some more details, error messages and such.

spectrumcomputing, 2019-04-07:
Thanks for the quick reply.  I'm using the free version of No-IP which I was under the impression does not support it.

MichaIng, 2019-04-07:
> which I was under the impression does not support it.

Did you try it? In case which top level domain did you choose?
AFAIK if it's inside this list, it will work definitely: https://publicsuffix.org/list/public_suffix_list.dat

spectrumcomputing, 2019-04-08:
Thanks, I will give it a go over the weekend.  I just took the info from here:

https://dietpi.com/phpbb/viewtopic.php?p=1061#p1062

NoIp is recommended for a URL/domain that points to your device. Free NoIp accounts do not work with Letsencrypt, you must own a domain name and use NoIp Premium features

Although I notice that post was from 2015.  I will update over the weekend. Thank you for your support and a great distro!

spectrumcomputing, 2019-04-17:
Apologies for the delay. I can confirm it does work on free accounts. Thank you.  Would it be possible to alter the installation instructions so that 'Free NoIp accounts do not work with Letsencrypt, you must own a domain name and use NoIp Premium features' is removed?

Thanks

MichaIng, 2019-04-17:
@spectrumcomputing 
Oh yeah, that is outdated info. Hum I thought we changed that already long time ago since this was mentioned before. Seems to have been forgotten again 🙈.

closed 2019-04-17 by MichaIng