MichaIng/DietPi #2695: Ampache does not playback .m4a files

spectrumcomputing, 2019-04-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
root@DietPi:~# #!/bin/bash
root@DietPi:~# G_DIETPI_VERSION_CORE=6
root@DietPi:~# G_DIETPI_VERSION_SUB=22
root@DietPi:~# G_DIETPI_VERSION_RC=3
root@DietPi:~# G_GITBRANCH=master
root@DietPi:~# G_GITOWNER=MichaIng

- Distro version | 
stretch
- Kernel version | 
Linux DietPi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B (armv7l)

- Power supply used | IQ Five Port Mains powered USB
- SDcard used | Sandisk Ultra

#### Additional Information (if applicable)
- Software title | Ampache
- Fresh Install
- Can this issue be replicated on a fresh installation of DietPi? Yes
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ... Place .m4a audio files in the music folder
2. ...Try and play with Ampache

#### Expected behaviour

- Music should play

#### Actual behaviour

- ... Ampache Freezes

#### Extra details

- ...

spectrumcomputing, 2019-04-08:
Hi again,

Following these instructions seems to fix it (at least in Subsonic). Is it possible to get this baked into DietPi, as well as upgrading to the new version of Subsonic:

http://pmdzsite.online.fr/index.php?article17/subsonic-on-raspberry-pi-3

Many thanks

MichaIng, 2019-04-09:
@spectrumcomputing 
Thanks for this info.

Ah I found the bug in our code. That was already found and fixed for Airsonic but with Subsonic it's the same. Will fix this, many thanks for reporting!

spectrumcomputing, 2019-04-09:
Subsonic now plays .m4a files perfectly. Thank you.  Ampache still does not.  This may help:

https://github.com/ampache/ampache/wiki/Transcoding

spectrumcomputing, 2019-04-10:
I enabled logging in Ampache by making some changes in the config file and this is the contents when I try and play .m4a files:

2019-04-10 19:47:32 [admin] (session) -> miudmi14q4v0gvd29vseo3g1m0 has been extended to Wed, 10 Apr 2019 20:47:32 +0100 extension length 3600 
2019-04-10 19:47:32 [admin] (session) -> Session created: 79c1c4dc84c22e022fc77ec029bd09ed 
2019-04-10 19:47:32 [admin] (session) -> miudmi14q4v0gvd29vseo3g1m0 has been extended to Wed, 10 Apr 2019 20:47:32 +0100 extension length 3600 
2019-04-10 19:47:32 [admin] (session) -> miudmi14q4v0gvd29vseo3g1m0 has been extended to Wed, 10 Apr 2019 20:47:32 +0100 extension length 3600 
2019-04-10 19:47:32 [admin] (ajax.server.php) -> Called for page: {index} 
2019-04-10 19:47:32 [admin] (session) -> miudmi14q4v0gvd29vseo3g1m0 has been extended to Wed, 10 Apr 2019 20:47:32 +0100 extension length 3600 
2019-04-10 19:47:32 [admin] (ajax.server.php) -> Called for page: {index} 
2019-04-10 19:47:35 [admin] (session) -> miudmi14q4v0gvd29vseo3g1m0 has been extended to Wed, 10 Apr 2019 20:47:35 +0100 extension length 3600 
2019-04-10 19:47:35 [admin] (ajax.server.php) -> Called for page: {stream} 
2019-04-10 19:47:35 [admin] (stream.ajax.php) -> Called for action {directplay} 
2019-04-10 19:47:35 [admin] (stream.ajax.php) -> Play type {} 
2019-04-10 19:47:36 [admin] (session) -> miudmi14q4v0gvd29vseo3g1m0 has been extended to Wed, 10 Apr 2019 20:47:36 +0100 extension length 3600 
2019-04-10 19:47:36 [admin] (session) -> miudmi14q4v0gvd29vseo3g1m0 has been extended to Wed, 10 Apr 2019 20:47:36 +0100 extension length 3600 
2019-04-10 19:47:36 [admin] (stream.php) -> Asked for {play_item}. 
2019-04-10 19:47:36 [admin] (stream.php) -> Stream Type: web_player Media IDs: [{"object_type":"song","object_id":"45"}] 
2019-04-10 19:47:36 [admin] (session) -> Session created: b0a3015d38735faab013fc9d85f5cdb8 
2019-04-10 19:47:36 [admin] (media) -> Using default target format 
2019-04-10 19:47:36 [admin] (media) -> Transcode settings: from m4a to mp3 
2019-04-10 19:47:36 [admin] (media) -> Command: ffmpeg Arguments: -i %FILE% -vn -b:a %SAMPLE%K -c:a libmp3lame -f mp3 pipe:1 
2019-04-10 19:47:36 [admin] (media) -> Changing play url type from {m4a} to {mp3} due to encoding settings... 
2019-04-10 19:47:36 [admin] (stream_playlist.class.php) -> Adding url {{"properties":["url","title","author","time","info_url","image_url","album","type","codec","track_num"]}}... 
2019-04-10 19:47:36 [admin] (stream_playlist) -> Generating a {web_player} object... 
2019-04-10 19:47:36 [admin] (media) -> Using default target format 
2019-04-10 19:47:36 [admin] (media) -> Transcode settings: from m4a to mp3 
2019-04-10 19:47:36 [admin] (media) -> Command: ffmpeg Arguments: -i %FILE% -vn -b:a %SAMPLE%K -c:a libmp3lame -f mp3 pipe:1 
2019-04-10 19:47:36 [admin] (webplayer.class.php) -> Types {{"real":"mp3","player":"mp3"}} 
2019-04-10 19:47:36 [admin] (webplayer.class.php) -> Return get_media_js_param {{"title":"Episode 2","artist":"BBC Radio 4 Extra","artist_id":"9","album_id":"9","replaygain_track_gain":"0.000000","replaygain_track_peak":"0.000000","replaygain_album_gain":"0.000000","replaygain_album_peak":"0.000000","media_id":"45","media_type":"song","filetype":"mp3","url":"http:\/\/192.168.1.167\/ampache\/play\/index.php?ssid=b0a3015d38735faab013fc9d85f5cdb8&type=song&oid=45&uid=3&name=BBC%20Radio%204%20Extra%20-%20Episode%202.mp3&transcode_to=mp3","poster":"http:\/\/192.168.1.167\/ampache\/image.php?object_id=9&object_type=album&auth=b0a3015d38735faab013fc9d85f5cdb8&thumb=3&name=art.jpg"}} 
2019-04-10 19:47:36 [admin] (media) -> Using default target format 
2019-04-10 19:47:36 [admin] (media) -> Transcode settings: from m4a to mp3 
2019-04-10 19:47:36 [admin] (media) -> Command: ffmpeg Arguments: -i %FILE% -vn -b:a %SAMPLE%K -c:a libmp3lame -f mp3 pipe:1 
2019-04-10 19:47:36 [admin] (webplayer.class.php) -> Types {{"real":"mp3","player":"mp3"}} 
2019-04-10 19:47:36 [admin] (webplayer.class.php) -> Return get_media_js_param {{"title":"Episode 2","artist":"BBC Radio 4 Extra","artist_id":"9","album_id":"9","replaygain_track_gain":"0.000000","replaygain_track_peak":"0.000000","replaygain_album_gain":"0.000000","replaygain_album_peak":"0.000000","media_id":"45","media_type":"song","filetype":"mp3","url":"http:\/\/192.168.1.167\/ampache\/play\/index.php?ssid=b0a3015d38735faab013fc9d85f5cdb8&type=song&oid=45&uid=3&name=BBC%20Radio%204%20Extra%20-%20Episode%202.mp3&transcode_to=mp3","poster":"http:\/\/192.168.1.167\/ampache\/image.php?object_id=9&object_type=album&auth=b0a3015d38735faab013fc9d85f5cdb8&thumb=3&name=art.jpg"}} 
2019-04-10 19:47:36 [admin] (session) -> miudmi14q4v0gvd29vseo3g1m0 has been extended to Wed, 10 Apr 2019 20:47:36 +0100 extension length 3600 
2019-04-10 19:47:36 [admin] (media) -> Using default target format 
2019-04-10 19:47:36 [admin] (media) -> Transcode settings: from m4a to mp3 
2019-04-10 19:47:36 [admin] (media) -> Command: ffmpeg Arguments: -i %FILE% -vn -b:a %SAMPLE%K -c:a libmp3lame -f mp3 pipe:1 
2019-04-10 19:47:36 [admin] (webplayer.class.php) -> Types {{"real":"mp3","player":"mp3"}} 
2019-04-10 19:47:36 [admin] (media) -> Using default target format 
2019-04-10 19:47:36 [admin] (media) -> Transcode settings: from m4a to mp3 
2019-04-10 19:47:36 [admin] (media) -> Command: ffmpeg Arguments: -i %FILE% -vn -b:a %SAMPLE%K -c:a libmp3lame -f mp3 pipe:1 
2019-04-10 19:47:36 [admin] (webplayer.class.php) -> Types {{"real":"mp3","player":"mp3"}} 
[... cut ...]
2019-04-10 19:51:33 [admin] (session) -> miudmi14q4v0gvd29vseo3g1m0 has been extended to Wed, 10 Apr 2019 20:51:33 +0100 extension length 3600 
2019-04-10 19:51:33 [admin] (ajax.server.php) -> Called for page: {index} 
2019-04-10 19:52:01 [admin] (stream) -> Process is not running, kill skipped. 
2019-04-10 19:52:01 [admin] (play) -> Stream ended at 6826476 (6826476) bytes out of  
2019-04-10 19:52:01 [admin] (stream) -> Process is not running, kill skipped. 
2019-04-10 19:52:01 [admin] (play) -> Stream ended at 6826476 (6826476) bytes out of

MichaIng, 2019-04-11:
@spectrumcomputing 
Thanks for the link. From there it seems quite clear how to enable:
```
G_CONFIG_INJECT 'transcode_m4a[[:blank:]]' 'transcode_m4a      = allowed' /var/www/ampache/config/ampache.cfg.php
```

But there are reported as well several issues with transcoding enabled on Ampache, which is why it is disabled by default: https://github.com/ampache/ampache/issues/1048

spectrumcomputing, 2019-04-11:
You sir are a genius :-) Subsonic and Ampache now working perfectly.  I will make a donation to the project over the weekend.

MichaIng, 2019-04-16:
@spectrumcomputing 
Thanks for reporting back.

I will add this as info to our online docs. As of above mentioned issues with transcoding on Ampache, we will not implement it by default, but instead direct users to the documentation about how to do so manually. Makes sense as well as you can enable much more sound formats that way.

MichaIng, 2019-04-17:
Info added to online docs: https://dietpi.com/phpbb/viewtopic.php?p=554#p554
Will mark this as closed. Feel free to reopen, if required.

closed 2019-04-17 by MichaIng