MichaIng/DietPi #2696: Kodi on Odroid C2: double free or corruption

ajgon, 2019-04-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.22.3
- Distro version | stretch
- Kernel version | 3.16.57+
- SBC device | Odroid C2 (aarch64)
- Power supply used | 5V 2A
- SDcard used | SanDisk 8GB

#### Additional Information (if applicable)
- Software title | kodi
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Steps to reproduce
1. Install kodi
2. Either set kodi as autostrat, or try to run `startkodi`

#### Expected behaviour
- Kodi should fire up

#### Actual behaviour
- Kodi crashes with an error:
```
ERROR: Unable to create GUI. Exiting
*** Error in `/usr/local/lib/kodi/kodi.bin`: double free or corruption (out): 0x000000556131d7d0 ***
Aborted
```

MichaIng, 2019-04-08:
@ajgon 
Many thanks for your report.

Simple first try, just in case some file is corrupted: `G_AGI --reinstall kodi-odroid`

To sort out our wrapper, when you are in terminal please try: `xinit kodi-standalone` (this is what autostart and `kodistart` from terminal should do.
Please also try to start Kodi from Desktop otherwise (if you have one installed or don't mind to install one?). The command from desktop terminal should be `kodi`.

ajgon, 2019-04-08:
Hey, thank you for quick reply!

I already tried all the solutions before submitting the ticket, and some other including:

* `G_AGI --reinstall kodi-odroid`
* `xinit kodi-standalone`
* `xinit kodi`
* `xinit kodi --standalone`
* `xinit xterm` (which launched Xserver properly), and then running up `kodi` from xterm
* removing `kodi-odroid` and installing raw `kodi` from debian repos

Unfortunatelly I don't have any xwindow manager, as I planned use Odroid solely as a HTPC (plus some extra apps running in background), so I haven't tried that. I'll try to install XFCE and fire up kodi from there and let you know.

Since this is vanilla system which I'm building from scratch, I don't mind doing more tests as long as SD card survives it :-).

MichaIng, 2019-04-08:
@ajgon 
Okay if starting from `xinit xterm` fails, then this should be the same for desktops. So no strong need to test on desktop.

Can you please try reinstalling the mali driver set:
```
G_AGI --reinstall mali450-odroid xf86-video-mali-odroid libump-odroid xf86-video-fbturbo-odroid
```

MichaIng, 2019-05-26:
@ajgon 
Any news about this? I mark the issue as closed for now. Feel free to reopen if required.

closed 2019-05-26 by MichaIng