MichaIng/DietPi #2697: Radio Player not working in Subsonic or Ampache

spectrumcomputing, 2019-04-07:
#### Required Information
``` 
root@DietPi:~# #!/bin/bash
root@DietPi:~# G_DIETPI_VERSION_CORE=6
root@DietPi:~# G_DIETPI_VERSION_SUB=22
root@DietPi:~# G_DIETPI_VERSION_RC=3
root@DietPi:~# G_GITBRANCH=master
root@DietPi:~# G_GITOWNER=MichaIng
```
- Distro version | 
stretch
- Kernel version | 
Linux DietPi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B (armv7l)

- Power supply used | IQ Five Port Mains powered USB
- SDcard used | Sandisk Ultra

#### Additional Information (if applicable)
- Software title | Ampache & Subsonic
- Fresh Install
- Can this issue be replicated on a fresh installation of DietPi? Yes
- `dietpi-bugreport`  Bug report sent, reference code: 0da7208c-3094-4722-b71b-ccf5f2e53432

#### Steps to reproduce

1. ... In Subsonic select Internet Radio, the settings
2. ...In Name enter 'Heart Sussex', in URL enter 'http://media-ice.musicradio.com/HeartSussex.m3u', in site URL enter 'https://www.heart.co.uk/sussex/'
3. Click Save
4. 

#### Expected behaviour

Click on Internet Radio, then click on the image and the radio plays via 'Flowplayer'

#### Actual behaviour
No play button appears and the section of the page is blank.

#### Extra details
Not working in AirSonic either.
- ...

spectrumcomputing, 2019-04-08:
Just to add more info, I have looked at the log and this is the error I get in SubSonic:

[4/8/19 6:37:55 PM BST] | INFO | InternetRadioStreamController | Starting converter for radio: /var/subsonic/transcode/ffmpeg -i http://bbcmedia.ic.llnwd.net/stream/bbcmedia_lrcorn_mf_p?s=1554737875&e=1554752275&h=71c817f98567e8f660f613ba6b0f5c3b -f mp3 -v 0 -
-- | -- | -- | --
[4/8/19 6:37:55 PM BST] | INFO | InputStreamReaderThread | (/var/subsonic/transcode/ffmpeg) /var/subsonic/transcode/ffmpeg: 1: /var/subsonic/transcode/ffmpeg: Syntax error: "(" unexpected

These work fine on the Windows build.

spectrumcomputing, 2019-04-08:
Hi again,

Following these instructions seems to fix it.  Is it possible to get this baked into DietPi, as well as upgrading to the new version of Subsonic:

http://pmdzsite.online.fr/index.php?article17/subsonic-on-raspberry-pi-3

Many thanks

MichaIng, 2019-04-09:
@spectrumcomputing 
Thanks for reporting, again.

The same answer applies: https://github.com/MichaIng/DietPi/issues/2695#issuecomment-481050941

> (/var/subsonic/transcode/ffmpeg) /var/subsonic/transcode/ffmpeg: 1: /var/subsonic/transcode/ffmpeg: Syntax error: "(" unexpected

Strange, we place the symlink as well on install. There was an issue with Airsonic where the `transcode` did not exist which lead to symlink was placed wrong. But on Subsonic this dir is pre-created by the deb package install, so no chance this fails.

However I took the change to update our Subsonic install code a bid and leave it now run as non-root user "subsonic".

spectrumcomputing, 2019-04-09:
OK, so I ran a clean install of DietPi and installed Subsonic, AirSonic and Amapache.  Subsonic now at latest version (thank you).

Subsonic now runs m4a files and radio streams perfectly!  Airsonic runs m4a files without issue.  Ampache does not play .m4a files or radio streams. 

This may help with Ampache:

https://github.com/ampache/ampache/wiki/Transcoding

spectrumcomputing, 2019-04-11:
I found the answer for Ampache here: 

https://github.com/ampache/ampache/issues/683

You need to download the .m3u file, open it up in a text editor then paste the contents of that into the path, and select codec mp3 (has to be lower case), then it seems to work. Thank you.

MichaIng, 2019-04-17:
@spectrumcomputing 
Thanks for info. Yeah generally it seems to depend on the actual web player, how the m3u file is served, to allow steaming or not from the link. As suggested in the link the final m3u URL after all redirects have been resolved, needs to be added.

I will mark this issue as closed. Feel free to reopen, if required.

closed 2019-04-17 by MichaIng