MichaIng/DietPi #2701: Odroid https://oph.mdrjr.net/ down?

tomtomb24, 2019-04-08:
I try to install dietpi on an odroid X4 and the link/server https://oph.mdrjr.net/ is timing out.
Is the server down or deprecated?
Tom

MichaIng, 2019-04-08:
@tomtomb24 
Indeed Meverics' repo is currently down.

I am just wondering (and verified) that there is indeed one single APT source still the original mirror (`/etc/apt/sources.list.d/meveric-all.list` => `deb http://oph.mdrjr.net/meveric all main testing`). Actually that should have been replaced by our own mirror (`http://fuzon.co.uk/meveric`). Although usually it should not play a role.

Somehow this survived the related step in the image preparation script: https://github.com/MichaIng/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L584

Ah now I see why: This one file contains the repo as `http://` while the others by default (fresh image from Meveric) contain them as `https://`.

When the error prompts on timeout, please exit to terminal. There run:
```
sed -i 's@http://oph.mdrjr.net/meveric@http://fuzon.co.uk/meveric@' /etc/apt/sources.list.d/meveric-all.list
```

Then `exit` to logout and login again to restart first run setup.

tomtomb24, 2019-04-09:
HI,
thanks for checking. I have updated the file meveric-all.list to the fuzon.co.uk link and it worked like a charm. 

Thanks

Tom

Fourdee, 2019-04-09:
@MichaIng 

Updated workaround, was missing the httpS. Tested fine on Odroid N2

MichaIng, 2019-04-09:
@Fourdee 
Perfect. I am still wondering what is going on with Meverics repo. It's still down. Perhaps we should report on Odroid forums. As well report the missing "S" since this was for sure not intended, although really no harm.

MichaIng, 2019-04-15:
https://oph.mdrjr.net/ is up again. For future Odroid images we fixed the possibly left sources entry to be switched to our mirror regardless. I mark this as closed.

closed 2019-04-15 by MichaIng

djmaze, 2022-05-29:
It's down again and the fuzon.co.uk seems to be gone as well. Any alternatives?

MichaIng, 2022-05-29:
Which DietPi version are you using? It should have been updated to `https://dietpi.com/meveric/` a long time ago.

djmaze, 2022-05-29:
@MichaIng Nevermind, I just did an DietPi update and it switched to the new repo automatically. Sorry for the noise.

MichaIng, 2022-05-29:
Okay great 🙂.