MichaIng/DietPi #2703: Keyboard without function to login

Trigger58, 2019-04-08:
Hello,
i tried the following 2 Distros with my RaspberryPi 3b+:

- DietPi_RPi_ARMv6-Stretch_Amiberry
- DietPi_RPi_ARMv6-Stretch

The 1. Distro is working fine and Amiberry runs great but every time i quit amiberry or i try to use another boot option without autologin into console, then i can't use the keyboard. 

I get the screen with please enter login to continue but i can't enter anything.

Only if i change in dietpi.txt the boot option **7=Console+auto root login** then the keyboard works. 

Same problem with the 2. distro. 

What i have to do to change all boot options to autologin?

At the moment my only possible way to change a file is with ftp-tool like FileZilla. I copy the dietpi.txt to pc, change it and overwrite the file on pi.

The implemented way that is described at your homepage with root:dietpi@192.168.0.100 don't work too. I tried different IP's ,including the IP that DietPi got from my fritzbox, but no way to get into it. Only FileZilla gets access but with FileZilla i can only transfer files but i can't get access to dietpi-config or another menu with it.

Trigger58, 2019-04-08:
Sorry, i forgot to say that i have the problem with the latest Version from DietPi-Homepage i downloaded yesterday.

MichaIng, 2019-04-09:
@Trigger58 
Many thanks for reporting.

Okay if you tried `DietPi_RPi_ARMv6-Stretch_Amiberry` + `DietPi_RPi_ARMv6-Stretch` then it has not to do with latest changes since the first is much older then the second (which I updated a few days ago).

You did hit `return` one time right? Our login banner overprints the initial `login: ` input field, to you need to hit return one time to have it appearing.
Aside from that very strange since the keyboard itself otherwise works fine. Did you try different USB ports or a different keyboard?

Trigger58, 2019-04-09:
Good Morning,
i tried it with hitting one time return but nothing happens. I can press every key but i get no response.
I tried 2 keyboards with usb cable. a cheap and an expensive one. both with the same result. 
Both are working only if i use the the autologin at startup. 

it seems that in the other boot options the keyboard driver was not loaded or something else.
But there is a strange thing. during loading the bootoption with amiberry i can hit ESC key and then the screen change the color. this means that the pi get a response from keyboard but after starting in amiberry and the i quit it, the login screen appears and the keyboard don't work.

Trigger58, 2019-04-09:
UPDATE:
i flashed now different microsd-cards and usb-sticks. i got the same result at all cards and sticks. no keyboard function during login screen.
i tried all usb-ports at pi too. no change.

Trigger58, 2019-04-09:
UPDATE2:

I got it to work. Finally.

There the steps i did:

1. Flashing Image to microsd-card
2. On PC i changed the bootoptions in dietpi.txt from 6 to 7
3. Started card with pi and let him do all Updates
4. After Updates i could use keyboard because i chosen autostart option 7
5. Entered dietpi-config and changed autostart  option to normal boot UAE4ARM NOT fastboot!
6. Saved all and reboot 
7. After quitting amiberry i get me logged in by Auto and can use my keyboard.

MichaIng, 2019-04-11:
@Trigger58 
I think now I understand.

You only face the non-functional keyboard after exiting AmiBerry from fastboot mode, right? So on first boot login works but when you install AmiBerry, enable fastboot, then booting into AmiBerry and exiting to console, keyboard does not work?
- The RPi AmiBerry image boots into fastboot mode automatically **after** first run setup (initial dietpi-software prompt) has finished which is why you need to actively adjust `dietpi.txt`.

I guess it has to do with the following:
- AmiBerry fastboot boots into AmiBerry UI on early boot stage, using TTY1. For this reason on RPi e.g. we move the kernel boot messages from TTY1 to TTY3 (when enabling fastboot) so they don't mess with the AmiBerry UI and fastboot output.
- The login console `getty@tty1.service` however is enabled on TTY1 as well. I guess either this service fails when AmiBerry is active (and bound to TTY1) or AmiBerry fastboot suppresses it actively from starting.
- Without `getty@tty1.service` being active, no login is possible, since no agetty is listening to STDIN, that for sure.
- This is not an issue when using any other boot mode than 6 (fastboot) since in all those cases `getty@tty1.service` is reached, either configured to require login or doing this automatically. So e.g. AmiBerry standard boot is starting as child process of `getty@tty1.service` (`agetty` is the actual process binary). So when exiting AmiBerry you land in the login console that was already active as parent.

AmiBerry fastboot is intended to interfere at very early boot stage and skip all the agetty stuff to allow starting that fast. Also preventing `getty@tty1.service` makes sense, so there are no two separate processes listening to the same TTY for STDIN. But what indeed would be great is:
- When exiting AmiBerry from fastboot, `getty@tty1.service` should be started so users are presented with the default login prompt.

@midwan @Fourdee 
Perhaps you have an idea how to achieve that?

midwan, 2019-04-12:
@MichaIng 
I've noticed this problem also, but only on DietPi so far.
I haven't had time to look for a solution however... :-/

MichaIng, 2019-04-12:
@midwan 
Possibly `systemd-logind` does the getty spawn on other systemd.

@Trigger58 
You can access the system via SSH right, so e.g. using PuTTY from remote Windows system? Otherwise skip the following, as we then need to get SSH running first.

Could you please try to enable systemd-logind:
```
systemctl unmask systemd-logind
systemctl --now enable systemd-logind
```

Then boot into fastmode (autostart option 6) and test if exiting then brings you into a login prompt. In case it doesn't (getty on TTY1 does not spawn automatically), then you can switch to TTY2 by pressing F2 button. systemd-logind should then automatically spawn a getty (login prompt) there. From there you can revert autostart option.

As said first assure that SSH works, just in case.

midwan, 2019-04-12:
@MichaIng 
In my tests, SSH does work normally. And so do other TTYs, if you switch to them (e.g. with Ctrl-Alt-F2) you can login normally.

Trigger58, 2019-04-12:
Hi,
I can Test it. SSH is working all time but i get only access through a PC Tools called FileZilla. The Tools in Dietpi to get access from a PC dont work. I cant get access through them.
I hole that the procedure you wrote works. Otherwise i cant get access to my System anymore.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/MichaIng/DietPi/issues/2703#issuecomment-482578190",
"url": "https://github.com/MichaIng/DietPi/issues/2703#issuecomment-482578190",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Trigger58, 2019-04-12:
I installed PuTTY und conntected to Pi.
Then i used your commands but i dont know if the second command was successful.
[image]

Trigger58, 2019-04-12:
So i changed autoboot to 6 and made a reboot. Amiberry starts fast. After quitting Amiberry it happens nothing. The bootprocess was killed during start from amiberry so ineeded the CTRL-ALT-F2 combo to get access.
TTY1 dont work.

Trigger58, 2019-04-12:
Ok, i have to correct my last answer. Now it seems to work.
To start in fastboot and then quit amiberry direct after loading is not helpful. Now i waited 2 Minutes and then i quit amiberry and hey, now i got a login screen :)

MichaIng, 2019-04-12:
@Trigger58 
Thanks for testing. Jep the second command above was expected to fail since systemd-logind is (usually) a static service. So it is always enabled, if not masked. I just added this as kind of failsafe.

I think AmiBerry does not really kill the boot process, but it goes on in background, printing to TTY3 (you should see boot process output when pressing F3, if still done). When exiting too fast, I guess systemd-logind is not yet at a stage to spawn on TTY1 automatically. When you hit F2 then I guess boot was already far enough to allow getty spawn by logind. And if you wait a bid longer than it is already when exiting AmiBerry.

However long story short:
- AmiBerry prevents getty@tty1.service because it runs on TTY1 itself.
- So when exiting AmiBerry the getty needs to be started outside of systemd boot order.
- This is done most easily by having logind active, which does this automatically when screen is attached to a free TTY.
- So we will unmask logind when installing AmiBerry.

As well on our pre-installed AmiBerry image(s) we need to have logind unmasked. I can update our RPi AmiBerry image this weekend, only need to know about the install procedure outside of DietPi-Software.
@Fourdee 
How do/did you create this image? Simply create (or use as basis) the regular RPi image and place AmiBerry code + service manually (+ symlink to enable) + autostart option to 6 in `dietpi.txt`?

MichaIng, 2019-04-17:
Okay fix merged for v6.23: https://github.com/MichaIng/DietPi/pull/2718

closed 2019-05-12 by MichaIng