MichaIng/DietPi #2704: Proxy server settings 'stick' even though proxy server is off

stephantual, 2019-04-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.22.3
- Distro version | not relevant (preimage raspian lite)
- Kernel version | not relevant
- SBC device | RPI3b+
- Power supply used -5V 2.4A RAVpower 60w
- SDcard used | sandisk 128gb ultra

#### Additional Information (if applicable)
- Software title | diet pi config screen
- Can this issue be replicated on a fresh installation of DietPi? YES

#### Steps to reproduce
- fresh burn of the latest dietpi
- load sd into pi
- start pi
- go to networks settings and set proxy server to 'yes'
- attempt network access, default of 'myproxyserver.com' is resolved then everything fail as expected
- go to networks settings and set proxy server to 'no' (save, reboot)

#### Expected behaviour
- the proxy server url should be ignored as the network proxy is now OFF

#### Actual behaviour
- the proxy server url is taken into account (in other words, it's as if the proxy server setting was still toggled ON'

#### Extra details
very annoying as it forces user to reflash (unless i'm missing something, it's not possible to mess around with the pre-image config files from the pi itself).

MichaIng, 2019-04-09:
@stephantual 
Many thanks for your report.

Will try to replicate.

Fourdee, 2019-04-11:
Confirmed bug, looking into it now.

Fourdee, 2019-04-11:
Notes:
- If user disables proxy, reboot required as can not unset outside of bash session?
- Reboot should not be required for enabling/updating creds, as we can export.

MichaIng, 2019-04-15:
I see, the `export` pattern was missing in the `sed` command. It was my error when adding `^[[:blank:]]*` to remove the lines more targeted: https://github.com/MichaIng/DietPi/commit/4b80658bc7d3b0bd02634eb533081d9dfc5a0bb6#diff-0e127c4cad3f8401a925d13d05b5ae87R5052
Sorry for that...

@Fourdee 
Is there any special reason why you reverted from `[On]`/`[Off]` to `[Enabled]`/`[Disabled]`?
Also I vote for adding proxy settings via `/etc/bashrc.d/dietpi-proxy.sh` drop-in to skip touching `/etc/bash.bashrc` and allow easier removal/adding of these.
And estimating current settings should be done by checking `$http_proxy` variable first, if empty `/etc/bashrc.d/dietpi-proxy.sh`. `dietpi.txt` settings should me converted to AUTO_SETUP_* since they themselves have no effect but are only applied on first boot.

closed 2019-05-12 by MichaIng