MichaIng/DietPi #2708: Failure to Install Emby Server

oleshkev, 2019-04-10:
#### Details:
- Date           | Wed 10 Apr 15:35:58 BST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | wget  -O 41.
- Exit code      | 1
- Software title | DietPi-Software

#### Additional Information (if applicable)
- the software title installed freshly
- this issue can be replicated on a fresh installation of DietPi

#### Steps to reproduce:
1. Run dietpi-software
2. Go to dietpi-software> Software Optimized > select Emby
3. Trigger Installation of software

#### Expected behaviour:
Emby installed

#### Actual behaviour:
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Emby Server: web interface media streaming server

[  OK  ] DietPi-Software | Connection test: https://api.github.com/repos/MediaBrowser/Emby.Releases/releases/latest
[FAILED] DietPi-Software | wget  -O 41.

#### Extra details:
It looks like the command "wget  -O 41" is missing some parameters.
I have tried solution for the similar issues, but it does not work:
wget https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-software -O /DietPi/dietpi/dietpi-software
--2019-04-10 15:46:08--  https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-software
Resolving github.com (github.com)... 140.82.118.4, 140.82.118.3
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://github.com/MichaIng/DietPi/raw/testing/dietpi/dietpi-software [following]
--2019-04-10 15:46:09--  https://github.com/MichaIng/DietPi/raw/testing/dietpi/dietpi-software
Reusing existing connection to github.com:443.
HTTP request sent, awaiting response... 404 Not Found
2019-04-10 15:46:09 ERROR 404: Not Found.

MichaIng, 2019-04-11:
@oleshkev 
Many thanks for your report.

Please see here for solution: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5782

oleshkev, 2019-04-12:
@MichaIng, thanks for your prompt reply.

Honestly, the solution you have proposed didn't work. As this line of code from the solution:
`sed -i 's#"$(curl -s "$INSTALL_URL_ADDRESS" | grep -m1 "browser_download_url.*emby-server-deb_.*_$arch\\.deb" | cut -d \\" -f 4)"#"https://github.com/MediaBrowser/Emby.Releases/releases/download/4.0.2.0/emby-server-deb_4.0.2.0_$arch.deb"#' /DietPi/dietpi/dietpi-software`
 just cleaned the file DietPi-software (so it became 0 byte in size).

So I downloaded the updated file with this command. 
`wget https://raw.githubusercontent.com/MichaIng/DietPi/ce890930d8239433f307b0bbe6219a84a718aedd/dietpi/dietpi-software -O /DietPi/dietpi/dietpi-software`

And then installation of Emby succeeded.

Thank you one more time.

closed 2019-04-12 by oleshkev

MichaIng, 2019-04-13:
@oleshkev 
?? Very strange, must have been a copy/paste error. Just tested again and works as expected:
```
2019-04-13 21:17:50 root@micha:/tmp# cp /DietPi/dietpi/dietpi-software .
2019-04-13 21:18:03 root@micha:/tmp# l
-rwxr-xr-x 1 root root 493K Apr 13 21:18 dietpi-software
2019-04-13 21:18:03 root@micha:/tmp# sed -i 's#"$(curl -s "$INSTALL_URL_ADDRESS" | grep -m1 "browser_download_url.*emby-server-deb_.*_$arch\\.deb" | cut -d \\" -f 4)"#"https://github.com/MediaBrowser/Emby.Releases/releases/download/4.0.2.0/emby-server-deb_4.0.2.0_$arch.deb"#' dietpi-software
2019-04-13 21:18:26 root@micha:/tmp# l
-rwxr-xr-x 1 root root 493K Apr 13 21:18 dietpi-software
2019-04-13 21:18:26 root@micha:/tmp# grep "https://github.com/MediaBrowser/Emby.Releases" dietpi-software
                        no_check_url=1 Download_Install "https://github.com/MediaBrowser/Emby.Releases/releases/download/4.0.2.0/emby-server-deb_4.0.2.0_$arch.deb"
```
- Within single no variables are expanded, so the search/replace pattern is very specific and sed works line by line.

However good that you found a solution. Note that downloading whole scripts from different branches can lead to unexpected issues since some changes might depend on changes in other scripts or even system changes as well. At best always pull from master branch and apply changes manually at the desired lines.