MichaIng/DietPi #2711: Installing Sonarr fails | ASUS Tinker Board S

holmesdanni, 2019-04-14:
#### Details:
- Date           | Sun 14 Apr 10:29:21 AEST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | ARMbian
- SBC device     | Asus Tinker Board (armv7l) (index=52)
- Kernel version | #5.77 SMP PREEMPT Wed Apr 3 17:06:29 CEST 2019
- Distro         | stretch (index=4)
- Command        | G_AGI sqlite3
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Select to install Sonarr via dietpi-software, then begin install.

#### Expected behaviour:
- Sonarr should install successfully

#### Actual behaviour:

- The install begins to run, but it then presents the follow message:

>DietPi-Software: G_AGI sqlite3                                                           
  Exit code: 100                                                                        
  DietPi version: v6.22.3 (Fourdee/master) | HW_MODEL:52 | HW_ARCH:2 | DISTRO:4         
  Image creator: DietPi Core Team                                                       
  Pre-image: ARMbian                                                                    
  Log file contents:                                                                       
  E: Unable to correct problems, you have held broken packages.                            
  Unable to continue, DietPi-Software will now terminate.

#### Extra details:

- The same thing happens when I try to install Radarr.

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.

MichaIng, 2019-04-15:
@holmesdanni 
Many thanks for your report.

I installed Sonarr several times on testing systems the last days, so the install code should be generally fine. In your case it's an APT issue.

Please paste the full output of: `apt install sqlite3`

Generally to have APT fix issues automatically: `apt -f install`
- **BUT: Before confirming with `y`, check out what it wants to do, especially that it does not remove a bunch of packages that you actually require.**

With the above output we should be able to identify the faulty dependency and correct targeted.

holmesdanni, 2019-04-15:
Thanks @MichaIng. After pasting `apt install sqlite3` I get the following output.

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 sqlite3 : Depends: libsqlite3-0 (= 3.16.2-5+deb9u1) but 3.27.2-2~bpo9+1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2019-04-16:
@holmesdanni
Hmm for some reason libsqlite3-0 is pulled from backports whole sqlite3 is not. Could you please send a `dietpi-bugreport` and paste the ID here? So I can check which packages were/are pulled from backports and why.

holmesdanni, 2019-04-17:
I just ran `dietpi-bugreport` and got the following, so will try again tomorrow.

```
[  OK  ] DietPi-Bugreport | Root access verified.
[  OK  ] DietPi-Bugreport | RootFS R/W access verified.

[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[  OK  ] DietPi-Bugreport | Connection test: ssh.dietpi.com
[FAILED] DietPi-Bugreport | Failed to connect to SFTP server. Please try again later or report this to DietPi forum or GitHub repo in the first place.
```

MichaIng, 2019-04-17:
@holmesdanni 
Strange, works well here (now).

Meanwhile could you please paste:
```
apt-cache policy
dpkg -l | grep bpo
```

MichaIng, 2019-05-26:
I mark this as closed. Feel free to reopen if the issue persists.

closed 2019-05-26 by MichaIng