MichaIng/DietPi #2714: Emby installation failure

graphwire, 2019-04-14:
#### Details:
- Date           | Sun 14 Apr 21:27:51 BST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | alkis
- Pre-image      | https://yadi.sk/d/pHxaRAs-tZiei/5.78_20190412/S905/Armbian_5.78_Aml-s905_Debian_stretch_default_5.1.0-rc1_20190412.img.xz
- SBC device     | Generic Device (aarch64) (index=22)
- Kernel version | #5.78 SMP PREEMPT Fri Apr 12 14:28:34 MSK 2019
- Distro         | stretch (index=4)
- Command        | wget  -O 41.
- Exit code      | 1
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:
Stops midways, a retry or exit prompt pops-up

#### Additional logs:
```
Log file contents:
wget: missing URL
Usage: wget [OPTION]... [URL]...

Try `wget --help' for more options.
```

MichaIng, 2019-04-15:
@graphwire 
Many thanks for your report.

Jep known issue with the last release of Emby Server. To resolve please do the following:
```
sed -i 's#"$(curl -s "$INSTALL_URL_ADDRESS" | grep -m1 "browser_download_url.*emby-server-deb_.*_$arch\\.deb" | cut -d \\" -f 4)"#"https://github.com/MediaBrowser/Emby.Releases/releases/download/4.0.2.0/emby-server-deb_4.0.2.0_$arch.deb"#' /DietPi/dietpi/dietpi-software
/DietPi/dietpi/software install 41 # to rerun the Emby install
```

graphwire, 2019-04-17:
It worked! Thank you!

closed 2019-04-24 by graphwire