MichaIng/DietPi #2715: DietPi.com/survey | Benchmarks reset for v6.23 released

Fourdee, 2019-04-15:
Reason:
- Incorrect RAM benchmarks due to low MB size test against sub second time result. Gives poor total time calulation.
- CPU temps on Pine A64 incorrect.

All we need to do is change ```dietpi-survey_report``` to only calculation benchmarks with version ```2```.

---

@MichaIng 

We could do fixes on ```dietpi-survey_report``` to check for invalid CPU temps (+cpu time), but the RAM results need resetting.

MichaIng, 2019-04-15:
@Fourdee 
Jep makes sense. Perhaps we can do the version check only when reading RAM values but keep reading the other ones.

Fourdee, 2019-04-19:
@MichaIng 

Made a start:
https://dietpi.com/survey/debug.html

MichaIng, 2019-04-20:
@Fourdee 
Added some tiny changes and committed to GitHub: https://github.com/MichaIng/DietPi/commit/458d8a0bfb01c9ed4e5a6da823d566d20c079b2a

Jump directly to desired table:
- Device: https://dietpi.com/survey/#device
- Distro: https://dietpi.com/survey/#distro
- Benchmark RAM: https://dietpi.com/survey/#bench_ram
- Derived from #benchmark you added (https://dietpi.com/survey/#benchmark). Love this `id="..."` 😃.

Fourdee, 2019-04-20:
@MichaIng 

Also with this update (https://github.com/MichaIng/DietPi/commit/b6044942e0bb6e800f776292ac230434fda5c68e), we seperatlely check for valid entries in each CPU/RAM/RootFS benchmark. So if user only has valid RAM benchmark, the others will be ignored.

Previously we expected CPU/RAM/RootFS to be done together. Changes allow for more control and safety net.

Also yep, colors to ID table easily.

MichaIng, 2019-04-23:
@Fourdee 
That is great. Long term we can also offer individual benchmarks, e.g. if user doesn't want to wear SDcard. I remember you wanted to keep it easy that time, all benchs + upload with a single. Perhaps keep that and have a sub menu for individual benchs, e.g. combined with per-drive bench as well.

I will fill the RAM bench table with the devices I have later. I think we can mark this as closed then?

Fourdee, 2019-04-24:
@MichaIng 

> Perhaps keep that and have a sub menu for individual benchs, e.g. combined with per-drive bench as well.

Yep, something we can do in the future. No rush and the benchmark suite is a nice touch to ensure user can quickly benchmark system and compare.

> I will fill the RAM bench table with the devices I have later. I think we can mark this as closed then?

:+1: 

I also added the customFS to survey. So yep, all done for now :+1:

closed 2019-04-24 by Fourdee