MichaIng/DietPi #2719: I need help installin gEmby on my Raspberry PI 3 B+

Havenlee13, 2019-04-18:
I can install software like SAMBA, but I can not install Emby or Plex. I am using the OS DietPI like requested. below is the error that I have gotten. I wasn't able to copy and paste. sorry for the inconvenience. Any help would be appreciated. 
[image]

Havenlee13, 2019-04-18:
I didn't think it was actually going to show the picture. I though it was just going to put a URL there instead.

MichaIng, 2019-04-18:
@Havenlee13 
Many thanks for your report.

Please see for solution: https://github.com/MichaIng/DietPi/issues/2714#issuecomment-483254053

closed 2019-05-12 by MichaIng