MichaIng/DietPi #2721: Attempted update failed

borpin, 2019-04-18:
#### Details:
- Date           | Thu 18 Apr 11:16:46 BST 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Virtual Machine (x86_64) (index=20)
- Kernel version | #1 SMP Debian 4.9.144-3 (2019-02-02)
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Ign:1 http://repo.mongodb.org/apt/debian jessie/mongodb-org/3.4 InRelease
Ign:2 http://ftp.debian.org/debian jessie-backports InRelease
Err:3 http://ftp.debian.org/debian jessie-backports Release
  404  Not Found
Hit:4 http://repo.mongodb.org/apt/debian jessie/mongodb-org/3.4 Release
Ign:6 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:7 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:8 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:9 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:10 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists...
E: The repository 'http://ftp.debian.org/debian jessie-backports Release' does no longer have a Release file.
```

MichaIng, 2019-04-18:
@borpin
Many thanks for your report.

Where does the Jessie backports repo come from? This is obsolete and even conflicting with Stretch AND was dropped by Debian some days ago: https://github.com/MichaIng/DietPi/issues/2665

Please check your drop-in APT sources (/etc/apt/sources.list.d) and remove the file/line that contains jessie-backports.

borpin, 2019-04-18:
Ok thanks.  It is coming from this [script](https://community.ubnt.com/t5/UniFi-Wireless/UniFi-Installation-Scripts-UniFi-Easy-Update-Scripts-Ubuntu-18/td-p/2375150) to install a UniFi controller so I am checking with the author.

borpin, 2019-04-18:
Ok thanks. I'm told it can be removed so no worries. Many thanks.

closed 2019-04-18 by borpin