MichaIng/DietPi #2724: cant install software on dietpi

aspinks, 2019-04-19:
i know this is old but i ran this:
wget https://raw.githubusercontent.com/Fourdee/DietPi/0f18aa4dc0af8ab910a0173dce8849d5b53c30b0/dietpi/func/dietpi-globals -O /DietPi/dietpi/func/dietpi-globals
now when i try to install it goes to the drive manager. i exit that and get this:
OK ] DietPi-Software | Installing Sonarr: automatically download TV shows
[ OK ] DietPi-Software | Free space check: path=/ | available=24428 MB | required=500 MB

then it goes to the command prompt.

MichaIng, 2019-04-19:
@aspinks 
Uiii, this fix (Ref: https://github.com/MichaIng/DietPi/issues/1923) is totally outdated. This was for v6.11, now we are on v6.22 and downloading a v6.11(comma 5) script (especially the globals) for sure breaks everything else (DietPi scripts) 😉.

Please reapply the current master version:
```
wget https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/func/dietpi-globals -O /DietPi/dietpi/func/dietpi-globals
. /DietPi/dietpi/func/dietpi-globals
```

Then please retry installing Sonarr and paste the output of any error (or the whole dietpi-software output) here, so we can investigate.

aspinks, 2019-04-19:
thank you!, so that fixed that part but now i am back to my original problem.  I tried to install filebot via command line and now i get this error:

#### Details:
- Date           | Fri 19 Apr 17:35:10 CDT 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Native PC (x86_64) (index=21)
- Kernel version | #1 SMP Debian 4.9.65-3+deb9u2 (2018-01-04)
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...running any software install
2. ...

#### Expected behaviour:

- ...installing software

#### Actual behaviour:

- ...

#### Extra details:

- ...this happened once when i tried to install sonarr via dietpi-software.  if fixed that by doing a clean install

#### Additional logs:
```
Log file contents:
Ign:1 https://get.filebot.net/deb universal-jdk8 InRelease
Get:2 https://get.filebot.net/deb universal-jdk8 Release [1,165 B]
Get:3 https://get.filebot.net/deb universal-jdk8 Release.gpg [587 B]
Get:4 https://get.filebot.net/deb universal-jdk8/main all Packages [547 B]
Ign:5 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:6 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:7 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:8 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:9 https://cdn-aws.deb.debian.org/debian stretch Release
Fetched 2,299 B in 1s (1,413 B/s)
Reading package lists...
E: Failed to fetch https://get.filebot.net/deb/dists/universal-jdk8/InRelease  Unable to find expected entry 'main/binary-amd64/Packages' in Release file (Wrong sources.list entry or malformed file)
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

MichaIng, 2019-04-19:
@aspinks 
Ah I see, the filebot APT repo fails like this. Reason is it does not contain an explicit `binary-amd64` dir: https://get.filebot.net/deb/dists/universal-jdk8/main/
Not sure why APT throws an error if instead `binary-all` is present (like here), but it does. I found the info that `universal`/`universal-jdk8` are actually made for Debain Buster and Ubuntu 18: https://www.filebot.net/forums/viewtopic.php?f=11&t=6028
Perhaps the APT version on Debian Buster (and Ubuntu 18) allow `binary-all`-only, but on Stretch it does not.

Please adjust the sources list entry like this: `deb https://get.filebot.net/deb/ stable main`
This branch contains the required `binary-amd64` dir: https://get.filebot.net/deb/dists/stable/main/

aspinks, 2019-04-19:
first off thank you for the fast and great responses.  I imagine i need github repository to do this.  in my own efforts to fix this before i found your help i uninstalled github repository and cannot reinstall it.  i get the same errors as listed above.  how do i clear those out so i can get the repository and try installing filebot?

MichaIng, 2019-04-22:
@aspinks 
Sorry for the late reply. Actually I already answered from module, but it seems to not have been posted... connection issue on the road...

> I imagine i need github repository to do this.
...
this is what i get when i try to install github or anything at all:

What do you mean by "github" in this relation? Did you run some installer script from a GitHub repo?

However, you can leave things as they are, just as said, `universal-jdk8` is no valid branch for Debian Stretch. Please do:
```
sed -i 's/universal-jdk8/stable/' /etc/apt/sources.list.d/*
G_AGUP
```

aspinks, 2019-04-26:
sorry for the delay, been out of town for work.

when i run sed -i 's/universal-jdk8/stable/' /etc/apt/sources.list.d/*
G_AGUP or any built in installer,

i get this:
─┤ DietPi Error Handler: ├──────────────────────────────
GUP
Exit code: 100
DietPi version: v6.22.3 (Fourdee/master) | HW_MODEL:21 | HW_ARCH:10 | DISTRO:4
Image creator: n/a
Pre-image: n/a

 file contents:
:1 https://get.filebot.net/deb stable InRelease
:2 https://get.filebot.net/deb stable Release [1,175 B]
:3 https://get.filebot.net/deb stable Release.gpg [587 B]
:4 https://cdn-aws.deb.debian.org/debian stretch InRelease
:5 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease [91.0 kB]
:6 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease [94.

:7 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease [91.8 kB]

 Retry         : Re-run the last command that failed
            DietPi-Config : Edit network, APT/NTP mirror settings etc

MichaIng, 2019-04-26:
@aspinks 
> expected entry 'main/binary-i386/Packages'

Strange since your system is x86_64. We do add i386 arch to APT on x86 images but that only "allows" it but not "requires" i386 to be present. The required `main/binary-amd64/Packages` is there: https://get.filebot.net/deb/dists/stable/Release

However please try to remove the arch:
```
dpkg --remove-architecture i386
apt update
```

aspinks, 2019-04-26:
i think that may have worked, fingers crossed...

MichaIng, 2019-04-27:
@aspinks 
If `apt update` did not show any error now, it did. I will mark this issue as closed. However feel free to reopen if the issue persists.

closed 2019-04-27 by MichaIng

aspinks, 2019-04-29:
sorry to be a bother but i tried running 

deb https://get.filebot.net/deb/ stable main

i get 

-bash: deb: command not found

MichaIng, 2019-04-29:
@aspinks 
`deb https://get.filebot.net/deb/ stable main` is not a command but the desired content of the related file in the directory `/etc/apt/sources.list.d/`. But according to your output above (https://github.com/MichaIng/DietPi/issues/2724#issuecomment-487217962) this has already been applied by the `sed ...` command I posted before.

Does `apt update` still show some error like this (?):
```
E: Failed to fetch https://get.filebot.net/deb/dists/stable/InRelease  Unable to find expected entry 'main/binary-i386/Packages' in Release file (Wrong sources.list entry or malformed file)
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

aspinks, 2019-04-29:
so when i run "apt update" i get no errors but this is what shows everytime.  
[image]

MichaIng, 2019-04-30:
@aspinks 
Jep as expected.

There is one updated package available. You can update it via: `G_AGUG`

aspinks, 2019-05-01:
I ran G_AGUG and it upgraded but I am still getting the same read out as above.

MichaIng, 2019-05-01:
@aspinks 
As said, this is the usual readout telling you about APT lists being updated. Everything is fine.

aspinks, 2019-05-01:
My apologies,  I misunderstood.  I am still learning linux.  So does this mean filebot is installed?  Is it possible to run it on the gui? I am using mate as the desktop.  Again thank you for all the help

MichaIng, 2019-05-01:
@aspinks 
Ah sorry no since the APT list update failed, of course not 🙈.

Do:
```
apt install filebot
```

aspinks, 2019-05-01:
so i ran that and thought i had it but got the following
[image]

MichaIng, 2019-05-01:
@aspinks 
Ah yeah the still remaining APT issue with Java + ca-certificates-java. Should be resolved with:
`apt -f install`