MichaIng/DietPi #2725: DietPi-Software | Budgie Desktop

camdenorrb, 2019-04-20:
#### Formal software information
- Software title | Budgie Desktop
- Short description | A really nice looking desktop environment

- GitHub URL | https://github.com/solus-project/budgie-desktop
- (Official) install documentation | https://linuxconfig.org/how-to-install-budgie-desktop-on-debian

---

#### Are there similar/alternative software titles available with DietPi-Software?
LXQT
LXDE
Mate

---

#### What makes your requested software better than the above solutions, if available?
Budgie has a growing community behind it and is one of the newer desktop environment coming to the Linux ecosystem. This desktop environment is more about the look and feel rather than extreme performance.

---

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

DietPi could help users avoid the struggle of compiling fbturbo and figuring out the correct PPA to use in order for it to run optimally. Also by giving the user the basic needs for a DE, such as LightDM

---

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

[Install xserver-xorg-video-fbturbo](https://github.com/ssvb/xf86-video-fbturbo/wiki/Installation) and set it in xorg config

Install a PPA giving the latest Budgie DE, possibly from [this PPA](https://launchpad.net/~ubuntubudgie/+archive/ubuntu/backports)

Install LightDM, Xorg-server, basic needs of a DE

---

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

I am willing to do whatever I need to in order to make this happen.

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/+67

MichaIng, 2019-04-21:
@camdenorrb 
Many thanks for your request.

The Ubuntu PPA of course will not work, but the desktop is available in official Debian repo: https://packages.debian.org/stretch/budgie-desktop
So:
- Install X.org
- `apt install budgie-desktop`
That's it 😃.

camdenorrb, 2019-04-21:
That package of budgie is kinda old, I’ve seen tutorials on installing Ubuntu PPA’s on debian but never tried it. Maybe it’s possible but not recommended?

MichaIng, 2019-04-21:
@camdenorrb 
Yeah I mean it's possible to add the sources.list entry manually + the key. But you need to specify an Ubuntu version. At least some of the packages have candidates for different Ubuntu versions, so they most likely depend on some specific library versions. So it's them basically try and error which version fits for Debian Stretch.

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not be addressed soon. The FeatHub request stays however open to allow votes and if anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng