MichaIng/DietPi #2731: Failed to start dhcpcd

JJGO, 2019-04-23:
### Failed to start dhcpcd on Startup

#### Required Information
- DietPi version | `6.22.3 master`
- Distro version | `stretch`
- Kernel version | `Linux Iroh 4.14.98-v7`
- SBC device | `RPi 3 Model B+ (armv7l)`
- Power supply used | `Official RPi Power Supply`
- SDcard used | SanDisk ultra 128GB

#### Description
1. Fresh install of latest Dietpi
2. Only thing I manually configured was WPA-EAP using `/etc/wpa_supplicant/wpa_supplicant.conf`
3. On startup I get a fail message saying  `failed to start dhcpcd on all interfaces` and to run `systemctl status dhcpd.service`
4. When I run `systemctl status dhcpd.service` it outputs `Unit dhcpd.service could not be found.`
5. Software seems to be installed, i.e. `sudo apt list --installed | grep dhcpcd` outputs `dhcpcd5/stable,now 1:6.11.5-1+rpt7 armhf [installed]`

MichaIng, 2019-04-24:
@JJGO 
Many thanks for your report.

Hmm, actually `dhcpcd` should not be installed and is not required. We use `isc-dhcp-client` which is also the reason the former fails.

Simply do `G_AGP dhcpcd5` to solve.

I check the RPi image, actually we actively purge this on image creation: https://github.com/MichaIng/DietPi/blob/dev/PREP_SYSTEM_FOR_DIETPI.sh#L883

MichaIng, 2019-05-26:
I mark this as closed. Feel free to reopen if required.

closed 2019-05-26 by MichaIng

Sopor, 2019-06-05:
I got this error message after a clean install with updates `failed to start WPA supplicant.`. What should i do to get rid of it?

MichaIng, 2019-06-06:
@Sopor 
Many thanks for your report. Jep I remember this was indeed one (though visual only) issue with the current life RPi image. One of the reasons I created the new one (check out the other topic you opened).

Simply do: `systemctl disable wpa_supplicant`
This service is started as instances for the particular network interface, so the global service is not required and conflicts with the prior one.

Sopor, 2019-06-07:
```
Removed /etc/systemd/system/dbus-fi.w1.wpa_supplicant1.service.
Removed /etc/systemd/system/multi-user.target.wants/wpa_supplicant.service.
```

MichaIng, 2019-06-07:
@Sopor 
Jep, the error should not show up again now.