MichaIng/DietPi #2737: How to setup Wifi-Hotspot (AP) without Ethernet just for local subnetwork?

malakhovks, 2019-04-25:
How to setup Wifi-Hotspot (AP) without Ethernet just for local subnetwork?
When I am trying to setup AP via `Dietpi-Software` `WiFi Hotspot: turn your device into a wifi hotspot` it says that Ethernet test `FAILED`.
#### Required Information
- DietPi version 
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH=master
G_GITOWNER=Fourdee
```
- Distro version
```
DietPi_v6.17_NanoPiNEOAir-ARMv7-Stretch
```
- Kernel version
```
Linux DietPi 4.14.52 #19 SMP Thu Oct 11 11:33:51 CST 2018 armv7l GNU/Linux
```
- SBC device
```
NanoPi NEO Air (armv7l)
No Ethernet
```
- Power supply used
```
5V 2A
```
- SDcard used
```
 SanDisk ultra 16 gb
```

malakhovks, 2019-04-25:
I took a bottle of **red wine** and solved the problem. Instructions for the owners of **NanoPi NEO Air**. Please add this somewhere in wiki or docs section.
# Setting up NanoPi NEO Air as an access point in a standalone network without Ethernet on Dietpi OS

Use the following to update first:

```bash
$ apt-get update
$ apt-get upgrade
```
Install all the required software in one go with this command: 
```bash
$ apt-get install dnsmasq hostapd dhcpcd5
```
Since the configuration files are not ready yet, turn the new software off as follows: 
```bash
$ systemctl stop dnsmasq
$ systemctl stop hostapd
```

### Configuring a static IP

We are configuring a standalone network to act as a server, so the Nanopi Neo Air needs to have a static IP address assigned to the wireless port. This documentation assumes that we are using the standard 192.168.x.x IP addresses for our wireless network, so we will assign the server the IP address 192.168.2.1. It is also assumed that the wireless device being used is `wlan0`.

To configure the static IP address, edit the dhcpcd configuration file with: 
```bash
$ nano /etc/dhcpcd.conf
```
Go to the end of the file and edit it so that it looks like the following:
```
interface wlan0
    static ip_address=192.168.2.1/24
```
Now restart the dhcpcd daemon and set up the new `wlan0` configuration:
```bash
$ service dhcpcd restart
```

### Configuring the DHCP server (dnsmasq)

The DHCP service is provided by dnsmasq. By default, the configuration file contains a lot of information that is not needed, and it is easier to start from scratch. Rename this configuration file, and edit a new one:

```bash
$ mv /etc/dnsmasq.conf /etc/dnsmasq.conf.orig  
$ nano /etc/dnsmasq.conf
```

Type or copy the following information into the dnsmasq configuration file and save it:

```
interface=wlan0      # Use the require wireless interface - usually wlan0
  dhcp-range=192.168.2.2,192.168.2.20,255.255.255.0,24h
```

So for `wlan0`, we are going to provide IP addresses between 192.168.2.2 and 192.168.2.20, with a lease time of 24 hours. If you are providing DHCP services for other network devices (e.g. eth0), you could add more sections with the appropriate interface header, with the range of addresses you intend to provide to that interface.

There are many more options for dnsmasq; see the [dnsmasq documentation](http://bashwww.thekelleys.org.uk/dnsmasq/doc.html) for more details.

### Configuring the access point host software (hostapd)

You need to edit the hostapd configuration file, located at /etc/hostapd/hostapd.conf, to add the various parameters for your wireless network. After initial install, this will be a new/empty file.

```bash
$ nano /etc/hostapd/hostapd.conf
```

Add the information below to the configuration file. This configuration assumes we are using channel `6`, with a network name of `dietpi`, and a password `1234567890`. Note that the name and password should **not** have quotes around them. 

```
interface=wlan0
driver=nl80211
ssid=dietpi
hw_mode=g
channel=6
wmm_enabled=1 # QoS support
ieee80211n=1 # 802.11n support
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=1234567890
wpa_key_mgmt=WPA-PSK
#wpa_pairwise=TKIP CCMP # You better do not use this weak encryption (only used by old client devices)
rsn_pairwise=CCMP
ht_capab=[HT40][SHORT-GI-20][DSSS_CCK-40]
```

We now need to tell the system where to find this configuration file.

```bash
$ nano /etc/default/hostapd
```

Find the line with `#DAEMON_CONF`, and replace it with this:

```
DAEMON_CONF="/etc/hostapd/hostapd.conf"
```

### INTERFACES

Since in the previous tutorial we set up the board to automatically connect to a network we must now adapt the file to its new behaviour.
`/etc/network/interfaces`

```
#CONNECT TO WIFI NETWORK
#auto wlan0
#allow-hotplug wlan0
#iface wlan0 inet dhcp
#wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
#iface default inet dhcp

#CREATE ACCESS POINT
auto wlan0
iface wlan0 inet static
  address 192.168.42.1
  netmask 255.255.255.0
  
#Local loopback
auto lo
iface lo inet loopback
```
### Start it up

Now start up the remaining services:

```bash
$ service hostapd start  
$ service dnsmasq start
$ reboot
```

### Connect to the `dietpi` network and SSH to 192.168.2.1

```bash
$ ssh root@192.168.2.1
```

MichaIng, 2019-04-25:
@malakhovks 
Great write-up. Jep indeed our WiFi Hotspot solution currently does not support local LAN-only setups. It is intended that you have an internet capable Ethernet beside so updates checks, installing packages etc still works and to assure there is always a left SSH way even if things break.

Just a little note:
- We use `isc-dhcp-server` for our Hotspot implementation.
- Major reason is that it works well with `isc-dhcp-client` which is mostly the default on Debian and allows to run and be started on a per-interface level via `ifupdown` hooks. It allows `iface eth0 inet dhcp` entries while `dhcpcd` requires `iface eth0 inet manual`, thus not so well integrated with `ifupdown`.
- But yeah in your case it doesn't matter at all and generally `dhcpcd` works perfectly well as client and server.

diveyez, 2019-05-16:
Just use this instead https://github.com/garywill/linux-router

malakhovks, 2019-05-16:
> Just use this instead https://github.com/garywill/linux-router

Thanks, I'll try to use it.

malakhovks, 2019-05-19:
> Just use this instead https://github.com/garywill/linux-router

I tried this option, **it does not work** for my setup with NanoPi NEO Air (armv7l) and No Ethernet :
```bash
root@DietPi:~/linux-router# ./lnxrouter -n --ap wlan0 dietpi --password 1234567890q
WARN: brmfmac driver doesn't work properly with virtual interfaces and
      it can cause kernel panic. For this reason we disallow virtual
      interfaces for your adapter.
      For more info: https://github.com/oblique/create_ap/issues/203
ERROR: Your adapter can not be a station (i.e. be connected) and an AP at the same time
```

diveyez, 2019-05-19:
You can use two WiFi adapters....

malakhovks, 2019-06-07:
So, maybe any new suggestions ?

diveyez, 2019-06-07:
Purchase the needed adapter.....

MichaIng, 2019-06-07:
@malakhovks 
Hmm it looks like linux-router also expects a second interface, or it just fails if the WiFi interface is still up, connected to another AP 🤔. You need to take care at least first that WiFi is completely down:
`ifdown wlan0`
And in `/etc/network/interfaces` comment all wlan0/WiFi related lines (the thirst block). Then retry to apply `lnxrouter`.

Joulinar, 2020-06-28:
Just for the record @MichaIng 

Another way how to setup Hotspot/AP if you like to use simultaneous AP and Managed Mode Wifi same time on a SBC that don't have `eth0` interface.

https://dietpi.com/phpbb/viewtopic.php?f=9&t=7828

MichaIng, 2020-06-28:
Totally makes sense. I already started to work on the new network setup, however lack of time and some other urgent topics prevented me from finishing it. Next regular release will definitely contain it!

jcgdjob, 2021-06-08:
I've tried @malakhovks method below (address changed to 192.168.2.1) but error occur:
`ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52`
Any idea? Thanks!

> I took a bottle of **red wine** and solved the problem. Instructions for the owners of **NanoPi NEO Air**. Please add this somewhere in wiki or docs section.
> 
> #CREATE ACCESS POINT
> auto wlan0
> iface wlan0 inet static
>   address 192.168.**2**.1
>   netmask 255.255.255.0
> ```