MichaIng/DietPi #2738: Disable swap on Asus TinkerBoard

GR3S, 2019-04-26:
How i can disable swap (incl. zram) on Asus TinkerBoard?
I tried `swapoff -a; /DietPi/dietpi/func/dietpi-set_dphys-swapfile 0 /var/swap` and removing any matching reference found in `/etc/fstab` but it didn't help.
`cat /proc/swaps` outputs zram partitions.

**Required Information:**
- DietPi Version: 6.18
- SBC Device: Asus TinkerBoard
- SD card used: SanDisk Ultra microSDHC 32GB 10 Class UHS-I [SDSQUNS-032G-GN3MN]

MichaIng, 2019-04-26:
@GR3S 
Just had had a look into zram:
- Those are no disk swaps that could be handled by swapoff (thus our script), but RAM swaps into a compressed block of the RAM.
- So I suggest this should be left in place since it allows to use the available RAM space more efficiently. Less used stuff can be "swapped" into these compressed blocks but stay in RAM. No additional disk I/O.

But I am wondering why those are active since zram seems to require an additional APT package: `zram-config`

Does `swapon -s` show the zram swaps as well?
______-
Futher reading: Hmm on Debian zram-config is not available, but zram-tools instead, although only on Buster: https://packages.debian.org/buster/zram-tools
- Wiki: https://wiki.debian.org/ZRam

Formaly it was called compcache and indeed there seems to be no Debian package to handle or enable them user friendly. But by default it should not be active. So I guess if this is indeed your case, that TinkerOS somehow hardcoded it or has some service active by default.

Could you print: `systemctl -l`
This should list all active boot services so we can check if there is anything unordinary or directly related to zram initialization.
________
But totally apart from the actual question:
- I find this very interesting.
- SBCs usually have very limited RAM for heavy desktop, video or larger web application usage. Or when some Java application is active that eats 500M...
- A swapfile is a solution but leads to disk I/O which is especially a bad thing for SDcards which do not stand very long with heavy swapfile usage. Yeah of course you can move the swap to an external drive (I do so) but then that is spinning more regularly, comsuming energy and producing noise.
- zram is actually great trade. It cannot do magic but since less used RAM content (that is about to be swapped) is only compressed but stays in RAM, there is no additional disk I/O but still more RAM space available.
- Yeah you limit the RAM initially, so this is not something that is useful when RAM limit is only reached very rarly (e.g. daily maintainance tasks by cron job or something), but if you are generally close to the limit so swap happens quite often, but with not much data overall, a zram swap IMO makes sense to keep those little swaps in RAM.

Perhaps we can add zram handling by times, although not before Debian Buster is released and the majority of our SBC images migrated.

GR3S, 2019-04-26:
Thx for the detailed answer.
I want to install Kubernetes and it requires a disabled swap.
I am familiar with the benefits of using zram, but not sure that the Kubernetes will work without problems.

I found the following solution to disable zram:
edit `/etc/default/armbian-zram-config` and set ENABLED=false
and `systemctl disable armbian-zram-config`

MichaIng, 2019-04-29:
@GR3S 
Ah, does this come from any APT package?
```
dpkg -S /etc/default/armbian-zram-config
```

Strange since I didn't find this on our image. When did you download the image?

rpecb, 2019-05-01:
```
root@DietPi:~# dpkg -S /etc/default/armbian-zram-config
dpkg-query: no path found matching pattern /etc/default/armbian-zram-config
root@DietPi:~# dpkg -S /usr/lib/armbian/armbian-zram-config
linux-stretch-root-next-tinkerboard: /usr/lib/armbian/armbian-zram-config
```

https://dietpi.com/downloads/images/DietPi_ASUSTB-ARMv7-Stretch.7z - downloaded on 19 apr

Checksums (sha256):
```
3d26fba8ae87f0d2d3674fee47a92bf254504b59caaa899f3129af35c2a1ea91  DietPi_ASUSTB-ARMv7-Stretch.7z
33707ed2f471f4b077a6116667d26aa2acaea2b385c2c06839c8b33a199b0b8b  DietPi_v6.18_ASUSTB-ARMv7-Stretch.img
```

MichaIng, 2019-05-02:
Okay I checked the ARMbian repo:
- The `linux-<distro>-root-<branch>-<device>` packages are core packages of the ARMbian system and we preserve it for our images.
- They inherit `Replaces: zram-config` so ship an own zRam implementation, at least with the current version.

So yeah actually I think it can be quite useful but should not be enabled automatically. I will think it through, perhaps even skip the whole APT package as it anyway doubles some of features.

MichaIng, 2019-06-23:
Conflicts/bugs RAMlog in case: https://dietpi.com/phpbb/viewtopic.php?p=18548#p18548
- [ ] Will add a patch to disable+remove all (new) ARMbian services with v6.25

closed 2019-07-03 by MichaIng