MichaIng/DietPi #2746: DietPi-Software | PiVPN: Check official installer for IP redirection

MichaIng, 2019-04-29:
Ref: https://github.com/MichaIng/DietPi/issues/2505#issuecomment-487706128

Should be included, at least optionally, but added to the PiVPN repo itself at best.

tohjg, 2019-06-13:
For those who **installed pihole and pivpn**, but **no internet when device connected to vpn**, you need to either:

1. set `net.ipv4.ip_forward` to `1` inside `/etc/sysctl.d/99-sysctl.conf` 
2. or, you may create symlink from `/etc/sysctl.d/99-sysctl.conf` to `/etc/sysctl.conf`.

PiVPN installer modified `/etc/sysctl.conf` but it was ignored, instead it reads config file(s) from `/etc/sysctl.d`. 

Reported using Raspberry Pi Model 3+, dietpi version 6.24.1.

MichaIng, 2019-06-14:
@tohjg 
A symlink is missing there, actually is should be: `/etc/sysctl.d/99-sysctl.conf => /etc/sysctl.conf`
To recover this symlink: `ln -sf /etc/sysctl.conf /etc/sysctl.d/99-sysctl.conf`
Actually I thought we recreated this already some DietPi versions ago via update... let me check...

MichaIng, 2019-06-17:
IPv6 support on PiVPN still outstanding, so we leave it as is for now: https://github.com/pivpn/pivpn/issues/259

The other issue is strange, since we already re-created missing symlink with v6.23 so it is still outstanding why/how it got removed again. At least we never touch it.

However I mark this issue as closed for now.

closed 2019-06-17 by MichaIng