MichaIng/DietPi #2750: Open Beta v6.23 | Please help testing and hardening the upcoming release

MichaIng, 2019-04-30:
RC version | v6.23.0
---------- | -------
Changelog | https://github.com/Fourdee/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/Fourdee/DietPi/compare/master...beta
How to apply | https://github.com/Fourdee/DietPi/blob/beta/BRANCH_SYSTEM.md
Release planned | Early Mai

#### When testing, especially have an eye on features mentioned in the change log above that you regularly use anyway.
_______________
### Known issues
#### DietPi functionality
- DietPi-Config | Enabling WiFi + Ethernet adapters, both on different subnets, breaks WiFi connection in some cases: https://github.com/Fourdee/DietPi/issues/2103
  - Current workaround is to disable the adapter that is not in use (not for internet connection).
  - Otherwise a custom routing table is required.

#### SBC/device related
- RPi | LXAppearance (on LXDE desktop) hangs on dbus-launch: https://github.com/Fourdee/DietPi/issues/1791
- RPi | Enabling OpenGL/VC4 causes sound card selection to be invalid: https://github.com/Fourdee/DietPi/issues/2173
- Pine A64 | Bluetooth add-on module currently not supported: https://github.com/Fourdee/DietPi/issues/528
- Odroid XU4 | Kodi freezes shortly on video playback: https://github.com/MichaIng/DietPi/issues/2584
- Rock64 | 3.5mm A/V jack is currently not functional: https://github.com/MichaIng/DietPi/issues/2522

#### Software title related
https://github.com/MichaIng/DietPi/commit/16d37ea0e58c3de53632ae37cb468f7d0cd08c67**
- Node-RED | Pre-installed modules cannot be updated via webUI: https://github.com/Fourdee/DietPi/issues/2073
- Raspimjpeg | With Lighttpd, streaming mjpeg does not work: https://github.com/Fourdee/DietPi/issues/1747
- WiFi Hotspot | Currently not working on some Odroid C2 and RPi devices: https://github.com/Fourdee/DietPi/issues/1955, https://github.com/MichaIng/DietPi/issues/2531

Fourdee, 2019-05-01:
REF Twitter: https://t.co/qxpVwpjkV4

MichaIng, 2019-05-12:
### Release done: https://github.com/MichaIng/DietPi/pull/2784
### Many thanks to all tester ❤️ !

closed 2019-05-12 by MichaIng