MichaIng/DietPi #2754: dietpi-config does not store settings

tatoosh, 2019-05-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH=master
G_GITOWNER=MichaIng

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretch

- Kernel version | `uname -a`
Linux z1 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
 RPi 3 Model B+ (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2A

- SDcard used | (EG: SanDisk ultra)
SanDisk

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated?
fresh image

- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

 `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`
-bash: 0ad854f7-5922-4a36-8e60-49cfe36c9ed6: Kommando nicht gefunden.
-bash: $'\r\E[K\E[0m\E[90m[\E[0m\E[32m': Kommando nicht gefunden.

#### Steps to reproduce
dietpi-config does not store settings

#### Expected behaviour
1. dietpi-config should be set from performance to ondemand
2. soundcard hifiberry dac should be set instead if rpi built in

MichaIng, 2019-05-01:
@tatoosh 
Many thanks for your report.

Could you be a bid more specific:
- You change CPU governor in DietPi-Config performance settings from `performance` to `ondemand` and after reboot it is reverted to `performance`? And the same for the sound card?

tatoosh, 2019-05-02:
@MichaIng 

> You change CPU governor in DietPi-Config performance settings from performance to ondemand and after reboot it is reverted to performance? And the same for the sound card?

Yes. i can reproduce it on other rPis

I solved the sound card issue it by editing /boot/config.txt and /DietPi/config.txt
I tried only changing one file, that did not work. Normally it is enough to change `/DietPi/config.txt`, right?

How to solve the cpu governor problem? my cpu is running warm :)

MichaIng, 2019-05-02:
@tatoosh 
Only `/DietPi/config.txt` should be required. So then possibly the RAMdisk service fails to sync the changes to disk on shutdown. Please paste:
```
journalctl -u dietpi-ramdisk
cat /var/tmp/dietpi/logs/dietpi-ramdisk.log
```

Meanwhile you can set the CPU governor as well via `/boot/config.txt` (disk location) + `/DietPi/config.txt` (RAM location): `CONFIG_CPU_GOVERNOR=ondemand`
Then to apply the governor to current session: `/DietPi/dietpi/func/dietpi-set_cpu`

tatoosh, 2019-05-02:
Thx for the fast help.

i edited both config.txt files and set the `CONFIG_CPU_GOVERNOR=ondemand` 
`/DietPi/dietpi/func/dietpi-set_cpu`

**Now it is on pwersave, not on ondemand** - is that correct?

> journalctl -u dietpi-ramdisk
> cat /var/tmp/dietpi/logs/dietpi-ramdisk.log

[image]

MichaIng, 2019-05-02:
Hmm powersafe is not ondemand AFAIK, how did you check?

```
cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
```
Should show ondemand.

The RAMdisk log ends with 28. April while journal shows it was started on boot successfully. Is the the whole log or does it have more lines? The very last entries from today are the interesting ones.

Did /DietPi/dietpi/func/dietpi-set_cpu throw any error about unsupported governor?

tatoosh, 2019-05-06:
Here are your suggested checks - not really correct or?

[image]

MichaIng, 2019-05-06:
@tatoosh 
> i edited both config.txt files and set the CONFIG_CPU_GOVERNOR=ondemand

You mean you edited `/DietPi/dietpi.txt` with this, right? Generally apply those changes simply via `dietpi-config` > Performance Options

Ah damn, I see I mentioned it wrong above. Yeah `config.txt` does not support setting the CPU governor, it's `dietpi.txt`. So as mentioned above, please try to use `dietpi-config`, skip the reboot and run `/DietPi/dietpi/func/dietpi-set_cpu`.

MichaIng, 2019-05-26:
@tatoosh 
Could you solve it (apply ondemand governor)? I mark the issue as closed, but feel free to reopen if the issue persists.

closed 2019-05-26 by MichaIng