MichaIng/DietPi #2755: Node Red installation seems to fail

axwax, 2019-05-02:
### Creating a bug report/issue
Despite Node-Red seemingly installing fine it doesn't appear to be installed where its service thinks it is (/usr/local/bin/node-red). Additionally, the node-red command can't be found.

#### Required Information
- DietPi version | 6.22.3 (master/Fourdee)
- Distro version | Stretch 9.9
- Kernel version | Linux homecontrol 4.14.98+ #1200 Tue Feb 12 20:11:02 GMT 2019 armv6l GNU/Linux
- SBC device | RPi Zero W (armv6l)
- Power supply used | 5V, 2.1A
- SDcard used | Kingston SDCS 16GB

#### Additional Information (if applicable)
- Software title | Node Red
- Was the software title installed freshly or updated/migrated? installed freshly
- Can this issue be replicated on a fresh installation of DietPi? it was a fresh install, but upgraded from 6.16 (I think)
- Bug report 12cecaf2-19ae-476c-9625-7163da6f0081

#### Steps to reproduce

1. upgrade DietPi 6.16 to 6.22.3
2. in dietpi-software choose mosquitto, node-red, influxdb, grafana-server
3. run systemctl status node-red.service
4. run node-red

#### Expected behaviour

    Loaded: loaded
    Active: active
    ...

```
node-red
Welcome to Node-RED
===================
```

#### Actual behaviour

```
● node-red.service - Node-Red (DietPi)
Loaded: loaded (/etc/systemd/system/node-red.service; disabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Thu 2019-05-02 15:11:03 BST; 14min ago
Process: 9716 ExecStart=/usr/local/bin/node-red -u /mnt/dietpi_userdata/node-red (code=exited, status=203/EXEC)
Main PID: 9716 (code=exited, status=203/EXEC)

May 02 15:11:03 homecontrol systemd[1]: Started Node-Red (DietPi).
May 02 15:11:03 homecontrol systemd[9716]: node-red.service: Failed at step EXEC spawning /usr/local/bin/node-red: No such file or direc
tory
May 02 15:11:03 homecontrol systemd[1]: node-red.service: Main process exited, code=exited, status=203/EXEC
May 02 15:11:03 homecontrol systemd[1]: node-red.service: Unit entered failed state.
May 02 15:11:03 homecontrol systemd[1]: node-red.service: Failed with result 'exit-code'.
```

```
node-red
-bash: node-red: command not found
```

#### Extra details
Other than the original install file being outdated (I think it was 6.16) it was a vanilla install, which I then upgraded to the current version before attempting any software install.

MichaIng, 2019-05-02:
Many thanks for your report.

Hmm, so the Node-RED install must have failed. Any error messages you remember from DietPi-Software run?

Please try to reinstall/verify manually:
npm i -g node-red

axwax, 2019-05-02:
Wow, thanks for the super quick response!
I just realised the problem seems to be with the node.js installation:
While it is ticked in Software Additional there is neither a node nor an npm command.

I just did a `dietpi-software  reinstall 9`
and the installer fails:
```
2019-05-02 21:08:32 (619 KB/s) - ‘node_install.sh’ saved [1943/1943]

Node Linux Installer by www.github.com/taaem
Running as Root User
Get Latest Version Number...
Done
Downloading latest stable Version ...
Done
Installing...

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
ERROR: Couldn't extract tar
```
I did choose the lightweight 7zip option when prompted during the upgrade process to 6.22.3 - could that be the problem?

MichaIng, 2019-05-02:
@axwax 
I just tested and on x86 VM it succeeds.

> I did choose the lightweight 7zip option when prompted during the upgrade process to 6.22.3 - could that be the problem?

Nope, the error is about `gzip`, not `7zip`. Hum let me check the Node install code.

MichaIng, 2019-05-02:
See my comments in the Node GitHub issue. Looking at the download page it really seems to be planned/wanted that latest Node does not support ARMv6 anymore, thus does not work on any Raspbian based RPi image....

I really hope they rething this, however until then, and as the installer does not handle this gracefully, we have to manually choose Node 11 for ARMv6 then...

MichaIng, 2019-05-02:
@axwax 
Okay found a solution that is not too complicated. We can fix the installer easily when downloaded. Please try:
```
wget https://raw.githubusercontent.com/taaem/nodejs-linux-installer/master/node-install.sh
chmod +x node_install.sh
sed -i 's|nodejs.org/dist/latest/|nodejs.org/dist/latest-v11.x/' node_install.sh
./node_install.sh
rm node_install.sh
```

axwax, 2019-05-02:
That's amazing! 
Ijust followed your above node.js install instructions, and got it all running.
I did have to  change `node_install` to `node-install` in lines 2-5 though.

I just looked at your commit, and I think you might have to check for that too: 
You changed `wget "$INSTALL_URL_ADDRESS" -O node_install.sh` to `wget "$INSTALL_URL_ADDRESS"`, so the downloaded file will be called `node-install.sh` making the following lines fail I think!

Anyway, my Node Red seems to be installing beautifully now - thanks for such a speedy fix, and I'm looking forward to 6.23! 😃

MichaIng, 2019-05-02:
@axwax 
Ah jep you're right. Just thought that defining the same if obsolete if the URL contains it already. But indeed the tiny difference.

closed 2019-05-12 by MichaIng

MichaIng, 2020-03-22:
For reference, PR up to fix the installer: https://github.com/taaem/nodejs-linux-installer/pull/11