MichaIng/DietPi #2759: Bluetooth problems 'sap'

SaturnusDJ, 2019-05-03:
```
root@HC:~# service bluetooth stop
root@HC:~# service bluetooth start
root@HC:~# service bluetooth status
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-05-03 12:35:23 CEST; 2s ago
     Docs: man:bluetoothd(8)
 Main PID: 2680 (bluetoothd)
   Status: "Running"
      CPU: 34ms
   CGroup: /system.slice/bluetooth.service
           └─2680 /usr/lib/bluetooth/bluetoothd

May 03 12:35:23 HC systemd[1]: Starting Bluetooth service...
May 03 12:35:23 HC bluetoothd[2680]: Bluetooth daemon 5.43
May 03 12:35:23 HC systemd[1]: Started Bluetooth service.
May 03 12:35:23 HC bluetoothd[2680]: Starting SDP server
May 03 12:35:23 HC bluetoothd[2680]: Bluetooth management interface 1.14 initialized
May 03 12:35:23 HC bluetoothd[2680]: Failed to obtain handles for "Service Changed" characteristic
May 03 12:35:23 HC bluetoothd[2680]: Sap driver initialization failed.
May 03 12:35:23 HC bluetoothd[2680]: sap-server: Operation not permitted (1)
```

Some weeks ago I had to reinstall DietPi because of Bluetooth trouble too. Not sure if it was the same as this. I tried some solutions provided online, but none working.

At first the Bluetooth didn't work at all any more. After disabling Bluetooth in `dietpi-config` (which also removes the software), rebooting, and reinstalling, it works again, but the above error still appears.
Note that without rebooting, it did not work.

cat /DietPi/dietpi/.version
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH=master
G_GITOWNER=MichaIng

echo $G_DISTRO_NAME` or `cat /etc/debian_version
-bash: or: command not found
stretchcat /etc/debian_version

uname -a
Linux HC 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

echo $G_HW_MODEL_DESCRIPTION
RPi 3 Model B+ (armv7l)

Power supply
Official

SD Card
Kingston SDCR/32GB

`dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`
[Show what is send] --> Error cmd not found + SSH session froze

MichaIng, 2019-05-04:
@SaturnusDJ 
Actually the SAP errors can be safely ignored: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=191073&sid=9db75ddad5861bed59784bd1dfc58d2f
It's the "SIM Access Profile" which does not apply to the RPi of course.

You can get rid of the warning by editing the service file to run `bluetoothd` with `--noplugin=sap` option, but the warnings should not hurt at all and I would just leave them instead of manually editing systemd units.

We could implement this option via drop-in config, since I hardly believe that anyone uses this on a SBC:
```
mkdir /etc/systemd/systemd/bluetooth.service.d
echo -e '[Service]\nExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=sap' > /etc/systemd/systemd/bluetooth.service.d/dietpi-no_sap.conf
```
But really, it does not hurt and limits one possible use case.

SaturnusDJ, 2019-05-04:
Understood. 👌 

I will keep watching Bluetooth though, because sometimes I am having Bluetooth issues that manifests itself through software tripping on BT functionality.