MichaIng/DietPi #2760: Docker 5 minutes delay on startup

SaturnusDJ, 2019-05-03:
```
root@HC:~# uptime
 12:42:32 up 29 min,  1 user,  load average: 0.12, 0.12, 0.12

root@HC:~# docker ps
CONTAINER ID     IMAGE                               STATUS
blablablabla     portainer/agent:linux-arm-1.2.1     Up 24 minutes
```

So it took 5 minutes before the first container was up.

```
May 03 12:13:09 HC systemd[1]: Starting Docker Socket for the API.
-- Subject: Unit docker.socket has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit docker.socket has begun starting up.
May 03 12:13:09 HC systemd[1]: Listening on Docker Socket for the API.
-- Subject: Unit docker.socket has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit docker.socket has finished starting up.
--
-- The start-up result is done.
```
```
May 03 12:13:24 HC systemd[1]: Started Docker Application Container Engine.
-- Subject: Unit docker.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit docker.service has finished starting up.
--
-- The start-up result is done.
```
```
May 03 12:17:42 HC dockerd[873]: time="2019-05-03T12:17:42.526984702+02:00" level=info msg="libcontainerd: started new docker-containerd process" pid=1248
```

Didn't find any other data that might help to debug.

SaturnusDJ, 2019-05-03:
Seems to be like this: https://www.raspberrypi.org/forums/viewtopic.php?p=1316562
```
May 04 00:23:43 HC kernel: random: crng init done
May 04 00:23:43 HC kernel: random: 7 urandom warning(s) missed due to ratelimiting
```

Don't know where it comes from. Can't remember having changed system critical stuff manually.
Here are all red coloured lines from log:
```
Nov 03 18:16:42 HC kernel: bcm2708_fb soc:fb: Failed to allocate GPU framebuffer (-22)
Nov 03 18:16:42 HC kernel: bcm2708_fb soc:fb: probe failed, err -22

May 04 00:19:02 HC kernel: vc_vchi_sm_init: failed to open VCHI service (-1)
May 04 00:19:02 HC kernel: [vc_sm_connected_init]: failed to initialize shared memory service

May 04 00:19:02 HC kernel: bcm2835-clk 3f101000.cprman: pllh: couldn't lock PLL

May 04 00:19:02 HC kernel: bcm2835-clk 3f101000.cprman: pllh: couldn't lock PLL

May 04 00:19:02 HC kernel: bcm2835-clk 3f101000.cprman: pllh: couldn't lock PLL

May 04 00:19:02 HC kernel: bcm2835-clk 3f101000.cprman: pllh: couldn't lock PLL

May 04 00:19:10 HC bluetoothd[785]: Failed to obtain handles for "Service Changed" characteristic

May 04 00:19:10 HC bluetoothd[785]: Sap driver initialization failed.
May 04 00:19:10 HC bluetoothd[785]: sap-server: Operation not permitted (1)

May 04 00:19:12 HC bluetoothd[785]: Loading LTKs timed out for hci0
```

SaturnusDJ, 2019-05-03:
Provided solution online works:
```
apt install haveged
systemctl enable haveged
```
Still no clue where this came from though.

MichaIng, 2019-05-04:
@SaturnusDJ 
Many thanks for your report.

Hmm I have the warnings as well on my RPi but never any issues on boot:
```
2019-05-04 13:52:26 root@micha:/DietPi# dmesg | grep urandom
[    3.093493] random: systemd: uninitialized urandom read (16 bytes read)
[    3.107900] random: systemd: uninitialized urandom read (16 bytes read)
[    3.108873] random: systemd: uninitialized urandom read (16 bytes read)
[   15.277467] random: 7 urandom warning(s) missed due to ratelimiting
```

Not sure how `/dev/random` works but many random generators are based on user inputs indeed. So on headless systems where no user input occurs for long times, there might be simply not enough random values generated. The package you installed seems to address this by "filling the random entropy pool" when dropping under certain range: https://packages.debian.org/stretch/haveged

However these messages occur on boot only and isn't it totally usual that during boot there is of course no user input done? However I read into this by times. In question if no random values can be generated this at best is a security issue when hashes and such are to be generated. So at least it is worth having a look.

So after installing haveged boot works fine, quick as expected?

MichaIng, 2019-05-04:
Further reading:
- Well known issue: https://github.com/systemd/systemd/issues/4167
- A Debian fix seemed to have it made worse, leading to the boot delay, you observe, in some cases: https://bugs.chromium.org/p/project-zero/issues/detail?id=1559
- Linux kernel 4.19 seems to allow solving it, which is expected soon for RPi: https://github.com/Hexxeh/rpi-firmware/tree/master

SaturnusDJ, 2019-05-04:
> 
> 
> @SaturnusDJ
> Many thanks for your report.

No problem, thanks for the kind responses always.

> So after installing haveged boot works fine, quick as expected?
> Was there finally any relation with docker?

Booting is quick again after installing the package.

In my case the main problem that rises from the delay is that Docker containers don't start. That means that critical services have an extra downtime of 5 minutes.

> And btw all kernel errors you pasted above are expected on headless systems with GPU memory share 16M and in case the headless mode selected via dietpi-config. Those can be safely ignored.
> 
> The bluetoothd SAP errors can be ignored as well, but see your other issue about that.

Thanks. 👍 

> 
> 
> Further reading:
> 
>     * Well known issue: [systemd/systemd#4167](https://github.com/systemd/systemd/issues/4167)
> 
>     * A Debian fix seemed to have it made worse, leading to the boot delay, you observe, in some cases: https://bugs.chromium.org/p/project-zero/issues/detail?id=1559
> 
>     * Linux kernel 4.19 seems to allow solving it, which is expected soon for RPi: https://github.com/Hexxeh/rpi-firmware/tree/master

Good news. :)

closed 2019-05-07 by MichaIng