MichaIng/DietPi #2761: remove banner and sudo question for standard user

Sopor, 2019-05-03:
I found [this](https://dietpi.com/phpbb/viewtopic.php?t=3602) post about using sudo at login for a standard user. I'm running DietPi 6.22.3 and i still have to enter sudo password at login. I have tried to disable it but i can't find where to do it. It is quite annoying to enter the password twice when i login.

Why isn't this fixed yet? The post i found was from May 20, 2018 and the problem still exists :(

MichaIng, 2019-05-03:
@Sopor 
Many thanks for your report.

Please see here for solution: https://github.com/MichaIng/DietPi/issues/2667

It is fixed for v6.23 which is currently in beta stage: https://github.com/MichaIng/DietPi/issues/2750

Sopor, 2019-05-04:
So i should only run this?
```
sed -i "/until sudo systemctl/c\		while pgrep -f '/DietPi/dietpi/postboot' &> /dev/null" /DietPi/dietpi/dietpi-login
```
What will it do, replace a line in the `dietpi-login` file?

MichaIng, 2019-05-04:
@Sopor 
Jep, it replaced the line containing the sudo call with a far simpler and quicker check without any root permission needs.

Sopor, 2019-05-04:
Now when i run `/DietPi/dietpi/login` it will not trigger the sudo. Thanks for your help @MichaIng

closed 2019-05-04 by Sopor

MichaIng, 2019-05-04:
@Sopor 
Perfect, many thanks for reporting back.

Sopor, 2019-05-07:
@MichaIng This work-a-round, isn't this persistent? After i rebooted i was forced to run this again :(
```
sed -i "/until sudo systemctl/c\		while pgrep -f '/DietPi/dietpi/postboot' &> /dev/null" /DietPi/dietpi/dietpi-login
```

MichaIng, 2019-05-07:
@Sopor 
Hmm should only be required once.

Did RAMdisk sync successfully?
```
systemctl status dietpi-ramdisk
tail /var/tmp/dietpi/logs/dietpi-ramdisk.log
```

Sopor, 2019-05-07:
So it use some kind of RAM disk? How often will this RAM disk save to disk?

MichaIng, 2019-05-07:
@Sopor 
Ahh a power failure, jep then it makes sense. It only saves to disk on regular shutdown.
Should be worth to investigate the reason for the power failure. RAMdisk not syncing to disk is not a real issue but when you have some database action of general disk I/O running, then you can easily end up with corrupted database or worse. Check `dmesg` for any possible reasons/kernel errors, like insufficient voltage and such. But that's another topic 😉.

So then to be safe, edit on disk as well:
```
sed -i "/until sudo systemctl/c\		while pgrep -f '/DietPi/dietpi/postboot' &> /dev/null" /boot/dietpi/dietpi-login
```

Or you apply to our open beta for v6.23, which contains the fix: https://github.com/MichaIng/DietPi/issues/2750

Sopor, 2019-05-07:
Is this something that is unique for DietPi to use a RAM disk? What is the reason to use RAM disk, speed? Will it read/write to boot so much that a RAM disk is needed?

MichaIng, 2019-05-07:
@Sopor 
Main reason is speed indeed. It is not writing very much, but the scripts run smoother on slow SDcards.