MichaIng/DietPi #2762: Update sabnzbd

19eighties, 2019-05-05:
How do I update sabnzbd to the latest version (2.3.8)? It would be great if I can update via the built-in updater, but it seems to lag behind releases for quite a while.

MichaIng, 2019-05-05:
@19eighties 
Many thanks for your question/report.

Jep indeed we currently install a fixed older version. I opened a PR to always pull the latest release automatically. I also reworked it a bid so `dietpi-software reinstall 139` can be used to update SABnzbd without affecting the existing config file + some other minor enhancements to the systemd unit and such: https://github.com/sabnzbd/sabnzbd/issues/1283

I will test on all distro versions and merge into v6.23 which is expected to be released in a few days.

19eighties, 2019-05-08:
Thanks for your efforts, much appreciated!

closed 2019-05-12 by MichaIng

19eighties, 2021-02-26:
Following up on this issue since I am trying to update sabnzbd via "dietpi-software reinstall 139", but I get an error about the version of python running. I tried running the linked script before running the above command, but it didn't solve the issue. Do I need to update python? If so, is there a standard way of updating it with the built-in dietpi tools?

`ERROR: This script does not work on Python 3.5 The minimum supported Python version is 3.6. Please use https://bootstrap.pypa.io/3.5/get-pip.py instead`

Joulinar, 2021-02-26:
hi, 

can you check the Debian version you are running on `echo $G_DISTRO_NAME`

MichaIng, 2021-02-26:
I fixed this already. Not long ago get-pip.py has dropped Python 3.5 support but as printed in the error output there are backwards compatible script versions available: https://github.com/MichaIng/DietPi/commit/c26703513c2135f84fd5efccad3a06de469ac4cb#diff-98445566b948cc76bb45181ccd463ee93a7a441dbc8b25763d1b88f05f4ef8c5

19eighties, 2021-02-27:
@MichaIng I manually ran the script from https://bootstrap.pypa.io/3.5/get-pip.py before the "dietpi-software reinstall 139", but the script still stalls at this point. Wouldn't it determine that pip is the latest version? How do I deal with the error?

19eighties, 2021-02-27:
@Joulinar Output is: stretch

19eighties, 2021-02-27:
I also tried reinstalling pip with "dietpi-software reinstall 130" and got the same error:

[image]

MichaIng, 2021-02-27:
Do the following:
```
sed -i 's|https://bootstrap.pypa.io/get-pip.py|https://bootstrap.pypa.io/3.5/get-pip.py|' /boot/dietpi/dietpi-software
dietpi-software reinstall 130
```

19eighties, 2021-02-27:
Ok, I did that and was able to install sabnzbd, but now it is not able to start. I checked the syslog and it shows these messages:

`
Feb 27 10:40:11 DietPi systemd[1]: Started SABnzbd (DietPi).
Feb 27 10:40:12 DietPi python3[6373]:   File "/etc/sabnzbd/SABnzbd.py", line 129
Feb 27 10:40:12 DietPi python3[6373]:     self._size: int = size
Feb 27 10:40:12 DietPi python3[6373]:               ^
Feb 27 10:40:12 DietPi python3[6373]: SyntaxError: invalid syntax
Feb 27 10:40:12 DietPi systemd[1]: sabnzbd.service: Main process exited, code=exited, status=1/FAILURE
Feb 27 10:40:12 DietPi systemd[1]: sabnzbd.service: Unit entered failed state.
Feb 27 10:40:12 DietPi systemd[1]: sabnzbd.service: Failed with result 'exit-code'.
Feb 27 10:40:12 DietPi systemd[1]: sabnzbd.service: Service hold-off time over, scheduling restart.
Feb 27 10:40:12 DietPi systemd[1]: Stopped SABnzbd (DietPi).
Feb 27 10:40:12 DietPi systemd[1]: Started SABnzbd (DietPi).
Feb 27 10:40:14 DietPi python3[6386]:   File "/etc/sabnzbd/SABnzbd.py", line 129
Feb 27 10:40:14 DietPi python3[6386]:     self._size: int = size
Feb 27 10:40:14 DietPi python3[6386]:               ^
Feb 27 10:40:14 DietPi python3[6386]: SyntaxError: invalid syntax
Feb 27 10:40:14 DietPi systemd[1]: sabnzbd.service: Main process exited, code=exited, status=1/FAILURE
Feb 27 10:40:14 DietPi systemd[1]: sabnzbd.service: Unit entered failed state.
Feb 27 10:40:14 DietPi systemd[1]: sabnzbd.service: Failed with result 'exit-code'.
Feb 27 10:40:14 DietPi systemd[1]: sabnzbd.service: Service hold-off time over, scheduling restart.
Feb 27 10:40:14 DietPi systemd[1]: Stopped SABnzbd (DietPi).`

MichaIng, 2021-02-27:
Ah, SABnzbd just raised the minimum Python dependency to v3.6 with yesterdays release: https://github.com/sabnzbd/sabnzbd/releases/tag/3.2.0

Okay, that means we need to either disable the install option on Stretch systems or install v3.1.1 manually 🤔. Okay I'll implement the downgrade, hang on...

19eighties, 2021-02-27:
Oh man, just missed it... so I'll be able to install 3.1.1?

MichaIng, 2021-02-27:
Yes either that or manually compiling Python 3.9 or upgrading the distro to Buster 😉.

Fixed with: https://github.com/MichaIng/DietPi/commit/cc092bb954d9e7e7aabecc3b1aa0848f75b4fc49
Changelog: https://github.com/MichaIng/DietPi/commit/fecc6e6bae761d9d3ead916a765be33e68cc9bc8

Joulinar, 2021-02-27:
Well going to Buster might be best option as Stretch is running out of support more and more.

19eighties, 2021-03-01:
I wasn't really aware of the new platform, so I will eventually update it, but I set it up so long ago it will take some time to figure out all the settings I made and moving over all the app data, etc.

Joulinar, 2021-03-01:
Well if you wait until summer, we will have already next Debian version Bullseye 🤣