MichaIng/DietPi #2764: Github organisation

FredericGuilbault, 2019-05-05:
I suggess you do a github organisation and transfer this repo to it. You deserve it. 

It would cut down noobs confusion about where the code live and where to fork.

As an organisation manager  you will still have access to commit directly on the organisation repository without having to submit pull request. 

It.s a 30 min to 1 hour job.

MichaIng, 2019-05-05:
@FredericGuilbault 
I also thought about this, to separate the repo from any individual user account. But since we just moved the repo from Fourdee over to mine, requiring adjustment of all download links and such, I will not do that soon. In the end it doesn't change much practically for end users.

Actually it would have been wise to make an organisation directly instead migration the repo to my account. However as said, practically no difference if it's `https://github.com/DietPi/DietPi` or `https://github.com/MichaIng/DietPi`.

FredericGuilbault, 2019-05-06:
It make sense. Thx for taking time to answer. Closing

closed 2019-05-06 by FredericGuilbault

FredericGuilbault, 2019-05-16:
One of the reasons for having a seperate Organisation repo. Is abstracting all those small commits that don't worth being described. They could published in a pull request that sum up all the changes into something that make a nice simple story for others. while the person who did the actual work keep trace of all the details.  

 
[image]

MichaIng, 2019-05-18:
@FredericGuilbault 
But for this no organisation is required, since we could also add all those small commits via pull requests, is it?

It is basically a question of effort vs visual aspects. I tend to add larger changes (software/feature requests, larger fixes) via PR, to have all related code changes + update patches + changelog entries and such in one squashed merge commit, which also allows to test the changes independently of possible parallel unrelated commits. But for smaller changes the effort is simply too high. And it is even good then to have them visible as single commit, so one can quickly check why/where this came from. Basically when selecting a file, the "Blame" option can then be used nicely to exactly track where which code part came from.

When having whole releases as single squashed commit, then one always needs to find the repo/branch where the single change was done, find the commit that is related to this change etc. etc. Not much benefit IMO. And if one only wants to see the version history, then the changelog is better for this anyway?

However I generally see the benefit of having the project as organisation separated from single user accounts. This totally makes sense. As mentioned above it might take a while until we go this step and then we will see how to practically use it. Such changes anyway require some time until a practical workflow has been looped in 😉.