MichaIng/DietPi #2765: Setting up samba (2:4.5.16+dfsg-1+deb9u1) failed

ErickPH, 2019-05-05:
#### Details:
- Date           | Sat  4 May 23:45:30 EST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Additional Information (if applicable)
- Software title : Dietpi-Software installing SAMBA
- Was the software title installed freshly or updated/migrated?
Software was removed due to inability to start after upgrade. Reinstalation caused issues.

- Can this issue be replicated on a fresh installation of DietPi?
not tried.

#### Steps to reproduce
Use Dietpi-Software to install SAMBA

#Error 
[  OK  ] DietPi-Software | Free space check: path=/ | available=12315 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/af7fffa7-80bc-d101-a057-ffa780bcd101/dietpi_userdata
[ .... ] DietPi-Software | (1/3) Testing connection to http://raspbian.raspberrypi.org/raspbian, ple[  OK  ] DietPi-Software | Connection test: http://raspbian.raspberrypi.org/raspbian
[  OK  ] NTPD: time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask all: avahi-daemon
[  OK  ] DietPi-Services | unmask all: nmbd
[  OK  ] DietPi-Services | unmask all: smbd
[  OK  ] DietPi-Services | unmask all: deluged
[  OK  ] DietPi-Services | unmask all: medusa
[  OK  ] DietPi-Services | unmask all: deluge-web
[  OK  ] DietPi-Services | unmask all: cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : deluge-web
[  OK  ] DietPi-Services | stop : medusa
[  OK  ] DietPi-Services | stop : deluged
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[  OK  ] DietPi-Services | stop : avahi-daemon

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT

[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
[  OK  ] DietPi-Software | G_AGUP
[ INFO ] DietPi-Software | Running apt simulation to check for errors, please wait...
[ INFO ] DietPi-Software | APT installation for: bash-doc -s, please wait...
Inst bash-doc (4.4-5 Raspbian:stable [all])
Conf bash-doc (4.4-5 Raspbian:stable [all])
Conf samba (2:4.5.16+dfsg-1+deb9u1 Raspbian:stable [armhf])
[  OK  ] DietPi-Software | G_AGI bash-doc -s
[ INFO ] DietPi-Software | APT upgrade, please wait...
Setting up samba (2:4.5.16+dfsg-1+deb9u1) ...
Samba is not being run as an AD Domain Controller, masking samba-ad-dc-service.
Please ignore the following error about deb-systemd-helper not finding samba-ad-dc-service.
Failed to preset unit: Unit file /etc/systemd/system/samba-ad-dc.service is masked.
/usr/bin/deb-systemd-helper: error: systemctl preset failed on samba-ad-dc.service: No such file or directory
Job for smbd.service failed because the control process exited with error code.
See "systemctl status smbd.service" and "journalctl -xe" for details.
invoke-rc.d: initscript smbd, action "start" failed.
● smbd.service - Samba SMB Daemon
   Loaded: loaded (/lib/systemd/system/smbd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-05-04 23:44:33 EST; 38ms ago
     Docs: man:smbd(8)
           man:samba(7)
           man:smb.conf(5)
  Process: 14076 ExecStart=/usr/sbin/smbd $SMBDOPTIONS (code=exited, status=1/FAILURE)
 Main PID: 14076 (code=exited, status=1/FAILURE)

May 04 23:44:33 DietPi systemd[1]: Starting Samba SMB Daemon...
May 04 23:44:33 DietPi systemd[1]: smbd.service: Main process exited, code=…LURE
May 04 23:44:33 DietPi systemd[1]: Failed to start Samba SMB Daemon.
May 04 23:44:33 DietPi systemd[1]: smbd.service: Unit entered failed state.
May 04 23:44:33 DietPi systemd[1]: smbd.service: Failed with result 'exit-code'.
Hint: Some lines were ellipsized, use -l to show in full.
dpkg: error processing package samba (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Errors were encountered while processing:
 samba
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Software | G_AGUG

--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Sat  4 May 23:45:30 EST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Setting up samba (2:4.5.16+dfsg-1+deb9u1) ...
Samba is not being run as an AD Domain Controller, masking samba-ad-dc-service.
Please ignore the following error about deb-systemd-helper not finding samba-ad-dc-service.
Failed to preset unit: Unit file /etc/systemd/system/samba-ad-dc.service is masked.
/usr/bin/deb-systemd-helper: error: systemctl preset failed on samba-ad-dc.service: No such file or directory
Job for smbd.service failed because the control process exited with error code.
See "systemctl status smbd.service" and "journalctl -xe" for details.
invoke-rc.d: initscript smbd, action "start" failed.
● smbd.service - Samba SMB Daemon
   Loaded: loaded (/lib/systemd/system/smbd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-05-04 23:44:33 EST; 38ms ago
     Docs: man:smbd(8)
           man:samba(7)
           man:smb.conf(5)
  Process: 14076 ExecStart=/usr/sbin/smbd $SMBDOPTIONS (code=exited, status=1/FAILURE)
 Main PID: 14076 (code=exited, status=1/FAILURE)

May 04 23:44:33 DietPi systemd[1]: Starting Samba SMB Daemon...
May 04 23:44:33 DietPi systemd[1]: smbd.service: Main process exited, code=…LURE
May 04 23:44:33 DietPi systemd[1]: Failed to start Samba SMB Daemon.
May 04 23:44:33 DietPi systemd[1]: smbd.service: Unit entered failed state.
May 04 23:44:33 DietPi systemd[1]: smbd.service: Failed with result 'exit-code'.
Hint: Some lines were ellipsized, use -l to show in full.
dpkg: error processing package samba (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Errors were encountered while processing:
 samba
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

root@DietPi:~#

MichaIng, 2019-05-05:
@ErickPH 
Many thanks for your report. Let me try to replice.

ErickPH, 2019-05-10:
> @ErickPH
> Many thanks for your report. Let me try to replicate.
> 
> Fresh install works fine here.
> Could you please paste the log:
> 
> ```
> cat /var/log/samba/log.smbd
> ```

My logs are cleared. no data is logged as the issue is with (re)installing and not with usage.

MichaIng, 2019-05-10:
@ErickPH 
The logs are most likely cleared due to RAMlog. Since `smbd.service: Main process exited, code=…LURE` there must be some error output of the Samba daemon itself.

Please try to force the error again and then check logs:
```
dpkg --configure samba
cat /var/log/samba/log.smbd
```

MichaIng, 2019-05-26:
I mark this as closed. Feel free to reopen if the issue persists.

closed 2019-05-26 by MichaIng