MichaIng/DietPi #2769: Installing OwnCloud

hamhandler, 2019-05-05:
#### Details:
- Date           | Sun  5 May 19:41:27 BST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | systemctl start redis-server
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Job for redis-server.service failed because the control process exited with error code.
See "systemctl status redis-server.service" and "journalctl -xe" for details.
```

MichaIng, 2019-05-05:
@hamhandler 
Many thanks for your report. Please paste: `journalctl -u redis-server`

hamhandler, 2019-05-05:
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Sun 2019-05-05 20:39:12 BST
. --
May 05 20:28:03 DietPi systemd[1]: [/lib/systemd/system/redis-server.ser
vice:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
May 05 20:28:03 DietPi systemd[1]: Starting Advanced key-value store...
May 05 20:28:04 DietPi run-parts[880]: run-parts: executing /etc/redis/redis-ser
ver.pre-up.d/00_example
May 05 20:28:04 DietPi redis-server[885]: *** FATAL CONFIG FILE ERROR ***
May 05 20:28:04 DietPi redis-server[885]: Reading the configuration file, at lin
e 163
May 05 20:28:04 DietPi redis-server[885]: >>> 'logfile /var/log/redis/redis-serv
er.log'
May 05 20:28:04 DietPi redis-server[885]: Can't open the log file: Permission de
nied
May 05 20:28:04 DietPi systemd[1]: redis-server.service: Control process
 exited, code=exited status=1
May 05 20:28:04 DietPi systemd[1]: Failed to start Advanced key-value st
ore.
May 05 20:28:04 DietPi systemd[1]: redis-server.service: Unit entered fa
iled state.
May 05 20:28:04 DietPi systemd[1]: redis-server.service: Failed with res
ult 'exit-code'.
May 05 20:28:04 DietPi systemd[1]: redis-server.service: Service hold-off time o
ver, scheduling restart.
May 05 20:28:04 DietPi systemd[1]: Stopped Advanced key-value store.
May 05 20:28:04 DietPi systemd[1]: Starting Advanced key-value store...
May 05 20:28:04 DietPi run-parts[891]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
May 05 20:28:05 DietPi systemd[1]: redis-server.service: Control process exited, code=exited status=1
May 05 20:28:05 DietPi systemd[1]: Failed to start Advanced key-value store.
May 05 20:28:05 DietPi systemd[1]: redis-server.service: Unit entered failed state.
May 05 20:28:05 DietPi systemd[1]: redis-server.service: Failed with result 'exit-code'.
May 05 20:28:05 DietPi systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
May 05 20:28:05 DietPi systemd[1]: Stopped Advanced key-value store.
May 05 20:28:05 DietPi systemd[1]: Starting Advanced key-value store...
May 05 20:28:05 DietPi run-parts[912]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
May 05 20:28:05 DietPi systemd[1]: redis-server.service: Control process exited, code=exited status=1
May 05 20:28:05 DietPi systemd[1]: Failed to start Advanced key-value store.
May 05 20:28:05 DietPi systemd[1]: redis-server.service: Unit entered failed state.
May 05 20:28:05 DietPi systemd[1]: redis-server.service: Failed with result 'exit-code'.
May 05 20:28:06 DietPi systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
May 05 20:28:06 DietPi systemd[1]: Stopped Advanced key-value store.
May 05 20:28:06 DietPi systemd[1]: Starting Advanced key-value store...
May 05 20:28:06 DietPi run-parts[1006]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
May 05 20:28:06 DietPi systemd[1]: redis-server.service: Control process exited, code=exited status=1
May 05 20:28:06 DietPi systemd[1]: Failed to start Advanced key-value store.
May 05 20:28:06 DietPi systemd[1]: redis-server.service: Unit entered failed state.
May 05 20:28:06 DietPi systemd[1]: redis-server.service: Failed with result 'exit-code'.
May 05 20:28:06 DietPi systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
May 05 20:28:06 DietPi systemd[1]: Stopped Advanced key-value store.
May 05 20:28:06 DietPi systemd[1]: Starting Advanced key-value store...
May 05 20:28:07 DietPi run-parts[1015]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
May 05 20:28:07 DietPi systemd[1]: redis-server.service: Control process exited, code=exited status=1
May 05 20:28:07 DietPi systemd[1]: Failed to start Advanced key-value store.
May 05 20:28:07 DietPi systemd[1]: redis-server.service: Unit entered failed state.
May 05 20:28:07 DietPi systemd[1]: redis-server.service: Failed with result 'exit-code'.
May 05 20:28:07 DietPi systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
May 05 20:28:07 DietPi systemd[1]: Stopped Advanced key-value store.
May 05 20:28:07 DietPi systemd[1]: redis-server.service: Start request repeated too quickly.
May 05 20:28:07 DietPi systemd[1]: Failed to start Advanced key-value store.
May 05 20:28:07 DietPi systemd[1]: redis-server.service: Unit entered failed state.
May 05 20:28:07 DietPi systemd[1]: redis-server.service: Failed with result 'exit-code'.
May 05 20:30:42 DietPi systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
May 05 20:30:43 DietPi systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
May 05 20:30:43 DietPi systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
May 05 20:30:43 DietPi systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
May 05 20:30:43 DietPi systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
May 05 20:30:43 DietPi systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
May 05 20:31:42 DietPi systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
[... cut ...]
May 05 20:36:19 DietPi systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
May 05 20:36:19 DietPi systemd[1]: Stopped Advanced key-value store.
May 05 20:36:19 DietPi systemd[1]: redis-server.service: Start request repeated too quickly.
May 05 20:36:19 DietPi systemd[1]: Failed to start Advanced key-value store.
May 05 20:36:19 DietPi systemd[1]: redis-server.service: Unit entered failed state.
May 05 20:36:19 DietPi systemd[1]: redis-server.service: Failed with result 'exit-code'.

MichaIng, 2019-05-05:
@hamhandler 
Hmm strange:
`redis-server[885]: Can't open the log file: Permission denied`
- But at this stage if the install have haven't touched any Redis config, log file or anything else at all, just did an APT install/upgrade and start it 🤔. Also works well here.
- However please try:
```
chmod redis:redis /var/log/redis
systemctl restart redis-server
```

Although it is just ignored, still strange:
`[/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'`
- This is simply a typo, it must be `RuntimeDirectory` without capital `T`.
- I just checked fresh install on Debian Stretch and there it is correct. Not sure if Raspbian Stretch ships a buggy package?
  - [ ] Checking Raspbian Stretch redis-server package

hamhandler, 2019-05-05:
Upon entering the first command it returns:
root@DietPi:~# chmod redis:redis /var/log/redis
chmod: invalid mode: ‘redis:redis’
Try 'chmod --help' for more information.

MichaIng, 2019-05-05:
I can't believe it, indeed they ship a package with typo: https://archive.raspbian.org/raspbian/pool/main/r/redis/redis-server_3.2.6-3+deb9u1_armhf.deb

Debian package doesn't have this: https://deb.debian.org/debian/pool/main/r/redis/redis-server_3.2.6-3+deb9u2_armhf.deb

Ah it's one maintainer version lower `deb9u1` vs `deb9u2`. And here is the related changelog: https://metadata.ftp-master.debian.org/changelogs//main/r/redis/redis_3.2.6-3+deb9u2_changelog

Okay that mystery is solved. Due to Raspbian being some days/weeks behind Debian repo. However nothing we need to fix since it does not cause service failure and will fix itself soon on Raspbian as well.

MichaIng, 2019-05-05:
@hamhandler 
> chmod: invalid mode: ‘redis:redis’

Ah whoops, sorry it should be: `chown redis:redis /var/log/redis`

hamhandler, 2019-05-05:
Got it working! Thank you so much for your help!

closed 2019-05-05 by hamhandler

MichaIng, 2019-05-05:
@hamhandler 
Great. Still not sure where the wrong ownership was come from, but glad it works now :).
You can finish ownCloud install via `dietpi-software install 47` if not yet done.