MichaIng/DietPi #2772: Unable to fetch scripts from github/dietpi website on fresh install getting connection error

raidensnake, 2019-05-07:
Unable to download scripts from dietpi or GitHub on fresh build saying connection failed yet apt-get update works.

MichaIng, 2019-05-07:
@raidensnake 
Many thanks for your report.

Could you please run and paste:
```
time wget --spider https://dietpi.com/downloads/binaries/all/cava.psf
time wget --spider https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
```
- It will not download anything, just checking and verifying file existence and report the time required for this.

raidensnake, 2019-05-07:
the response came back as 500 internal server error on both.

MichaIng, 2019-05-07:
@raidensnake 
Both? 🤔

Please paste the full output regardless.

raidensnake, 2019-05-07:
`root@DietPi:~# time wget --spider https://dietpi.com/downloads/binaries/all/cava.ps
Spider mode enabled. Check if remote file exists.
--2019-05-07 13:28:20--  https://dietpi.com/downloads/binaries/all/cava.ps
Resolving dietpi.com (dietpi.com)... 104.27.179.199, 104.27.178.199, 2606:4700:30::681b:b3c7, ...
Connecting to dietpi.com (dietpi.com)|104.27.179.199|:443... connected.
HTTP request sent, awaiting response... 500 Internal Server Error
Failed to get a valid response!!!`

raidensnake, 2019-05-07:
`root@DietPi:~# time wget --spider https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Spider mode enabled. Check if remote file exists.
--2019-05-07 13:28:06--  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 500 Internal Server Error
Failed to get a valid response!!!`

raidensnake, 2019-05-07:
there are the outputs also my shell didn't copy the first part properly.

MichaIng, 2019-05-07:
Hmm since it's two totally dedicated servers it's perhaps related to wget.

Could you try it with replacing `wget --spider` with `curl -IL`?

raidensnake, 2019-05-13:
Sorry the command failed to connect.

MichaIng, 2019-05-13:
@raidensnake 
Do you have any local adblocker, VPN, Pi-hole or something similar installed? I guess opening this link in browser works, right? https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt 

Does downloading/connecting to any other server work? E.g.
```
cd /tmp
wget --spider https://s3-eu-west-1.amazonaws.com/subsonic-public/download/subsonic-6.1.5.deb
wget https://s3-eu-west-1.amazonaws.com/subsonic-public/download/subsonic-6.1.5.deb
```

MichaIng, 2019-05-26:
Ref: https://github.com/MichaIng/DietPi/issues/2834

MichaIng, 2019-06-08:
I mark this as closed, feel free to reopen if issue persists and is not related to Pi-hole blocklist (referenced issue).

closed 2019-06-08 by MichaIng