MichaIng/DietPi #2775: Error in Lighttpd Nextcloud Config file

WilburWalsh, 2019-05-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.22.3
- Distro version | 9.9
- Kernel version | Linux raspi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)

#### Additional Information (if applicable)
- Software title | Nextcloud + Lighttpd

#### Problem

For quite some time I stumbled across this warning in my journal, but ignored it...

> raspi lighttpd[32755]: 2019-05-08 04:28:57: (server.c.1295) WARNING: unknown config-key: dir-listing.active (ignored)

Now, as I did some work on my Nextcloud anyway, I had a closer look. The problem seems to be this line
https://github.com/MichaIng/DietPi/blob/b293ba874d4497679471c882d0a6eba5e135476a/.conf/dps_114/lighttpd.nextcloud.conf#L19

which should read
	`dir-listing.activate = "disable"`

see [here](https://redmine.lighttpd.net/projects/lighttpd/wiki/Docs_ModDirlisting)

MichaIng, 2019-05-08:
@WilburWalsh 
Many thanks for your report. The error is expected. The directive requires a certain enabled Lighttpd module, but we add it regardless.

The issue is if we do not have this directive and end users enable the module for their needs, directory listing inside the Nextcloud install dir is possible which is not desired for security reasons.

Note that it is a warning, not an error, so will be ignored gracefully and does not cause any further issues.

MichaIng, 2019-05-08:
@WilburWalsh 
Ohh, I will verify that, many thanks for reporting! Perhaps it changed during Lighttpd versioning.

MichaIng, 2019-05-08:
Resolved:
- Nextcloud: https://github.com/MichaIng/DietPi/commit/676447c1f4251a19011bb67b6cfae95e9206b98c
- ownCloud: https://github.com/MichaIng/DietPi/commit/3cac9da9d89c84184206bfa91facfb1204d9bfca
- Changelog: https://github.com/MichaIng/DietPi/commit/066a6b502d8d11cc0b252f9f0bd1b34a17f37aec

closed 2019-05-12 by MichaIng