MichaIng/DietPi #2776: Diverging Partition layout across architectures.

FredericGuilbault, 2019-05-08:
RPi have a small boot partition on fat32 and a root partition on ext4 
And X86_64 debian have only one ext4 partition. 

They both make sense in their own context. But this create different layouts for the same Dietpi OS. This seem like an extra source of `platform specific code` to maintain and debug. 

AFAIK sadly the fat32 first partition can"t be avoid on the SDcards.  And IDK about limitations of others platforms. but if it's possible to unite them all under 1 partition layout even if it's not the optimal one for some infrastructure. I vote for this. 

( Im reporting this, but I have no plan to work on this for now. )

MichaIng, 2019-05-09:
@FredericGuilbault
Thanks for your suggestion. 

Indeed it depends on the SBC model if they require bootloader on fat partition or not. The benefit out having this is that one can configure dietpi.txt and kernel/boot options directly from the Windows desktop where the SDcard was flashed.

Since GNU/Linux systems strictly require UNIX permissions support on the root partition, there is no way of uniting this. As well some SBCs require a GPT partition table for EFI boot, so we must stay with different partition layouts to support all of them. However for end users there is nearly no difference, it only requires a more individual image creation process for us, with is most probably why you opened this issue?

FredericGuilbault, 2019-05-09:
> with is most probably why you opened this issue?

Yes, and for the extra work that take to detect the partition layout cuz I can't have Global expectation on how it's set.  And the open door to possible case where a think will break on A and C but not on B and D...  If they could had all the same layout it would have been a KISS way to do.

I was expecting something that would have been the lowest denominator. Like having a FAT32 boot partition on X86_64 althrou it's not optimal simply cuz RPI require it.  But as you said MBR vs GPT make it impossible.

closed 2019-05-12 by MichaIng