MichaIng/DietPi #2778: Unable to install rTorrent via dietpi-software

AndiTails, 2019-05-09:
Looks like the v3.9 of ruTorrent file it tries to pull from GitHub, no longer exists.

```
#### Details:
- Date           | Thu  9 May 11:46:04 BST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (Fourdee/master)
- Img creator    | yumiris
- Pre-image      | DietPi for VMware
- SBC device     | Virtual Machine (x86_64) (index=20)
- Kernel version | #1 SMP Debian 4.9.144-3.1 (2019-02-19)
- Distro         | stretch (index=4)
- Command        | Connection test: https://github.com/Novik/ruTorrent/archive/v3.9.tar.gz
- Exit code      | 4
- Software title | DietPi-Software

#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. Open dietpi-software
2. Choose 'rtorrent' to install
3. Attempt to run Install

#### Expected behaviour:
<!-- What SHOULD be happening? -->
- rTorrent+ruTorrent should install ;)

#### Actual behaviour:
<!-- What IS happening? -->
- Unable to download ruTorrent component from GitHub

#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-05-09 11:43:27--  https://github.com/Novik/ruTorrent/archive/v3.9.tar.gz
Resolving github.com (github.com)... 140.82.118.3
Connecting to github.com (github.com)|140.82.118.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Novik/ruTorrent/tar.gz/v3.9 [following]
Spider mode enabled. Check if remote file exists.
--2019-05-09 11:43:27--  https://codeload.github.com/Novik/ruTorrent/tar.gz/v3.9
Resolving codeload.github.com (codeload.github.com)... 0.0.0.0, ::
Connecting to codeload.github.com (codeload.github.com)|0.0.0.0|:443... failed: Connection refused.
Connecting to codeload.github.com (codeload.github.com)|::|:443... failed: Cannot assign requested address.
Giving up.
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

```

MichaIng, 2019-05-09:
@AndiTails 
Many thanks for your report. I think I found the issue. Please edit /DietPi/dietpi/dietpi-software, find the line `https://github.com/Novik/ruTorrent/archive/$version_string.tar.gz` and add `no_check_url=1 ` in front of the Download_install command.

Not sure why we missed this but these GitHub recources usually throw an error when accessing via wget --spider (URL check) while direct download works well.

MichaIng, 2019-05-10:
Works fine here. I was mistaken, the archive files can be checked via wget --spider, so nothing to be changed:
```
[  OK  ] DietPi-Software | Connection test: https://github.com/Novik/ruTorrent/archive/v3.9.tar.gz
[ INFO ] DietPi-Software | G_THREAD_START_0 | wget https://github.com/Novik/ruTorrent/archive/v3.9.tar.gz -O 107.tar
```

Perhaps it was just a temporary server issue?

closed 2019-05-12 by AndiTails