MichaIng/DietPi #2780: Minor bug - TLS errors when trying to update firmware prior to first boot

stephantual, 2019-05-09:
Latest everything on rpi3b+
If you try to run 'firmware upgrade' from the main menu (which i beleive is just apt-get dist upgrade) - it will fail with a 'tls not terminated properly'

This is easily fixed by running the 'first boot' (not sure how you call this), in other words the part where it asks you if you want to be part of the survey etc... I imagine it has something to do with diet pi running an apt-update and the pre-patch before running the wrapper. 

Minor issue but likely confusing if you're new.

MichaIng, 2019-05-10:
@stephantual 
Many thanks for your report.

Possibly the APT error is due to system clock not yet being synced with time server. This is assured during first run setup. Why do you want to do a firmware upgrade before finishing first run setup? Usually this is not intended and it requires to exit the login scripts manually, right?

You are right, firmware update on RPi simply runs `apt-get dist-upgrade` (with our error handled + non-interactive wrapper: `G_AGDUG`). Actually on RPi there is no point in doing this, since the firmware packages are (differently compared to x86 and most other SBC firmware repos) is no meta-package. This means a simply `apt-get upgrade` will update firmware already, which is automatically done on first run update and software installs.

So actually to avoid confusion we should remove the menu option on RPi. It is mainly a relict from older times where we installed an ran `rpi-update`. But this is nowadays obsolete and even dangerous since it installes in-dev testing kernel versions. The APT package is updated regularly to stable versions which is the desired firmware.

MichaIng, 2019-05-10:
Done: https://github.com/MichaIng/DietPi/commit/239b6e8d7035ab1e969ea6ee88dcbd69f78a18a6

closed 2019-05-12 by MichaIng