MichaIng/DietPi #2781: Unable to reach raspbian servers

linleshi, 2019-05-10:
#### Details:
- Date           | Fri 10 May 10:59:00 BST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | Connection test: http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian
- Exit code      | 4
- Software title | DietPi-Software

#### Steps to reproduce:

1. sudo apt update freezes.
2. If you change mirror, it changes but nothing happens.

#### Expected behaviour:

- I should be going straight to the mirror and getting a connection.
#### Actual behaviour:

- Was working ok yesterday.

linleshi, 2019-05-10:
I think this is a local issue. Am not able to tunnel to the raspi using Dataplicity and the DIetpi is not showing up on JuiceSSH  via the internet. Secure Shell and JuiceSSH work locally. Have been trying to get WireGuard to work but am being thwarted by not being able to make port 51820 available. The only ports that I have made available recently have been 22, 80 and 443. My ISP is Talktalk. Maybe that's the problem. In the past running full Raspbian Stretch I was able to port forward 5900/5901 and get RealVNC working.

MichaIng, 2019-05-10:
@linleshi 
Many thanks for your report.

You can change the WireGuard port to one that your router/firewall/ISP is able to forward. Do you use WireGuard as server or client? In case of client, the default gateway is usually changed to the configured port and if this is closed, no traffic is possible. As server, outgoing traffic should be not affected.

linleshi, 2019-05-10:
Thanks for the WireGuard port advice.

Later today will revert to my second Dietpi microSD and see how it goes.

Am not sure what caused this but have been experimenting with iptables
rules and may have done something wrong.

Have removed iptables and iptables-persistent thinking that that might
solve the problem but after reboot no change.

Will probably start from scratch with the affected microSD if all is well
with the other one.

linleshi, 2019-05-13:
#### Details:
- Date           | Mon 13 May 09:39:32 BST 2019
- Bug report     | N/A
- DietPi version | v6.23.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | Connection test: http://raspbian.raspberrypi.org/raspbian
- Exit code      | 4
- Software title | DietPi-Software

#### Steps to reproduce:

1. Updated via dietpi-update this morning... no issues
2. Have the RASPI running as a Nextcloud and WireGuard server
3. Decided to also configure it as a media server
4.Loaded Subsonic, but didn't configure a file server before the Subsonic load

#### Expected behaviour:

-Expected everything to work

#### Actual behaviour:

- Can't access raspbian server
#### Extra details:

- Maybe caused by my failure to provide a Dietpi file server before Subsonic load

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-05-13 09:38:25--  http://raspbian.raspberrypi.org/raspbian
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... failed: Connection timed out.
wget: unable to resolve host address ‘raspbian.raspberrypi.org’
```

MichaIng, 2019-05-13:
@linleshi 
If indeed just the timeout is too short, you can now raise it via: `dietpi-config` > Network Options: Misc > G_CHECK_URL_Timeout

linleshi, 2019-05-13:
Ok. Will try that before anything more dramatic. Thanks for the speedy
response. Having done personal long-winded Raspi App loads over the last
three years, I am very impressed with the ease of use of Dietpi. It's a
wonderful system.

iondarie, 2019-05-22:
any update on this issue ? i have the same problem it is hanging..

+ apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
0% [Working]  

It has been like this for the last 10 minutes.

MichaIng, 2019-05-22:
@iondarie 
Do you have IPv6 enabled? Please try to disable it: `dietpi-config` > Network Options: Adapters

If that does not work, please try to switch the APT mirror: `dietpi-config` > Network Options: Misc

MichaIng, 2019-06-08:
@iondarie 
Another guess, since I experienced hanging network for ~10 minutes after boot on my kernel 4.19 Buster VM due to low entropy. Some kernel version seems to have changed behaviour to hold network until entropy pool is fully initialised: https://github.com/MichaIng/DietPi/issues/2806#issuecomment-498457875

After ~10 minutes APT update goes on and afterwards has no issue anymore.

Since RPi just got kernel 4.19 as well, perhaps this matches in your case. If so, please try to resolve via: `G_AGI haveged`
And if this indeed solves the issue, if you don't mind, please send us a `dietpi-bugreport` (and paste the ID here) since I want to investigate in which cases systems are affected by too low entropy on boot.

closed 2019-06-25 by MichaIng