MichaIng/DietPi #2782: Infinite loop

FredericGuilbault, 2019-05-10:
### Creating a bug report/issue

#### Required Information
RPI

#### Steps to reproduce
On the first boot, If I log as dietpi user instead of root (for debugging purpus )
Im stuck in a endless loop telling me that I need root access. 

[image]

#### Expected behaviour

It's ok to warn, but it should drop to prompt.

MichaIng, 2019-05-10:
@FredericGuilbault 
Many thanks for your report.

Indeed it is expected to login as root on first run. However logging is as different user should as you say only lead to a prompt. AFAIK I already fixed that for v6.23, let me check the made changes.

MichaIng, 2019-05-10:
Hmm, this should have been fixed already with v6.22, there is a whiptail prompt (when automated first run setup chosen, then a info message) and the script exits: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-login#L139

Is the RPi image on a lower version? In case I will update it after v6.23 release.

FredericGuilbault, 2019-05-10:
oh you are right it's v6.21  but it's a fresh install and I required the `beta` branch in `dietpi.txt` 

Im confused, ill dig that.

MichaIng, 2019-05-10:
@FredericGuilbault 
Since the login script starts the initial dietpi-update (including Git branch changes) the issue cannot be fixed without updating the image, or what you mean?

Fourdee, 2019-05-12:
@MichaIng 

Just a note, our online docs reference using ```root``` and ```dietpi``` for login. https://dietpi.com/phpbb/viewtopic.php?f=8&t=9 dietpi.com/download

As Micha already mentioned, we'd need to update all images to v6.23~ before we can change docs to ```dietpi``` user login.

Not critical in my eyes as we want the initial setup to be fluid for end user, and, free from requiring ```sudo```, however, would be nice to have at some point in the future.

MichaIng, 2019-05-12:
@Fourdee 
Jep indeed not critical. However the loop only occurs after dietpi-login rework (v6.21) and before v6.22, where we solved this with an exit prompt. Earlier images should have a different error message without loop. I will anyway update the RPi image and want to provide one on Buster basis.

FredericGuilbault, 2019-05-12:
Idk why, my brain had a false memory that dietpi codebase was re-downloaded from the internet prior to start the first run process. But looking at the code it's not the case at all.

closed 2019-05-12 by FredericGuilbault

MichaIng, 2019-05-14:
Updated RPi image ready for testing: https://dietpi.com/downloads/testing/DietPi_RPi-ARMv6-Stretch.7z
Buster image is following...