MichaIng/DietPi #2783: Dietpi-Software | UrBackup Server Update to 2.3.8

DeathIsUnknown, 2019-05-10:
There is a new version of UrBackup server available (2.3.8). Download it [here](https://hndl.urbackup.org/Server/2.3.8/)

Might it be possible to use the `latest` endpoint, rather than manually specifying versions? e.g. `https://hndl.urbackup.org/Server/latest/urbackup-server_X.X.X_ARCH.deb`. There is probably good reason to manually specify versions that I'm unaware of.

Required Information
DietPi version | 6.22.3

MichaIng, 2019-05-10:
@DeathIsUnknown 
Many thanks for the info.

I tried to find a "latest" download link but couldn't find it until now. This is probably something to address to the UrBackup devs which also allows them to update without having to change their websites download links.

MichaIng, 2019-05-10:
Hmm there is actually the dir https://hndl.urbackup.org/Server/latest/ but the files inside still contain version strings. But probably we can scrape the version string from there. I have to run some tests about this.

MichaIng, 2019-05-10:
I hijacked the other PR to add the new UrBackup version: https://github.com/MichaIng/DietPi/pull/2779
For v6.24 we will find a way to get the latest version string automatically.

DeathIsUnknown, 2019-06-04:
Just tried to get this working on my end. Here's what I've come up with so far: `wget -r -l1 -np -nH -A 'urbackup-server_*_armhf.deb' https://hndl.urbackup.org/Server/latest/`

MichaIng, 2019-06-04:
@DeathIsUnknown 
Quicker and cleaner method:
```
file=$(curl -s https://hndl.urbackup.org/Server/latest/ 2> /dev/null | grep -m1 'urbackup-server_2.3.8_amd64.deb' | cut -d \" -f 8)
wget "https://hndl.urbackup.org/Server/latest/$file"
```

DeathIsUnknown, 2019-06-04:
@MichaIng Love it!

MichaIng, 2019-06-04:
Done: https://github.com/MichaIng/DietPi/commit/220cf136e0e42f2e5cf425c17be907ee93cac043

closed 2019-07-03 by MichaIng