MichaIng/DietPi #2791: Seperate first boot configurations from reguliar boot procedures.

FredericGuilbault, 2019-05-12:
It's just to let you know that im working on this. 

https://github.com/FredericGuilbault/DietPi/tree/improvement/Seperate_first_boot

MichaIng, 2019-05-12:
@FredericGuilbault 
I was already thinking about the same. It should be even possible to move first run update and installs (when AUTO_SETUP_AUTOMATED=1 chosen) all into a systemd unit so that it runs before first login: https://github.com/MichaIng/DietPi/issues/2520

It's just the following things that need to be taken care off:
- Everything that does not need network and network pre-configuration (`/etc/network/interface`, applying chosen settings from `dietpi.txt`) needs to be done before the network interfaces are brought up (`Before=network-pre.target`).
- Then bringing network interfaces up including the workarounds for some devices, especially on Jessie where it is not 100% reliable (`After=network.target`). Waiting for/reading back online status and running time sync etc. `DietPi-Update` could then be run automatically + the first run steps that are currently part of `DietPi-Software` or even chosen automated software installs.

FredericGuilbault, 2019-05-14:
In `dietpi/boot` line 62 
```
#----------------------------------------------------------------
		# WiFi Country | Additional fallback for (older kernel?) devices that fail with wpa_supplicant.conf https://github.com/MichaIng/DietPi/issues/838
		command -v iw &> /dev/null && iw reg set "$(grep -m1 '^[[:blank:]]*CONFIG_WIFI_COUNTRY_CODE=' /DietPi/dietpi.txt | sed 's/^[^=]*=//')" &
		#----------------------------------------------------------------

```
IDK much about wifi,Does it really need to check if the country code exist at each boot ?

MichaIng, 2019-05-14:
@FredericGuilbault 
As you can read from the link, it sets the country code as some devices (SBC/kernel/network combination) does not seem to apply the related setting from `wpa_supplicant.conf` correctly.

We could recheck this with the devices reported. Since it was an RPi, perhaps one of the many kernel/firmware updates meanwhile fixed this.

The same applies for some other WiFi workarounds (ifup/ifdown). The networking service (re)start is currently indeed required on Jessie systems, but the manual ifup/ifdown could be rechecked.

FredericGuilbault, 2019-05-16:
Does it have to apply the settings over and over at each reboot ?

MichaIng, 2019-05-18:
@FredericGuilbault 
Which settings you mean? Those not inside the first run functions yes, but some review would not hurt.

Fourdee, 2019-05-20:
@FredericGuilbault 

> Does it have to apply the settings over and over at each reboot ?

Yes, as not all WiFi chipsets support the options set in ```wpa_supplicant.conf```, last time I checked.

MichaIng, 2019-06-04:
Related PR: https://github.com/MichaIng/DietPi/pull/2833

closed 2019-07-03 by MichaIng