MichaIng/DietPi #2792: Update issue rock64 6.22 to 6.23

josemahj, 2019-05-13:
#### Details:
- Date           | Mon 13 May 08:58:19 CEST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | ARMbian
- SBC device     | Rock64 (aarch64) (index=43)
- Kernel version | #4 SMP Wed May 8 19:34:01 CEST 2019
- Distro         | stretch (index=4)
- Command        | unzip update.zip
- Exit code      | 9
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Archive:  update.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of update.zip or
        update.zip.zip, and cannot find update.zip.ZIP, period.
```

MichaIng, 2019-05-13:
@josemahj 
Many thanks for your report.

Hmm that is strange since it works well here. Perhaps it was a corrupted download but could you try to extract the archive manually first:
```
cd /tmp
wget https://github.com/MichaIng/DietPi/archive/master.zip
unzip master.zip
# If that works, then remove the test files and retry die update
rm -R master.zip DietPi-master
dietpi-update 1
```

Fourdee, 2019-05-13:
@MichaIng @josemahj 

Could also point to a underlying system/filesystem instability/corruption, paste ```dmesg | grep ext```, we'll check for any errors.

MichaIng, 2019-05-13:
Good point. When downloading/extracting in /tmp (RAM) works but on disk not, this would be also a strong indication for disk issue/file system corruption.

josemahj, 2019-05-13:
> @MichaIng @josemahj
> 
> Could also point to a underlying system/filesystem instability/corruption, paste `dmesg | grep ext`, we'll check for any errors.

```
root@DietPi:/tmp# dmesg | grep ext
[    0.000000] Kernel command line: earlycon=uart8250,mmio32,0xff130000 swiotlb=1 kpti=0 root=UUID=37725f38-d37c-429c-89a6-9fc49158dcfe rootwait rootfstype=ext4 console=ttyS2,1500000 console=tty1 panic=10 consoleblank=0 loglevel=7 ubootpart=b275bf8f-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
                     .text : 0xffffff8008080000 - 0xffffff8008bd0000   ( 11584 KB)
[    3.632886] BTRFS: selftest: Running extent only tests
[    3.632925] BTRFS: selftest: Running bitmap and extent tests
[    3.632954] BTRFS: selftest: Running space stealing from bitmap to extent
[    3.634403] BTRFS: selftest: Running extent buffer operation tests
[    3.850866] BTRFS: selftest: Running btrfs_get_extent tests
[    3.854681] BTRFS: selftest: Running hole first btrfs_get_extent test
[    3.858385] BTRFS: selftest: Running outstanding_extents tests
[    7.185844] Adding 524284k swap on /var/swap.  Priority:-1 extents:15 across:1114112k SS`
```

josemahj, 2019-05-13:
> @josemahj
> Many thanks for your report.
>
> Hmm that is strange since it works well here. Perhaps it was a corrupted download but could you try to extract the archive manually first:
>
> ```
> cd /tmp
> wget [...]

this works..... rock64 updated to 6.23.3

MichaIng, 2019-05-13:
@josemahj 
Okay great. Not sure if you hit the 1 of 1 million cases where a download is corrupt due to transfer issues, but just in case something similar happens again, keep in mind to check `dmesg` as well since file system corruption (at least on SDcards) is the much more common problem and also the one that needs fast reaction to avoid further data loss. `dmesg` should show such I/O errors in red colour so they cannot be overseen 😉.

closed 2019-05-13 by MichaIng

elRadix, 2019-05-13:
I had the same issue, in my case this was a DNS issue - I had to turn off adblocker at router level - I'm using https://diversion.ch/diversion/diversion.html

MichaIng, 2019-05-13:
Strange that an adblocker blocks github.com. In case add github.com and dietpi.com to the whitelist as these are the only addresses (aside from software installs) that DietPi connects to. Ah and APT mirrors  of course.