MichaIng/DietPi #2793: DietPi-Software | Transmission: Web UI settings changes not boot persistent

MichaIng, 2019-05-13:
Ref: https://dietpi.com/phpbb/viewtopic.php?p=17927#p17927

Trying to replicate...

depach, 2019-05-13:
to get this working I had to make `debian-transmission` user the owner of `/etc/transmission-daemon`

before it was `root:debian-transmission`

weird thing is the directory had group write permissions but I still got on error about writing to a temp config file in `/etc/transmission-daemon` on a transmission restart 

`May 13 17:28:44 DietPi transmission-daemon[1570]: [2019-05-13 17:28:44.159] Couldn't save temporary file "/etc/transmission-daemon/settings.json.tmp.5MH5fk": Permission denied (variant.c:1280)`

MichaIng, 2019-05-13:
@depach 
Thanks for reporting.

Hmm strange that it is a permissions issue since we don't change anything about that from the APT package install.

We add running the service as group "dietpi" via: `/etc/systemd/system/transmission-daemon.service.d/dietpi-group.conf`
However that should only add "dietpi" group permissions, not limit the existing ones. But I will verify that. Would be at least an `explanation` if that overrides the "debian-transmission" group permissions.

MichaIng, 2019-05-13:
Just checked the postinst script and the permissions are as intended:
```
case "$1" in
  configure)
    if ! getent passwd debian-transmission > /dev/null; then
        adduser --quiet \
                --system \
                --group \
                --no-create-home \
                --disabled-password \
                --home /var/lib/transmission-daemon \
                debian-transmission
    fi

    chperms debian-transmission debian-transmission 4750 /var/lib/transmission-daemon/.config/transmission-daemon

    chperms debian-transmission debian-transmission 4775 /var/lib/transmission-daemon/downloads

    chperms debian-transmission debian-transmission 0660 /etc/transmission-daemon/settings.json

    chperms root debian-transmission 2775 /etc/transmission-daemon

  ;;
esac
```

Hmm I hope the dietpi group is not the issue since there is no other way to create downloads as dietpi group. And this is true for many software installs.
However running `chown debian-transmission /etc/transmission-daemon` is not hard fix :wink:.

MichaIng, 2019-06-16:
Okay applying `dietpi` group via drop-in config is indeed the issue. Strange it that I tested this already long before and now again: Using Group= in systemd units does definitely NOT override system group memberships/permissions of the run user. So it must be Transmission internal check/issue.

MichaIng, 2019-06-16:
Mysterious:
```
root@VM-Stretch:/tmp# systemctl cat test
# /etc/systemd/system/test.service
[Service]
Type=notify
User=www-data
Group=dietpi
ExecStart=/bin/dash -c 'echo success >> /tmp/dir/test'
root@VM-Stretch:/tmp# l -a dir/
total 0
drwxrwxr-x 2 root www-data  60 Jun 17 01:06 .
drwxrwxrwt 9 root root     200 Jun 17 01:25 ..
-rw-rw-r-- 1 root www-data   0 Jun 17 01:19 test
root@VM-Stretch:/tmp# systemctl start test
root@VM-Stretch:/tmp# cat /tmp/dir/test
success
root@VM-Stretch:/tmp# > /tmp/dir/test
root@VM-Stretch:/tmp# chown -R :debian-transmission dir
root@VM-Stretch:/tmp# sed -i 's/www-data/debian-transmission/' /etc/systemd/system/test.service
root@VM-Stretch:/tmp# systemctl daemon-reload
root@VM-Stretch:/tmp# systemctl start test
Job for test.service failed because the control process exited with error code.
See "systemctl status test.service" and "journalctl -xe" for details.
root@VM-Stretch:/tmp# cat /tmp/dir/test
```
- The same that works with any other user, does not work with `debian-transmission`

Where the hack is the difference?...

MichaIng, 2019-06-17:
Resolved: https://github.com/MichaIng/DietPi/pull/2918

closed 2019-07-03 by MichaIng