MichaIng/DietPi #2794: Update to v6.23 seems to have bricked my installation...

maartenlangeveld, 2019-05-13:
Update to v6.23 seems to have completely bricked my installation...

Update halted with error. Got a message that advised to run 'apt --fix-broken install' but that won't work.:

apt --fix-broken install
Reading package lists... Done
Building dependency tree
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  php7.0-cli php7.0-common php7.0-curl php7.0-fpm
The following packages will be upgraded:
  php7.0-cli php7.0-common php7.0-curl php7.0-fpm
4 upgraded, 0 newly installed, 0 to remove and 12 not upgraded.
12 not fully installed or removed.
Need to get 0 B/2,723 kB of archives.
After this operation, 1,370 kB disk space will be freed.
Do you want to continue? [Y/n] y
E: Sub-process /usr/bin/dpkg exited unexpectedly

Machine won't reboot:
root@DietPi:~# reboot
Illegal instruction

So what to do now?

#### Details:
- Date           | Mon 13 May 18:31:18 CEST 2019
- Bug report     | N/A
- DietPi version | v6.23.3 (MichaIng/master)
- Img creator    |
- Pre-image      |
- SBC device     |  (index=)
- Kernel version | #1200 Tue Feb 12 20:11:02 GMT 2019
- Distro         | NULL (index=)
- Command        | G_AGI rsync
- Exit code      | 100
- Software title | DietPi-Backup

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

DietPi, 2019-05-13:
Details:

    Date | Mon 13 May 18:31:18 CEST 2019
    Bug report | N/A
    DietPi version | v6.23.3 (MichaIng/master)
    Img creator |
    Pre-image |
    SBC device | RPI B (armv61) (index=1)
    Kernel version | #1200 Tue Feb 12 20:11:02 GMT 2019
    Distro | stretch (index=4)
    Command | G_AGUG
    Exit code | 100
    Software title | DietPi-Software

Steps to reproduce:

 Regular install. System hangs with errors.

Expected behaviour:

Full install with pi-hole support.

Actual behaviour:

 System hangs, sometimes disabling the network adapter. I tried with a raspberry pi zero W through the wireless connection and a raspberrypi B through ethernet. Same behaviour. 

Extra details:
This is my first time using DietPi. I have installed it yesterday on the pi zero, and it worked without a hitch. It was a version 6.22.?  Then I tried to update to the latest version, and it broke. Now I can't get it to work at all.

Additional logs:

Log file contents: 
E: Sub-process /usr/bin/dpkg exited unexpectedly

Fourdee, 2019-05-13:
@maartenlangeveld 

> Illegal instruction

Thanks for the report, appears your system is experiencing possible stability issues.

Please can you reply with the following:
- SBC device used (eg: RPI 2)
- Power supply used (+brand volts, amps)
- SD/eMMC used (+ brand)
- Please paste the full output of dmesg ```dmesg```

Fourdee, 2019-05-13:
@DietPi 

> System hangs with errors.

Might be a different issue, does the issue occur only when installing software with ```dietpi-software```?

Try running ```G_AGF```, please paste output of any errors if it occurs.

> System hangs, sometimes disabling the network adapter. 

Could also be related to a stability issue, and/or networking (bad WiFi signal). Please reply with following:
- ```dietpi-config``` > networking adapters > WiFi > please paste signal strength
- Power supply used (+brand volts, amps)
- SD/eMMC used (+ brand)
- Please paste the full output of dmesg dmesg
- Any manual changes to Linux files/configs, outside of DietPi scripts?
- Any manual software installations, outside of dietpi-software?

maartenlangeveld, 2019-05-13:
Strange... Have never ever experienced any stability issues with my setup. Running for about half year....

But here you are:

SBC: RPi Zero W
Powersupply: Samsung 5.0V 2.0A
SD: SanDisk Ultra Micro SD 400 GB

Cannot paste full ouput of dmseg since my installation won't boot anymore...
dietpi-config > networking adapters > WiFi > please paste signal strength: idem (remember) that wifi adapter had disappeared from config while I had SSH terminal open (over wifi network)....

Any manual changes to Linux files/configs, outside of DietPi scripts? --> No
Any manual software installations, outside of dietpi-software? --> Yes : see below

- upmpdcli
- minimserver

MichaIng, 2019-05-13:
@maartenlangeveld 
Seems that either something depends on PHP7.0 packages or those were not fully purged.

Could you please paste the output of: `apt purge *php7.0*`

Before confirming with yes, check with other packages are about to be removed.

Btw you installed PHP (e.g. a webserver stack) via dietpi-software, right? Just to assure that indeed the PHP7.3 migration caused the issue and the PHP7.3 repository + packages have been installed already. To check: `dpkg -l | grep '^php'`

maartenlangeveld, 2019-05-13:
Hi Micha.

I cannot since I cannot boot machine anymore. Tried to reboot machine by pulling the plug but won't reboot anymore. As far as I remember I have installed PHP via dietpi-software but I cannot guarantee.

Thanks for your concerns !!!

DietPi, 2019-05-13:
> 
> 
> @DietPi
> 
> > System hangs with errors.
> 
> Might be a different issue, does the issue occur only when installing software with `dietpi-software`?
Yes. I can install dietpi without any additional software. As soon as I try to install anything else, it hangs.

> Try running `G_AGF`, please paste output of any errors if it occurs.
 I get the DietPi Error Handler window. G_AGF
- Exit code: 100
- DietPi version (as above).
Log file contents:
E: Sub-process /usr/bin/dpkg exited unexpectedly
Unable to continue, will now terminate. 
Retry: Re-run last command
Diet-Pi config: Edit network, APT/NPT etc
> > System hangs, sometimes disabling the network adapter.
> 
> Could also be related to a stability issue, and/or networking (bad WiFi signal). Please reply with following:
> 
>     * `dietpi-config` > networking adapters > WiFi > please paste signal strength

Right now I have it only on the RPi B, it has no wifi. It says Ethernet: Not Found | [On] | Disconnected.
> 
>     * Power supply used (+brand volts, amps)
Anidees 5V 3A, used with 16 gauge 20 cm usb wire. This is RPI B, so it should be enough.

>     * SD/eMMC used (+ brand)
SD is Kingston 16GB.
>     * Please paste the full output of dmesg dmesg
I'll try to find how to do that, complete noob here.
>     * Any manual changes to Linux files/configs, outside of DietPi scripts?
No.
> 
>     * Any manual software installations, outside of dietpi-software?
No

DietPi, 2019-05-13:
Shutdown didn't work, Illegal command. Now it doesn't start any more, hangs in Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000004. ramdom: crng init done. I checked var/logs for dmesg file, but there were no files there. All I can do now is to redo the image and start again, but I've done that at least 5 times now. It ends up here all the time.

MichaIng, 2019-05-13:
@maartenlangeveld @DietPi 
Ah sorry I missed the illegal instructions part on reboot. Yeah then forget about the APT (PHP) error since this is most likely just a symptom of the underlying issue.

My guess goes to SDcard corruption then. The update includes a raspberrypi-firmware/kernel update (last dated April while our images is dated March) and this writes a bunch of files on the SDcard and moves then onto the /boot partition. Most likely a bad block is written to which then corrupts the related file, hence boot fails on kernel level (systemd was not yet initiated).

One chance is to mount the SDcard onto another Linux system and run `fsck` on both partitions. But in case of a bad block the error will just reappear on any other random file write. So at best copy all files from SDcard to a backup drive and write a fresh image onto a new SDcard.

> I checked var/logs for dmesg file, but there were no files there.

`dmesg` is a command, not a log file. You can run it on the running system to check kernel logs, but since it is not boot persistent sadly not chance to read those from an external system.
To read kernel/system logs from an external system it is required to disable DietPi-RAMlog and make the journald logs persistent: `mkdir /var/log/journal`
The included files are no text file, but you can read that from an external system with the `journalctl` command: `journalctl --file /mnt/<mount_point>/var/log/journal/<random_string>/system.journal`

maartenlangeveld, 2019-05-13:
Hi Micha,

Really appreciate your fantastic help !! Many many thanks !

Seems I simply have bad luck. Will do a fresh install.

Kind regards,

Maarten

maartenlangeveld, 2019-05-13:
Hi Micha,

Fresh install. get following error:

 ┌───────────────────────────────┤ DietPi Error Handler: ├────────────────────────────────┐
                                   │ DietPi-Software: G_AGI php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring
                                   │ php7.3-xml php7.3-zip php7.3-mysql
                                   │  - Exit code: 100
                                   │  - DietPi version: v6.23.3 (MichaIng/master) | HW_MODEL:1 | HW_ARCH:1 | DISTRO:4
                                   │  - Image creator: DietPi Core Team
                                   │  - Pre-image: Raspbian Lite
                                   │
                                   │ Log file contents:
                                   │ Setting up php7.3-fpm (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b) ...^M
                                   │ Job for php7.3-fpm.service failed because a fatal signal was delivered to the control
                                   │ process.^M
                                   │ See "systemctl status php7.3-fpm.service" and "journalctl -xe" for details.^M
                                   │ invoke-rc.d: initscript php7.3-fpm, action "start" failed.^M
                                   │ ^[[0;1;31m●^[[0m php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager^M
                                   │    Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; enabled; vendor preset:
                                   │
                                   │                Retry         : Re-run the last command that failed
                                   │                DietPi-Config : Edit network, APT/NTP mirror settings etc
                                   │
                                   │
                                   │                        <Ok>                            <Exit>
                                   │

maartenlangeveld, 2019-05-13:
Hi Micha,

Additional problem, the dietpi with installed software gets corrupted, many packages installed with dietpi-software are not listed in dietpi-software uninstall menu as process was interrupted when setting up php7.3-fpm, see above.

Will have to do fresh install again...

Are you sure to roll out v6.23 ?

MichaIng, 2019-05-13:
@maartenlangeveld 
First of all no need to re-flash just because `dietpi-software` run fails. Since software titles are marked as installed after everything has finished, it is a known issue that the ones already processed cannot be uninstalled via the menu. In such case at best simply install the software titles again, after fixing the occurred issue of course.

> Maybe the problemn is rpi-zero-w specific?

The PHP install part should not be RPi Zero specific since at that level all RPi should behave exactly the same.

> Will have to do fresh install again... Since same problem again.

You mean again the "illegal instruction" on reboot command and kernel error on boot? Yeah then most likely a hardware failure since the PHP package cannot cause such an issue.
___________
Okay so to debug, after DietPi first run update has finished, do not mark any software for install, but continue with minimal image. Reboot one or two times to verify that is working fine so far.

Lets try then to manually install PHP7.3 and please paste the output:
```
curl -sSL https://packages.sury.org/php/apt.gpg | apt-key add -
echo 'deb https://packages.sury.org/php/ stretch main' > /etc/apt/sources.list.d/dietpi-php.list
apt update
apt install php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip
```

maartenlangeveld, 2019-05-13:
Dit not get " "illegal instruction" this time but could not re-install software because everytime apt is started, it tries to repair php7.3-fpm which resulted evertytime in the same error. So i got stuck.

MichaIng, 2019-05-13:
@maartenlangeveld 
Ah okay.

Then please paste: `journalctl -u php7.3-fpm`
In case there should be a log file as well: cat `/var/log/php-fpm.log` (or similar)

maartenlangeveld, 2019-05-13:
I already did reinstall of image. So cannot paste.

MichaIng, 2019-05-13:
@maartenlangeveld 
Okay, then go with the above manual PHP7.3 install steps first.

maartenlangeveld, 2019-05-13:
Hi Micha,

Kernel panic... after apt update and upgrade have to do refresh again...

I'm giving up, I am very sorry. Will revert to v6.22 and not upgrade for a while...

Thanks, 

Maarten

DietPi, 2019-05-13:
> Kernel panic... after i followed your instructions and issued apt update followed by apt upgrade just before "apt install php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip"... have to do refresh reinstall again...

That's very similar to what I'm experiencing.

> I'll give it one more try, if that does not work I will revert to v6.22 and not upgrade for a while...

I was planning to do the same, but I wasn't able to stop the upgrade to v6.23 from the newly flashed sd-card.

MichaIng, 2019-05-14:
@DietPi 
Pressing <ctrl>+<c> will cancel any running process.

@maartenlangeveld 
Kernel panic (system crash) occurred while installing the PHP packages? Any error messages before that? Any chance to paste the last output?

I still don't believe that APT failure and reboot/boot/kernel error are related. Perhaps the I/O or CPU load triggers the hardware issue or such.

Did you follow my advice above to not install any software first (finish initial dietpi-software without marking software, as minimal image) and do 1-2 reboots? Also check `dmesg` output as advised by Fourdee to check for any I/O or voltage errors.
______________
I need to sleep now but will run tests on my RPi tomorrow.

maartenlangeveld, 2019-05-14:
Hi Micha,

Followed your instructions (manual install php7.3), exact same things went wrong when (1) updating v6.22 (installed base) to v6.23  and (2) fresh install of image (v6.21, which auto updates to v6,23) and then installing sofware (including php) through config-software menu (repeated two times).

see below for output, hope it will help you find the root cause and solve the problem. I will now reinstall backup (v6.22) since it serves no issue i think to repeat it for a fourth time.

root@DietPi:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
root@DietPi:~# curl -sSL https://packages.sury.org/php/apt.gpg | apt-key add -
OK
root@DietPi:~# echo 'deb https://packages.sury.org/php/ stretch main' > /etc/apt/sources.list.d/dietpi-php.list
root@DietPi:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
root@DietPi:~# apt install php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-apcu' instead of 'php7.3-apcu'
The following additional packages will be installed:
  fontconfig-config fonts-dejavu-core libargon2-1 libbsd0 libedit2 libexpat1 libfontconfig1 libfreetype6 libgd3 libicu57 libjbig0 libjpeg62-turbo libmagic-mgc
  libmagic1 libpcre2-8-0 libpng16-16 libsodium23 libtiff5 libwebp6 libx11-6 libx11-data libxau6 libxcb1 libxdmcp6 libxml2 libxpm4 libxslt1.1 libzip4
  mime-support php-common php7.3-cli php7.3-common php7.3-json php7.3-opcache php7.3-readline ucf
Suggested packages:
  libgd-tools file php-pear
Recommended packages:
  xml-core file php-apcu-bc
The following NEW packages will be installed:
  fontconfig-config fonts-dejavu-core libargon2-1 libbsd0 libedit2 libexpat1 libfontconfig1 libfreetype6 libgd3 libicu57 libjbig0 libjpeg62-turbo libmagic-mgc
  libmagic1 libpcre2-8-0 libpng16-16 libsodium23 libtiff5 libwebp6 libx11-6 libx11-data libxau6 libxcb1 libxdmcp6 libxml2 libxpm4 libxslt1.1 libzip4
  mime-support php-apcu php-common php7.3-cli php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json php7.3-mbstring php7.3-opcache php7.3-readline
  php7.3-xml php7.3-zip ucf
0 upgraded, 43 newly installed, 0 to remove and 3 not upgraded.
Need to get 17.5 MB of archives.
Setting up php7.3-zip (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b) ...

Creating config file /etc/php/7.3/mods-available/zip.ini with new version
Setting up php7.3-mbstring (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b) ...

Creating config file /etc/php/7.3/mods-available/mbstring.ini with new version
Setting up php7.3-json (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b) ...

Creating config file /etc/php/7.3/mods-available/json.ini with new version
Setting up libx11-6:armhf (2:1.6.4-3+deb9u1) ...
Setting up php7.3-cli (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b) ...
update-alternatives: using /usr/bin/php7.3 to provide /usr/bin/php (php) in auto mode
update-alternatives: using /usr/bin/phar7.3 to provide /usr/bin/phar (phar) in auto mode
update-alternatives: using /usr/bin/phar.phar7.3 to provide /usr/bin/phar.phar (phar.phar) in auto mode

Creating config file /etc/php/7.3/cli/php.ini with new version
Setting up libxpm4:armhf (1:3.5.12-1) ...
Setting up php7.3-fpm (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b) ...

Creating config file /etc/php/7.3/fpm/php.ini with new version
Created symlink /etc/systemd/system/multi-user.target.wants/php7.3-fpm.service → /lib/systemd/system/php7.3-fpm.service.
Job for php7.3-fpm.service failed because a fatal signal was delivered to the control process.
See "systemctl status php7.3-fpm.service" and "journalctl -xe" for details.
invoke-rc.d: initscript php7.3-fpm, action "start" failed.
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Tue 2019-05-14 01:09:43 BST; 165ms ago
[... 28 lines cut ...]
Reading state information... Done
3 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  libpcre3 libssl1.1 openssl
3 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 2,441 kB of archives.
After this operation, 124 kB disk space will be freed.
Do you want to continue? [Y/n] y
Get:1 https://packages.sury.org/php stretch/main armhf libssl1.1 armhf 1.1.1b-1+0~20190228092419.10+stretch~1.gbp729d89 [1,314 kB]
Get:2 https://packages.sury.org/php stretch/main armhf libpcre3 armhf 2:8.42-1+0~20190203125157.5+stretch~1.gbp79d75d [322 kB]
Get:3 https://packages.sury.org/php stretch/main armhf openssl armhf 1.1.1b-1+0~20190228092419.10+stretch~1.gbp729d89 [806 kB]
Fetched 2,441 kB in 2s (951 kB/s)
Preconfiguring packages ...
(Reading database ... 17555 files and directories currently installed.)
Preparing to unpack .../libssl1.1_1.1.1b-1+0~20190228092419.10+stretch~1.gbp729d89_armhf.deb ...
Unpacking libssl1.1:armhf (1.1.1b-1+0~20190228092419.10+stretch~1.gbp729d89) over (1.1.0j-1~deb9u1) ...
Preparing to unpack .../libpcre3_2%3a8.42-1+0~20190203125157.5+stretch~1.gbp79d75d_armhf.deb ...
Unpacking libpcre3:armhf (2:8.42-1+0~20190203125157.5+stretch~1.gbp79d75d) over (2:8.39-3) ...
E: Sub-process /usr/bin/dpkg exited unexpectedly

maartenlangeveld, 2019-05-14:
> @DietPi
> Pressing `<ctrl>+<c>` will cancel any running process.
>
> @maartenlangeveld
> Kernel panic (system crash) occurred while installing the PHP packages? Any error messages before that? Any chance [...]

Yes yes yes yes. System worked flawlessy last six months, cannot believe it's hardware related. When update tot php7.3 takes place, kernel gets corrupted.

MetalGearSolidNerd, 2019-05-14:
I am facing similiar issue after trying to update to 6.23 from 6.22.

HW - Raspberry pi 1B (256 MB)
SW - Pi-hole

salt-lick, 2019-05-14:
> I am facing similiar issue after trying to update to 6.23 from 6.22.
> 
> HW - Raspberry pi 1B (256 MB)
> SW - Pi-hole

Not confused... wrong emoji selected.

I'm also experiencing this same issue.

Pi Zero W
SW - Pi-hole

maartenlangeveld, 2019-05-14:
PLEASE HELP.

Restored backup after fresh install (v6.21 image, that automatically update to v6.23...argh), seemed to go  well but after rebooted nothing happens... cannot connect through SSH (since it's a headless install I cannot see what's going wrong).

A real disaster.... !!

When I do fresh install of DietPi images (v6.21) if forces me to upgrade to v6.23. But I don not want that.

Is it possible to install DietPi v6.21 or v6.22 without that it first upgrades to v6.23 ? How to manage that ?

ZZFrag, 2019-05-14:
Not sure if its related, but when I tried to update to v6.23 I was met with an error. `curl: (7) Failed to connect to codeload.github.com port 443: Connection refused
[FAILED] DietPi-Update | Download failed, unable to run update. Please try running dietpi-update again.
`
Now my pihole-ftl service is broken and to fix required a reboot of the system. If I run the update again, the same happens.

Fourdee, 2019-05-14:
@MichaIng 

I'll try to replicate on RPi Zero on v6.22 with Pi-hole installed, seems a possible ARMv6 issue:
https://twitter.com/Cederqvist/status/1128196744850935809

Fourdee, 2019-05-14:
@maartenlangeveld 

> Powersupply: Samsung 5.0V 2.0A

These are insufficient for SBC use, they can drop voltage to 4.5v~ under load. May not be the cause in this instance, however, you can check for any reported undervolts with:
```
dmesg | grep volt
```
Also can test with a multimeter and/or USB power meter.

maartenlangeveld, 2019-05-14:
@Fourdee,

Thanks !!
When I've everything up and running, will check reported undervolts.

I did not run Pi-hole on my RPi Zero W by the way

maartenlangeveld, 2019-05-14:
@Fourdee 

 `dmesg | grep volt`

gives no results (as I expected)  :-)

Kind regards, Maarteb

MichaIng, 2019-05-14:
@Fourdee 
I will now also start on RPi2. To keep it clean, could you:
- Verify that it is not an issue with the latest RPi kernel/firmware update.
- Install PHP7.3 manually as mentioned above to rule out any interference (and save time for any other install steps 😉): https://github.com/MichaIng/DietPi/issues/2794#issuecomment-492017274

Actually IF it is really an issue with the Ondrej repo packages that cause a kernel panic, then the chance is high that it is the OpenSSL1.1 upgrade instead of PHP actually, what you think? I can't believe that there is such a major critical issue with these packages, I mean it's from the official Debian PHP maintainer and THE well known and most used method to install PHP7.3... But something must cause this, so...

maartenlangeveld, 2019-05-14:
@Fourdee @MichaIng
Thanks for the hard work you are doing !!!

MichaIng, 2019-05-14:
Okay based on reports here and in forum it is an ARMv6 issue, hence I blocked update for those via pre-patches: https://github.com/MichaIng/DietPi/commit/bb93e025376bb2483e392886bc7311de05a3c021

My guess is the openssl1.1 package. We need to block those and e.g. use the package we host on dietpi.com for Jessie images. Needs to be reported to Ondrej to fix this in his repo. Since we only ship APT packages SSL related (also the Jessie one ia from Debian APT repo) this most likely is not related to DietPi.

But lets see what testing results show.

PeterLacknase, 2019-05-14:
It's definitely php7.3 (at least the fpm portion, that is). Installed everything step by step from scratch. fail2ban, ssh, GPIO-sets, python3, pip, lighttpd etc. When installing php7.3 (for pihole) apt gives the 2 fingers salute:

`Setting up php7.3-fpm (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b) ...
Job for php7.3-fpm.service failed because a fatal signal was delivered to the control process.
See "systemctl status php7.3-fpm.service" and "journalctl -xe" for details.
invoke-rc.d: initscript php7.3-fpm, action "start" failed.
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Tue 2019-05-14 19:30:34 CEST; 134ms ago
     Docs: man:php-fpm7.3(8)
  Process: 14096 ExecStart=/usr/sbin/php-fpm7.3 --nodaemonize --fpm-config /etc/php/7.3/fpm/php-fpm.conf (code=killed, signal=ILL)
 Main PID: 14096 (code=killed, signal=ILL)

May 14 19:30:34 sphinx systemd[1]: Starting The PHP 7.3 FastCGI Process Man…r...
May 14 19:30:34 sphinx systemd[1]: php7.3-fpm.service: Main process exited,…/ILL
May 14 19:30:34 sphinx systemd[1]: Failed to start The PHP 7.3 FastCGI Proc…ger.
May 14 19:30:34 sphinx systemd[1]: php7.3-fpm.service: Unit entered failed …ate.
May 14 19:30:34 sphinx systemd[1]: php7.3-fpm.service: Failed with result '…al'.
Hint: Some lines were ellipsized, use -l to show in full.
dpkg: error processing package php7.3-fpm (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 php7.3-fpm
`

RaspberryPi Zero W (v.1.1)
SPI / i2c enabled
AC: 2.5 A
Sandisk Extreme 64 GB as well as Extreme 32 GB.

zerbes, 2019-05-14:
Same here, Raspi 1 bricked as result of upgrading php 7.3 within upgrade of DietPi to 6.23. Same issue appears during reinstallation of pi-hole.

MichaIng, 2019-05-14:
Just verified that it works well on RPi2, for RPi3 as well we have positive reports.

Also I was wrong: `libssl1.1` ist not pulled from sury repo because we explicitly block this as it breaks Lighttpd: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software#L3453
However it does not matter: @maartenlangeveld step-by-step test outside of DietPi-Software (so openssl+libssl1.1 block was not active), these got installed, but same brick.

Did anyone run into this issue with Apache, so without `php7.3-fpm`?

@Fourdee 
Could you run some tests on your RPi ARMv6?

PeterLacknase, 2019-05-14:
dietpi@sphinx:~$ journalctl -u php7.3-fpm

`-- Logs begin at Tue 2019-05-14 19:16:55 CEST, end at Tue 2019-05-14 21:39:43 CEST. --
Mai 14 19:29:34 sphinx systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Mai 14 19:29:34 sphinx systemd[1]: php7.3-fpm.service: Main process exited, code=killed, status=4/ILL
Mai 14 19:29:34 sphinx systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
Mai 14 19:29:34 sphinx systemd[1]: php7.3-fpm.service: Unit entered failed state.
Mai 14 19:29:34 sphinx systemd[1]: php7.3-fpm.service: Failed with result 'signal'.
Mai 14 19:30:34 sphinx systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Mai 14 19:30:34 sphinx systemd[1]: php7.3-fpm.service: Main process exited, code=killed, status=4/ILL
Mai 14 19:30:34 sphinx systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
Mai 14 19:30:34 sphinx systemd[1]: php7.3-fpm.service: Unit entered failed state.
Mai 14 19:30:34 sphinx systemd[1]: php7.3-fpm.service: Failed with result 'signal'.`

No 

/var/log/php-fpm.log 

since it never started.

`curl -sSL https://packages.sury.org/php/apt.gpg | apt-key add -`

`OK`

`echo 'deb https://packages.sury.org/php/ stretch main' > /etc/apt/sources.list.d/dietpi-php.list`

<prompt>

`apt update`

`OK:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
OK:2 https://packages.sury.org/php stretch InRelease
OK:3 https://archive.raspberrypi.org/debian stretch InRelease
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Aktualisierung für 1 Paket verfügbar. Führen Sie »apt list --upgradable« aus, um es anzuzeigen.`

` apt list --upgradable`

`Auflistung... Fertig
libpcre3/unbekannt 2:8.42-1+0~20190203125157.5+stretch~1.gbp79d75d armhf [aktualisierbar von: 2:8.39-3]
N: Es gibt 1 zusätzliche Version. Bitte verwenden Sie die Option »-a«, um sie anzuzeigen.`

`apt install php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip`

`Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Hinweis: »php-apcu« wird an Stelle von »php7.3-apcu« gewählt.
php-apcu ist schon die neueste Version (5.1.17+4.0.11-1+0~20190217111312.9+stretch~1.gbp192528).
php7.3-curl ist schon die neueste Version (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b).
php7.3-fpm ist schon die neueste Version (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b).
php7.3-gd ist schon die neueste Version (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b).
php7.3-mbstring ist schon die neueste Version (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b).
php7.3-xml ist schon die neueste Version (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b).
php7.3-zip ist schon die neueste Version (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
1 nicht vollständig installiert oder entfernt.
Nach dieser Operation werden 0 B Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] 
php7.3-fpm (7.3.5-1+0~20190503093827.38+stretch~1.gbp60a41b) wird eingerichtet ...
Job for php7.3-fpm.service failed because a fatal signal was delivered to the control process.
See "systemctl status php7.3-fpm.service" and "journalctl -xe" for details.
invoke-rc.d: initscript php7.3-fpm, action "start" failed.
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Tue 2019-05-14 22:03:09 CEST; 145ms ago
     Docs: man:php-fpm7.3(8)
  Process: 15803 ExecStart=/usr/sbin/php-fpm7.3 --nodaemonize --fpm-config /etc/php/7.3/fpm/php-fpm.conf (code=killed, signal=ILL)
 Main PID: 15803 (code=killed, signal=ILL)

Mai 14 22:03:09 sphinx systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Mai 14 22:03:09 sphinx systemd[1]: php7.3-fpm.service: Main process exited, code=killed, status=4/ILL
Mai 14 22:03:09 sphinx systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
Mai 14 22:03:09 sphinx systemd[1]: php7.3-fpm.service: Unit entered failed state.
Mai 14 22:03:09 sphinx systemd[1]: php7.3-fpm.service: Failed with result 'signal'.
dpkg: Fehler beim Bearbeiten des Paketes php7.3-fpm (--configure):
 Unterprozess installiertes post-installation-Skript gab den Fehlerwert 1 zurück
Fehler traten auf beim Bearbeiten von:
 php7.3-fpm
E: Sub-process /usr/bin/dpkg returned an error code (1)`

MichaIng, 2019-05-14:
@PeterLacknase 
Many thanks for pasting.

Does running the executable manually show any further errors?
```
/usr/sbin/php-fpm7.3 --nodaemonize --fpm-config /etc/php/7.3/fpm/php-fpm.conf
```

And do you as well run into kernel errors and failing reboot?

PeterLacknase, 2019-05-14:
I'll try, since I'll have to recover from (last-step-)backup. No reboot possible. 

It always ends with `illegal instructions` (deutsch: "Ungültiger Maschinenbefehl")

Under gentoo I'd say bad compiler flags, "check your march options" ...

MichaIng, 2019-05-14:
@PeterLacknase 
Fixing things by purging the PHP7.3 packages again does not work as well, most likely with the same illegal instructions error, right?

PeterLacknase, 2019-05-14:
Affirmative.

MichaIng, 2019-05-14:
I opened an issue on sury.org GitHub page. Hope this can be debugged past: https://github.com/oerdnj/deb.sury.org/issues/1167

The extremely annoying this issue must be for you guys, the much I am (besides being shocked of course) excited for the reason.
```
Processing triggers for systemd (232-25+deb9u11) ...
Errors were encountered while processing:
 php7.3-fpm
```
- Possibly there is a major bug in the post-processing script that indeed breaks systemd completely.
- Would explain why all systemd commands fail with `illegal instructions` (systemctl, journalctl, reboot, shutdown etc. while `apt` commands seems to still work).
- Then on boot when the bootloader wants to initiate `systemd` it fails as well.

But sadly I have absolutely not idea how to debug this.
_____________
If this cannot be fixed fast, another PHP7.3 source would be the Buster repo: https://packages.debian.org/buster/php7.3-fpm
- The only issue is that it for sure pulls several other dependency packages from there and I am not keen to risk a whole accidental dist upgrade to Buster, like it happened in few cases with the Sid repo for WireGuard.
___________
If anyone is in mood: `apt install php7.2-fpm`
- Does this fail the same way?

MichaIng, 2019-05-14:
Okay clear response:
- Debian armhf (including sury.org) is not compatible with ARMv6 devices.
- I thought this cannot be possible since we have many non-RPi armv6l devices, but, nope that was a wrong assumption: All out supported ARMv6 devices are the RPi1/Zero generation: https://dietpi.com/survey/#device

Only chance I see is to use the Raspbian Buster repo then.

fjvt, 2019-05-14:
Ran into same issue today with 2 fresh installs on 2 differen raspberry pi 1.

maartenlangeveld, 2019-05-14:
@MichaIng 

Why not stay with Raspbian for Armv6 ? As with v6.22.3 ? Is that not possible ?

Again, thanks a lot for the good and hard work you and Fourdee do for DietPi. Really appreciate.

MichaIng, 2019-05-15:
@maartenlangeveld 
Jep we will use Raspbian Buster for the PHP packages on ARMv6 then (since all ARMv6 are RPi anyway). The solution is not optimal since as mentioned above some other Buster (testing) packages will be pulled. sury.org has the advantage that it has a Stretch branch so no risk to accidentally upgrade large parts of the system to the testing branch.

PeterLacknase, 2019-05-15:
> @PeterLacknase
> Many thanks for pasting.
> 
> Does running the executable manually show any further errors?
> 
> ```
> /usr/sbin/php-fpm7.3 --nodaemonize --fpm-config /etc/php/7.3/fpm/php-fpm.conf
> ```
> 
> And do you as well run into kernel errors and failing reboot?
> 
> Possibly related? [oerdnj/deb.sury.org#1000](https://github.com/oerdnj/deb.sury.org/issues/1000)

Again: `illegal instructions`

MichaIng, 2019-05-15:
Fix ready for testing: https://github.com/MichaIng/DietPi/pull/2799
But note that I did not even do syntax check, have to go to work now. Will finish in the evening and release as v6.24 hotfix.

MichaIng, 2019-05-15:
Damn thing. We need to either raise the priority of Buster to 500+ (which means that potentially many other packages can be pulled on apt upgrade) or install via `-t buster`. But libc6 is upgraded to the Buster branch in any case, which I really don't like:
```
root@VM-Stretch:~# apt-get install -t buster php7.3-apcu
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-apcu' instead of 'php7.3-apcu'
The following additional packages will be installed:
  libargon2-1 libbsd0 libc-bin libc-l10n libc6 libedit2 libicu63 libmagic-mgc libmagic1 libncurses6 libpcre2-8-0 libsodium23 libtinfo6 libxml2 locales
  mime-support php-common php7.3-cli php7.3-common php7.3-json php7.3-opcache php7.3-phpdbg php7.3-readline
Suggested packages:
  glibc-doc file php-gd php-pear
Recommended packages:
  manpages libidn2-0 libgpm2 file xz-utils php-apcu-bc
The following NEW packages will be installed:
  libargon2-1 libbsd0 libedit2 libicu63 libmagic-mgc libmagic1 libncurses6 libpcre2-8-0 libsodium23 libtinfo6 libxml2 mime-support php-apcu php-common
  php7.3-cli php7.3-common php7.3-json php7.3-opcache php7.3-phpdbg php7.3-readline
The following packages will be upgraded:
  libc-bin libc-l10n libc6 locales
4 upgraded, 20 newly installed, 0 to remove and 190 not upgraded.
Need to get 23.0 MB of archives.
```

However I see no other way, since there is not really a way around installing PHP > 7.0..
_________
Things are even more complicated:
- On Raspbian Buster there are php7.2 packages available (luckily) but not for APCu and Redis.
- Since ownCloud and phpBB do not yet support php7.3, we cannot support those two software titles on ARMv6 RPi devices. There is even no chance to stay with php7.0 since support for this will be dropped soon on them as well.
- A pain... only hope is that phpBB 3.3 and ownCloud 10.3 will be released soon as both will support PHP7.3 then.

PeterLacknase, 2019-05-15:
Who said it couldn‘t get worse? Rebooted the 3B+ (arm7l) today ... php7.3 pid and socket were gone under /var/run and couldn‘t be created (during postboot). Had to kill things and restart the service. I think php7.3-implementation is more severly b0rked.

MichaIng, 2019-05-15:
@PeterLacknase 
Very strange, during my tests I just ran into the same issue with MariaDB (but with Buster packages on Stretch, not the sury.org). Error was:
`ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=127)`

Does this match your `systemctl status php7.3-fpm`?

A reboot fixed it.

```
PermissionsStartOnly=true
ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld
```
Should lead to service having sufficient permissions to to create the PID/process dir, but somehow it did not have or failed for another reason until I rebooted.
__________
Okay ran into the same when removing /run/php (is the same as /var/run, as this is a symlink). However this is a tmpfs so on reboot it is clear. As long as it cannot create the dir itself (why ever, it starts as root!) is must fail on boot. But I just successfully rebooted with PHP-FPM up 🤔.... Just tested again.
Even the following works:
systemctl stop php7.3-fpm
rmdir /run/php
reboot
....
After reboot, PHP is up as it should. Not sure how the dir is created there while it is not when manually removing the file and restarting the service 🤔.
Ah found it: `/etc/init/php7.3-fpm.conf`
This creates the run dirs on boot. Unreliable implementation, this should be done within the service start, IMO. However does this file exist on your RPi3?

PeterLacknase, 2019-05-15:
The file exists. I‘ll get some failover pi running and test  rebooting. (Btw. i always restart services  SysV-style.)

PeterLacknase, 2019-05-15:
After 

`sudo systemctl status php7.3-fpm.service`

i get

`php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Wed 2019-05-15 22:42:09 CEST; 3min 55s ago
     Docs: man:php-fpm7.3(8)
  Process: 604 ExecStart=/usr/sbin/php-fpm7.3 --nodaemonize --fpm-config /etc/php/7.3/fpm/php-fpm.conf (code=kil
 Main PID: 604 (code=killed, signal=TERM)

Mai 15 22:40:32 dietpi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Mai 15 22:42:09 dietpi systemd[1]: php7.3-fpm.service: Start operation timed out. Terminating.
Mai 15 22:42:09 dietpi systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
Mai 15 22:42:09 dietpi systemd[1]: php7.3-fpm.service: Unit entered failed state.`

MichaIng, 2019-05-15:
@PeterLacknase 
> Btw. i always restart services SysV-style.

Hmm, which command exactly, you mean with `/etc/init.d/php7.3-fpm` as argument? As long as the systemd unit is inactive this should run as well. I would wonder if both used different PID+socket locations. However when the systemd unit is active, the sysvinit service must fail and the other way round.

I strongly advice to use systemd here. Many packages do not install any sysvinit service any more, so this does not work at all. Even if there is only a sysvinit service available, systemd generates a systemd unit of it.

So `systemctl restart php7.3-fpm` is the way to go.
_____________________
> php7.3-fpm.service: Start operation timed out. Terminating.

Okay so there is no PID file or socket creation issue in your case. Please paste: `cat /var/log/php7.3-fpm.log`

PeterLacknase, 2019-05-15:
`sudo cat /var/log/php7.3-fpm.log`

gives nothing. 

`systemctl restart php7.3-fpm`

gives

```
Failed to restart php7.3-fpm.service: The name org.freedesktop.PolicyKit1 was not provided by any .service files
See system logs and 'systemctl status php7.3-fpm.service' for details.
```

`sudo systemctl status php7.3-fpm.service`

gives

```
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Wed 2019-05-15 23:15:23 CEST; 4min 29s ago
     Docs: man:php-fpm7.3(8)
  Process: 597 ExecStart=/usr/sbin/php-fpm7.3 --nodaemonize --fpm-config /etc/php/7.3/fpm/php-fpm.conf (code=killed, signal=TERM)
 Main PID: 597 (code=killed, signal=TERM)

Mai 15 23:13:46 dietpi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Mai 15 23:15:23 dietpi systemd[1]: php7.3-fpm.service: Start operation timed out. Terminating.
Mai 15 23:15:23 dietpi systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
Mai 15 23:15:23 dietpi systemd[1]: php7.3-fpm.service: Unit entered failed state.
Mai 15 23:15:23 dietpi systemd[1]: php7.3-fpm.service: Failed with result 'timeout'.
```

MichaIng, 2019-05-15:
@PeterLacknase 
Searched the `org.freedesktop.PolicyKit1 was not provided` error and it seems to show up in cases where `sudo` was forgotten. But at least on x86 I get a different error. However did you use sudo, as you your paste above does not contain it?

PeterLacknase, 2019-05-15:
Checked that (after another reboot and 62 secs postbox):

`sudo systemctl status php7.3-fpm.service`

gives

```
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Wed 2019-05-15 23:32:24 CEST; 8s ago
     Docs: man:php-fpm7.3(8)
  Process: 598 ExecStart=/usr/sbin/php-fpm7.3 --nodaemonize --fpm-config /etc/php/7.3/fpm/php-fpm.conf (code=killed, signal=TERM)
 Main PID: 598 (code=killed, signal=TERM)

Mai 15 23:30:47 dietpi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Mai 15 23:32:24 dietpi systemd[1]: php7.3-fpm.service: Start operation timed out. Terminating.
Mai 15 23:32:24 dietpi systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
Mai 15 23:32:24 dietpi systemd[1]: php7.3-fpm.service: Unit entered failed state.
Mai 15 23:32:24 dietpi systemd[1]: php7.3-fpm.service: Failed with result 'timeout'.
```

`sudo systemctl restart php7.3-fpm`

now gives

```
Job for php7.3-fpm.service failed because a timeout was exceeded.
See "systemctl status php7.3-fpm.service" and "journalctl -xe" for details.
```

`sudo journalctl -xe`

gives

```
-- The start-up result is done.
Mai 15 23:32:24 dietpi systemd[1]: Reached target Graphical Interface.
-- Subject: Unit graphical.target has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit graphical.target has finished starting up.
-- 
-- The start-up result is done.
Mai 15 23:32:24 dietpi systemd[1]: Starting Update UTMP about System Runlevel Changes...
-- Subject: Unit systemd-update-utmp-runlevel.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit systemd-update-utmp-runlevel.service has begun starting up.
Mai 15 23:32:24 dietpi systemd[1]: Started Update UTMP about System Runlevel Changes.
-- Subject: Unit systemd-update-utmp-runlevel.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit systemd-update-utmp-runlevel.service has finished starting up.
-- 
-- The start-up result is done.
Mai 15 23:32:24 dietpi systemd[1]: Starting Lighttpd Daemon...
-- Subject: Unit lighttpd.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit lighttpd.service has begun starting up.
Mai 15 23:32:25 dietpi systemd[1]: Started Lighttpd Daemon.
-- Subject: Unit lighttpd.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit lighttpd.service has finished starting up.
[... 34 lines cut ...]
Mai 15 23:32:33 dietpi sudo[1384]: pam_unix(sudo:session): session closed for user root
Mai 15 23:32:46 dietpi sudo[1393]:   dietpi : TTY=pts/0 ; PWD=/home/dietpi ; USER=root ; COMMAND=/bin/systemctl restart php7.3-fpm
Mai 15 23:32:46 dietpi sudo[1393]: pam_unix(sudo:session): session opened for user root by dietpi(uid=0)
Mai 15 23:32:46 dietpi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
-- Subject: Unit php7.3-fpm.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit php7.3-fpm.service has begun starting up.
Mai 15 23:32:57 dietpi CRON[1131]: pam_unix(cron:session): session closed for user root
Mai 15 23:34:16 dietpi systemd[1]: php7.3-fpm.service: Start operation timed out. Terminating.
Mai 15 23:34:16 dietpi systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
-- Subject: Unit php7.3-fpm.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit php7.3-fpm.service has failed.
-- 
-- The result is failed.
Mai 15 23:34:16 dietpi systemd[1]: php7.3-fpm.service: Unit entered failed state.
Mai 15 23:34:16 dietpi systemd[1]: php7.3-fpm.service: Failed with result 'timeout'.
Mai 15 23:34:16 dietpi sudo[1393]: pam_unix(sudo:session): session closed for user root
Mai 15 23:35:44 dietpi sudo[1425]:   dietpi : TTY=pts/0 ; PWD=/home/dietpi ; USER=root ; COMMAND=/bin/journalctl -xe
Mai 15 23:35:44 dietpi sudo[1425]: pam_unix(sudo:session): session opened for user root by dietpi(uid=0)
```

MichaIng, 2019-05-15:
@PeterLacknase 
And `cat /var/log/php7.3-fpm.log` still shows nothing? Note that due to RAMlog this is cleaned every hour. So in case you might need to re-trigger the error to fill again the log.

PeterLacknase, 2019-05-15:
Right after boot it's empty. After

`sudo systemctl start php7.3-fpm`

it shows

```
[15-May-2019 23:39:31] NOTICE: fpm is running, pid 1450
[15-May-2019 23:39:31] NOTICE: ready to handle connections
[15-May-2019 23:39:31] NOTICE: systemd monitor interval set to 10000ms
```

(BTW. thanks for all your patience!)

MichaIng, 2019-05-15:
@PeterLacknase 
So looks good so far. Is this also the only content after the service fails with timeout?

> BTW. thanks for all your patience!

The least I can do. It is of course our aim to have the PHP migration stable, so tracking down all issues before I can release v6.24 with the ARMv6 hotfix 😉.

[8 more replies]

closed 2019-05-15 by MichaIng