MichaIng/DietPi #2795: error while reinstalling SABnzbd

sifyer, 2019-05-14:
#### Details:
- Date           | Tue 14 May 14:16:29 AEST 2019
- Bug report     | N/A
- DietPi version | v6.23.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid XU3/XU4/HC1/HC2 (armv7l) (index=11)
- Kernel version | #1 SMP PREEMPT Thu Apr 5 12:46:33 UTC 2018
- Distro         | stretch (index=4)
- Command        | G_AGI par2 p7zip-full libffi-dev libssl-dev
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Error found when updating to 6.23.3; and also when running: dietpi-software reinstall 139
2. ...

#### Expected behaviour:

- Updating SABnzbd to latest version

#### Actual behaviour:

- Exits with error

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2019-05-14:
@sifyer 
Many thanks for your report.

Please paste the full output of:
apt-get -y install par2 p7zip-full libffi-dev libssl-dev

sifyer, 2019-05-14:
Hi, here's the output of that command:

`Reading package lists... Done
Building dependency tree
Reading state information... Done
libffi-dev is already the newest version (3.2.1-6).
p7zip-full is already the newest version (16.02+dfsg-3+deb9u1).
p7zip-full set to manually installed.
par2 is already the newest version (0.6.14-2).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libssl-dev : Depends: libssl1.1 (= 1.1.1b-1+0~20190228092419.10+stretch~1.gbp72                                              9d89) but 1.1.0j-1~deb9u1 is to be installed
E: Unable to correct problems, you have held broken packages.
`

MichaIng, 2019-05-15:
@sifyer 
AH okay, this must be due to our block of ssl packages from the PHP repo, since those break Lighttpd on Stretch.

Please do the following:
```
echo -e '# libssl1.1 from sury.org breaks Lighttpd install
Package: openssl libssl*\nPin: origin packages.sury.org\nPin-Priority: -1' > /etc/apt/preferences.d/dietpi-openssl
apt upgrade
```

sifyer, 2019-05-15:
Thanks. Did you mean those three lines to be distinct on newlines like that? The apt upgrade didn't modify anything...

```
root@DietPi:~# echo -e '# libssl1.1 from sury.org breaks Lighttpd install
> Package: openssl libssl*\nPin: origin packages.sury.org\nPin-Priority: -1' > /etc/apt/preferences.d/dietpi-openssl
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libltdl7 libmcrypt4
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

MichaIng, 2019-05-15:
@sifyer 
Jep exactly. `cat /etc/apt/preferences.d/dietpi-openssl` should show:
```
# libssl1.1 from sury.org breaks Lighttpd install
Package: openssl libssl*
Pin: origin packages.sury.org
Pin-Priority: -1
```

Okay APT upgrade also works as expected now. You can remove the non-required packages via: `G_AGA`

closed 2019-05-15 by MichaIng