MichaIng/DietPi #2796: Software request: sourcehat webservices

AKonia, 2019-05-14:
## Sourcehat services

#### Formal software information
- Software title: *Sourcehat Services*
- lightweight Git, CI, building, task-management and other webservices.
- Official URL 
- It's official git repositories [Sourcehat](https://git.sr.ht/) 
- (Official) install documentation [Official Man](https://man.sr.ht/installation.md)

#### Are there similar/alternative software titles available with DietPi-Software?
- Gitea
- Gogs

#### What makes your requested software better than the above solutions, if available?
- lightweight, no JS, other performance advances.
- security
- full and CI for different Linux and BSD distros(FreeBSD, Ubuntu, Alpine, Archlinux)
- sub-services could be customized, you can choose which sub-services turn on, it will be good not only for software developers:
1. git or hg as version control system, 
2. todo: same as Issues on github, 
3. builds: CI with ability to run independent(from repo) build scenarios - you can build any source repo on different distribution
and other).
4. mailing list
5. wiki
6. dispatch service
#### There exist official manual [man.sr.ht](https://man.sr.ht/installation.md). DietPi could automate those steps, which saves end users a lot of time and allows installation as well to less experienced Linux users.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
- installation process and all dependencies listed on official site manual [man.sr.ht](https://man.sr.ht/installation.md) and all packages listed if follow link `Packages installation` on `man.sr.ht`, more see in `man.sr.ht` and  [Official repositories with sources](https://git.sr.ht/%7Esircmpwn/?search=sr.ht).

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
- I could, but I have not much time till end of summer.

#### Vote for this software on [FeatHub](https://feathub.com/MichaIng/DietPi/+77)

MichaIng, 2019-05-15:
@AKonia 
Many thanks for your request. We'll have a look when we can.

MichaIng, 2019-05-18:
It's sourceh**u**t not sourcehad, right 😉? https://sourcehut.org/

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not be addressed soon. The FeatHub request stays however open to allow votes and if anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng