MichaIng/DietPi #2797: dietpi-update v6.23.3 failed on Odroid xu4

Jmi91, 2019-05-14:
--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Tue 14 May 16:28:33 CEST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid XU3/XU4/HC1/HC2 (armv7l) (index=11)
- Kernel version | #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018
- Distro         | stretch (index=4)
- Command        | G_AGP
- Exit code      | 100
- Software title | DietPi-Patchfile

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
(Reading database ... 41028 files and directories currently installed.)
Preparing to unpack .../php-ssh2_1.1.2+0.13-4+0~20190220145736.7+stretch~1.gbpc540cd_armhf.deb ...
WARNING: Module ssh2 ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module ssh2 ini file doesn't exist under /etc/php/7.3/mods-available
Unpacking php-ssh2 (1.1.2+0.13-4+0~20190220145736.7+stretch~1.gbpc540cd) over (1.0+0.13-2) ...
(Reading database ... 41035 files and directories currently installed.)
Removing libapache2-mod-php7.0 (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
ERROR: Module php7.0 does not exist!
dpkg: error processing package libapache2-mod-php7.0 (--remove):
 subprocess installed pre-removal script returned error exit status 1
Removing php7.0-zip (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
Removing php7.0-xml (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
Removing php7.0-curl (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
Removing php7.0-gd (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
Removing php7.0-mbstring (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
Removing php7.0-mcrypt (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
Removing php7.0-mysql (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
dpkg: php7.0-cli: dependency problems, but removing anyway as you requested:
 libapache2-mod-php7.0 depends on php7.0-cli.

Removing php7.0-cli (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
dpkg: php7.0-common: dependency problems, but removing anyway as you requested:
 php7.0-soap depends on php7.0-common (= 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991).
 php7.0-json depends on php7.0-common (= 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991).
 libapache2-mod-php7.0 depends on php7.0-common (= 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991); however:
  Package php7.0-common is to be removed.
 php7.0-opcache depends on php7.0-common (= 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991).
 php7.0-readline depends on php7.0-common (= 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991).

Removing php7.0-common (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
dpkg: php7.0-json: dependency problems, but removing anyway as you requested:
 libapache2-mod-php7.0 depends on php7.0-json; however:
  Package php7.0-json is to be removed.

Removing php7.0-json (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
dpkg: php7.0-opcache: dependency problems, but removing anyway as you requested:
 libapache2-mod-php7.0 depends on php7.0-opcache; however:
  Package php7.0-opcache is to be removed.

Removing php7.0-opcache (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
Removing php7.0-readline (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
Errors were encountered while processing:
 libapache2-mod-php7.0
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

--------------------------------------------------------------------

[FAILED] DietPi-Patchfile | Unable to continue, DietPi-Patchfile will now terminate.
[  OK  ] DietPi-Update | Patching to v6.23.3 completed

[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com
[  OK  ] DietPi-Survey | Successfully sent survey data

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Completed

[ INFO ] DietPi-Update | Current version : v6.23.3
[ INFO ] DietPi-Update | Latest version  : v6.23.3
[  OK  ] DietPi-Update | Syncing new DietPi scripts to disk
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : avahi-daemon
[  OK  ] DietPi-Services | restart : vsftpd
[  OK  ] DietPi-Services | restart : nmbd
[  OK  ] DietPi-Services | restart : smbd
[  OK  ] DietPi-Services | restart : mariadb
[  OK  ] DietPi-Services | restart : apache2
[  OK  ] DietPi-Services | restart : cron
[ SUB2 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process_tool | Completed

The problem seems to be with PHP 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991
I cannot remove php 7.0.33

MichaIng, 2019-05-14:
@Jmi91 
Many thanks for reporting.

`apt purge libapache2-mod-php7.0` throws the same error? Strange one... It does not find the related Apache module even that it is installed and about the be removed. Why not simply ignoring when the module does not exist... However

Please try: `apt -f install && apt purge libapache2-mod-php7.0`

If this does not work, try to install it again, verify the module is there and then purge again...
```
apt install --reinstall libapache2-mod-php7.0
a2enmod php7.0 # Check if this does not throw an error
apt purge libapache2-mod-php7.0
G_AGP *php7.0*
```

Jmi91, 2019-05-14:
Hi,

I don't know how to remove PHP 7.0...

root@DietPi:~# apt -f install && apt purge libapache2-mod-php7.0
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up php7.0-fpm (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...

Creating config file /etc/php/7.0/fpm/php.ini with new version
NOTICE: Not enabling PHP 7.0 FPM by default.
NOTICE: To enable PHP 7.0 FPM in Apache2 do:
NOTICE: a2enmod proxy_fcgi setenvif
NOTICE: a2enconf php7.0-fpm
NOTICE: You are seeing this message because you have apache2 package installed.
Created symlink /etc/systemd/system/multi-user.target.wants/php7.0-fpm.service → /lib/systemd/system/php7.0-fpm.service.
Processing triggers for systemd (232-25+deb9u11) ...
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  libapache2-mod-php7.0*
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
After this operation, 2,501 kB disk space will be freed.
Do you want to continue? [Y/n]
(Reading database ... 42110 files and directories currently installed.)
Removing libapache2-mod-php7.0 (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
ERROR: Module php7.0 does not exist!
dpkg: error processing package libapache2-mod-php7.0 (--remove):
 subprocess installed pre-removal script returned error exit status 1
Errors were encountered while processing:
 libapache2-mod-php7.0
E: Sub-process /usr/bin/dpkg returned an error code (1)

root@DietPi:~# apt -f install libapache2-mod-php7.0
Reading package lists... Done
Building dependency tree
Reading state information... Done
libapache2-mod-php7.0 is already the newest version (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

root@DietPi:~# apt install --reinstall libapache2-mod-php7.0
Reading package lists... 0%
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
Need to get 1,052 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://packages.sury.org/php stretch/main armhf libapache2-mod-php7.0 armhf 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991 [1,052 kB]
Fetched 1,052 kB in 0s (2,635 kB/s)
(Reading database ... 42111 files and directories currently installed.)
Preparing to unpack .../libapache2-mod-php7.0_7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991_armhf.deb ...
Unpacking libapache2-mod-php7.0 (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) over (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
Setting up libapache2-mod-php7.0 (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
libapache2-mod-php7.0: not switching MPM - already enabled

root@DietPi:~# a2enmod php7.0
ERROR: Module php7.0 does not exist!

Fourdee, 2019-05-15:
@Jmi91 
Try reinstalling the deb9u3_armhf.deb, then remove:
```
wget http://ftp.us.debian.org/debian/pool/main/p/php7.0/libapache2-mod-php7.0_7.0.33-0+deb9u3_armhf.deb
dpkg - i *.deb
rm *.deb
apt purge libapache2-mod-php7.0
```

Fourdee, 2019-05-15:
@MichaIng 

Appears to be an external bug with ```20190503101027.13+stretch~1.gbp26f991``` version, unable to remove itself in this instance. Will flag as external for now.

Jmi91, 2019-05-15:
root@DietPi:/tmp# dpkg -i libapache2-mod-php7.0_7.0.33-0+deb9u3_armhf.deb
dpkg: warning: downgrading libapache2-mod-php7.0 from 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991 to 7.0.33-0+deb9u3
(Reading database ... 42111 files and directories currently installed.)
Preparing to unpack libapache2-mod-php7.0_7.0.33-0+deb9u3_armhf.deb ...
Unpacking libapache2-mod-php7.0 (7.0.33-0+deb9u3) over (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
dpkg: dependency problems prevent configuration of libapache2-mod-php7.0:
 libapache2-mod-php7.0 depends on php7.0-common (= 7.0.33-0+deb9u3); however:
  Version of php7.0-common on system is 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991.

dpkg: error processing package libapache2-mod-php7.0 (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libapache2-mod-php7.0

Fourdee, 2019-05-15:
@Jmi91 

```
G_AGF
```

Jmi91, 2019-05-15:
root@DietPi:/tmp# dpkg -i libapache2-mod-php7.0_7.0.33-0+deb9u3_armhf.deb
dpkg: warning: downgrading libapache2-mod-php7.0 from 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991 to 7.0.33-0+deb9u3
(Reading database ... 42111 files and directories currently installed.)
Preparing to unpack libapache2-mod-php7.0_7.0.33-0+deb9u3_armhf.deb ...
Unpacking libapache2-mod-php7.0 (7.0.33-0+deb9u3) over (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
dpkg: dependency problems prevent configuration of libapache2-mod-php7.0:
 libapache2-mod-php7.0 depends on php7.0-common (= 7.0.33-0+deb9u3); however:
  Version of php7.0-common on system is 7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991.

dpkg: error processing package libapache2-mod-php7.0 (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libapache2-mod-php7.0

root@DietPi:/tmp# G_AGF
[ INFO ] APT fix, please wait...
(Reading database ... 42111 files and directories currently installed.)
Preparing to unpack .../libapache2-mod-php7.0_7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991_armhf.deb ...
Unpacking libapache2-mod-php7.0 (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) over (7.0.33-0+deb9u3) ...
Setting up libapache2-mod-php7.0 (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
libapache2-mod-php7.0: not switching MPM - already enabled
[  OK  ] G_AGF

root@DietPi:/tmp# apt purge libapache2-mod-php7.0
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  libapache2-mod-php7.0*
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
After this operation, 2,501 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 42110 files and directories currently installed.)
Removing libapache2-mod-php7.0 (7.0.33-7+0~20190503101027.13+stretch~1.gbp26f991) ...
ERROR: Module php7.0 does not exist!
dpkg: error processing package libapache2-mod-php7.0 (--remove):
 subprocess installed pre-removal script returned error exit status 1
Errors were encountered while processing:
 libapache2-mod-php7.0
E: Sub-process /usr/bin/dpkg returned an error code (1)

Fourdee, 2019-05-15:
@Jmi91 

> ERROR: Module php7.0 does not exist!

I can only assume this can be removed, once php7.0 is installed, then manually remove ```libapache2-mod-php7.0``` before ```php7.0```.

Try:
```
G_AGI php7.0
G_AGF
G_AGP libapache2-mod-php7.0
G_AGP php7.0
G_AGA
```

Failing the above, ```dpkg``` has an option to force broken package removal (https://stackoverflow.com/questions/48431372/removing-broken-packages-in-ubuntu):
```
dpkg --remove --force-remove-reinstreq libapache2-mod-php7.0
```

MichaIng, 2019-05-15:
@Jmi91 @Fourdee 
> ERROR: Module php7.0 does not exist!

I think it's about the Apache2 module, so another approach is to manually create it:
```
> /usr/lib/apache2/modules/libphp7.0.so
cat << _EOF_ > /etc/apache2/mods-avaialble/php7.0.load
LoadModule php7_module /usr/lib/apache2/modules/libphp7.0.so
_EOF_
```

Jmi91, 2019-05-16:
@MichaIng 
echo 'LoadModule php7_module /usr/lib/apache2/modules/libphp7.0.so' > /etc/apache2/mods-available/php7.0.load

That did the trick !!! Thank you !
Now I have removed php7.0

Should I try to re install the DIetPi patch ? Is there a command to force a reinstall ?

Thanks !

Jmi91, 2019-05-16:
Ok just updated to DietPi v6.24.0
All went smoothly !
Thanks again !

MichaIng, 2019-05-16:
@Jmi91 
Okay perfect. I will leave the issue opened to create a bug report on Debian bug tracker. Not a big issue that many users will face but annoying that an uninstall fails because some file `that is about to be removed` is not found...

MichaIng, 2020-03-03:
Bug reported to Debian bug tracker: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=953042

closed 2020-03-03 by MichaIng