MichaIng/DietPi #2801: Plex, Emby and access to mounted SMB shares

ilgrank, 2019-05-15:
Hi!
I'm currently having trouble making Emby access SMB shares:

I installed Emby, and used `dietpi-drive_manager` to setup access to a windows SMB share.

While logged in as `root`, I can access the share and its content with no problem.

Emby process however, runs with the `emby `user, and it gets an access denied on the same shares. I verified it by logging in as emby:
`su -s /bin/bash -l emby`
and doing an `ls -la`:
`drwxrwx---  2 dietpi dietpi 4096 Feb 22 19:45 MP`

What do i need to do to allow Emby (and Plex) to access the mounted shared in the "dietpi-way"? (ie without manually altering /etc/fstab , which in turn would break `dietpi-drive_manager`)?

Thanks for your great work!

MichaIng, 2019-05-15:
@ilgrank 
Many thanks for your report.

> Is adding emby and plex users to the dietpi group a viable and secure option?

Jep that is the desired setup. Actually this should be already done by the DietPi-Software. Will verfy and in case add it, if missing.

```
usermod -a -G dietpi plex
usermod -a -G dietpi emby
```

ilgrank, 2019-05-15:
Thanks! Don't know why, but they were just part of `video`, and not of `dietpi`. Anyway, did it 
```
root@DietPi:/mnt# groups emby
emby : video dietpi
root@DietPi:/mnt# groups plex
plex : video dietpi

```
and now both Plex and Emby are able to access the SMB shares.. they both do scan the folder as libraries, yet none of them seems to be able to stream from them.. :/

ilgrank, 2019-05-15:
For some reason, in my install both `plex `and `emby ` (with _emby_ installed some time ago, but _plex_ just installed) were not part of the `dietpi `group)
Doing as @MichaIng  suggested, ie adding both users to the dietpi group and then restarting both services fixed the problem! 
Again, thanks @MichaIng !

closed 2019-05-15 by ilgrank

MichaIng, 2019-05-15:
@ilgrank 
Indeed those steps were missing. Added them for v6.24: https://github.com/MichaIng/DietPi/commit/157970ef14bf2f329e0188531013b07072622ae5
Will be released in the evening as it contains a critical hotfix for ARMv6 devices.

Plex server should have had already access since the service itself runs as dietpi group. But for Emby this was indeed not the case. Regardless now both users are added to dietpi group.

Many thanks for reporting!