MichaIng/DietPi #2802: dietpi-wifi-monitor	failed (Result: exit-code)

GulyFMG, 2019-05-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=23
G_DIETPI_VERSION_RC=3
G_GITBRANCH=master
G_GITOWNER=MichaIng`
- Distro version | 9.9
- Kernel version | Linux blacksun 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | Standart
- SDcard used | SanDisk 16Gb

#### Additional Information (if applicable)

- Bug report ID | 36406472-2e79-47f0-b6c3-82d2e78d0784

#### Steps to reproduce

Just boot RPI and every single time the script fails.

#### Expected behaviour

Should start without any error.

#### Actual behaviour

dietpi-wifi-monitor	failed (Result: exit-code)

#### Extra details
[FAILED] DietPi-Services | dietpi-wifi-monitor	failed (Result: exit-code) since Wed 2019-05-15 18:20:07 BST; 1min 14s ago
● dietpi-wifi-monitor.service - DietPi-WiFi-Monitor
   Loaded: loaded (/etc/systemd/system/dietpi-wifi-monitor.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-05-15 18:20:07 BST; 1min 15s ago
  Process: 753 ExecStart=/var/lib/dietpi/services/dietpi-wifi-monitor.sh (code=exited, status=1/FAILURE)
  Process: 750 ExecStartPre=/bin/dash -c [ -e /sys/class/net/wlan$(sed -n 2p /DietPi/dietpi/.network) ] (code=exited, status=0/SUCCESS)
  Process: 739 ExecStartPre=/DietPi/dietpi/func/obtain_network_details (code=exited, status=0/SUCCESS)
 Main PID: 753 (code=exited, status=1/FAILURE)

May 15 18:20:07 blacksun systemd[1]: Starting DietPi-WiFi-Monitor...
May 15 18:20:07 blacksun systemd[1]: Started DietPi-WiFi-Monitor.
May 15 18:20:07 blacksun dietpi-wifi-monitor.sh[753]: ERROR: Concurrent execution detected. Please exit the running instance of DietPi-WiFi-Monitor first. Exiting...
May 15 18:20:07 blacksun systemd[1]: dietpi-wifi-monitor.service: Main process exited, code=exited, status=1/FAILURE
May 15 18:20:07 blacksun systemd[1]: dietpi-wifi-monitor.service: Unit entered failed state.
May 15 18:20:07 blacksun systemd[1]: dietpi-wifi-monitor.service: Failed with result 'exit-code'.

MichaIng, 2019-05-15:
@GulyFMG 
Many thanks for reporting.

Indeed the concurrency check fails. Somehow `pgrep` does not exclude the own process any more.

To fix this, please do:
```
sed -i "s|pgrep -f 'dietpi-wifi-monitor\.sh' \&> /dev/null|(( $(pgrep -cf 'dietpi-wifi-monitor\.sh') > 1 ))|" /var/lib/dietpi/services/dietpi-wifi-monitor.sh
```

MichaIng, 2019-05-15:
Resolved with just released v6.24. Done as hotfix due to critical issue on ARMv6.

GulyFMG, 2019-05-16:
Just to do a follow up... there is no more errors on boot but i noticed on journalctl:

May 16 14:42:12 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:12 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:12 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: sleep: invalid time interval ‘{TICKRATE:-10}’
May 16 14:42:13 blacksun dietpi-wifi-monitor.sh[746]: Try 'sleep --help' for more information.
root@blacksun:~# sleep --help
Usage: sleep NUMBER[SUFFIX]...
  or:  sleep OPTION
Pause for NUMBER seconds.  SUFFIX may be 's' for seconds (the default),
'm' for minutes, 'h' for hours or 'd' for days.  Unlike most implementations
that require NUMBER be an integer, here NUMBER may be an arbitrary floating
point number.  Given two or more arguments, pause for the amount of time
specified by the sum of their values.

Fourdee, 2019-05-16:
@GulyFMG 

Confirmed, will fix.

MichaIng, 2019-05-16:
Ah did already fix. Syntax. Will merge this into stable silently.

Fourdee, 2019-05-16:
@MichaIng 

Just about to commit lol :+1: 

Recommend workaround, loop will run at full speed.

MichaIng, 2019-05-16:
Actually the MOTD implementation via script allows to push such small fixes live. So sed faulty line and restart service if it runs (with &> /dev/null). Will verify this on notebook later and adjust MOTD. Should be not used on regular basis but for such a small ones it should work well.

Since MOTD is enabled by default this should cover most systems and all newly updated ones.

Fourdee, 2019-05-16:
@MichaIng 

> Actually the MOTD implementation via script allows to push such small fixes live.

Nice idea :+1: 

Personally, I would not use this feature. Updates/fixes/changes should only be applied when the user selects them (end user trust). Risky if it goes wrong aswell?

MichaIng, 2019-05-18:
I mark this as closed. Will release v6.24.1 with this fix until the night, which is mostly required as hotfix for ARMv6 anyway.

closed 2019-05-18 by MichaIng

GulyFMG, 2019-05-18:
thank you for the feedback... btw this build is solid as hell. congrats

MichaIng, 2019-05-18:
@GulyFMG 
Thanks for feedback! Fine tuning/fixes for ARMv6 are ready as well to ship later: https://github.com/MichaIng/DietPi/pull/2809